esphome/issues #6161: Tuya Fan: oscillation_datapoint behaves read-only

Ceralor, 2024-08-16:
### The problem

As stated, the `oscillation_datapoint` feature for Tuya fans does not appear to allow changing the oscillation state. It updates and changes state based on toggling on the tower fan itself. Adding an additional Tuya switch allows altering this datapoint to toggle oscillation, but is a separate switch entity.

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.1

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

tuya_fan

### Example YAML snippet

```yaml
fan:
  - platform: tuya
    name: Fan
    switch_datapoint: 1
    speed_datapoint: 3
    oscillation_datapoint: 5
    speed_count: 3

select:
  - platform: tuya
    name: Fan Mode
    enum_datapoint: 2
    icon: 'mdi:wind-power'
    options:
      0: "None"
      1: Breeze
      2: Sleep

number:
  - platform: tuya
    name: Fan Sleep Timer
    number_datapoint: 22
    min_value: 0
    max_value: 15
    step: 1
    unit_of_measurement: hours
    entity_category: config
    icon: 'mdi:fan-clock'

switch:
  - platform: tuya
    name: Fan Oscillation
    entity_category: config
    icon: 'mdi:arrow-oscillating'
    switch_datapoint: 5
```

### Anything in the logs that might be useful for us?

```txt
[23:45:15][C][tuya:041]: Tuya:
[23:45:15][C][tuya:056]:   Datapoint 1: switch (value: ON)
[23:45:15][C][tuya:062]:   Datapoint 2: enum (value: 0)
[23:45:15][C][tuya:058]:   Datapoint 3: int value (value: 1)
[23:45:15][C][tuya:056]:   Datapoint 5: switch (value: OFF)
[23:45:15][C][tuya:062]:   Datapoint 22: enum (value: 0)
[23:45:15][C][tuya:074]:   Product: '{"p":"olswa3cicr6nlcz0","v":"1.0.0","m":2}'
#### Below is when I selected Oscillation On in Home Assistant
[23:45:25][D][fan:021]: 'Fan' - Setting:
[23:45:25][D][fan:027]:   Oscillating: YES
#### Seconds later during DP update
[23:45:32][D][tuya:335]: Datapoint 1 update to ON
[23:45:32][D][fan:120]: 'Fan' - Sending state:
[23:45:32][D][fan:121]:   State: ON
[23:45:32][D][fan:123]:   Speed: 1
[23:45:32][D][fan:126]:   Oscillating: NO
[23:45:32][D][tuya:355]: Datapoint 2 update to 0
[23:45:32][D][select:015]: 'Fan Mode': Sending state None (index 0)
[23:45:32][D][tuya:343]: Datapoint 3 update to 1
[23:45:32][D][fan:120]: 'Fan' - Sending state:
[23:45:32][D][fan:121]:   State: ON
[23:45:32][D][fan:123]:   Speed: 1
[23:45:32][D][fan:126]:   Oscillating: NO
[23:45:32][D][tuya:335]: Datapoint 5 update to OFF
[23:45:32][D][fan:120]: 'Fan' - Sending state:
[23:45:32][D][fan:121]:   State: ON
[23:45:32][D][fan:123]:   Speed: 1
[23:45:32][D][fan:126]:   Oscillating: NO
[23:45:32][D][tuya:355]: Datapoint 22 update to 0
[23:45:32][D][number:012]: 'Fan Sleep Timer': Sending state 0.000000
```

### Additional information

Logs show that ESPHome received the Oscillation ON request from Home Assistant, but it doesn't actually change anything. The next DP query, the log outputs that it received nothing different.

pethans, 2024-10-16:
I am seeing the same behavior with ESPHome 2024.9.2 and a Holmes Tuya fan with similar datapoints. As above, a switch entity can change oscillation but the fan entity, while correctly showing "Oscillating: YES/NO" in the logs, cannot change the oscillation mode. Incidentally I replaced the Tuya chip in this fan with an ESP32-C3-supermini board.

BenJamesAndo, 2024-10-31:
I'm also experiencing this exact same issue. I swapped out a Tuya chip with an ESP-C3-12F in my Dream Maker Tuya fan.
I used to have a different ESPHome Tuya fan and the oscillating worked. This was over a year ago that I last used it so perhaps it's a recent change that has caused this issue.

As a workaround I tried to see if I could get ESPHome to store the logs of oscillating enabling or disabling and then toggle the tuya switch entity for oscillating, but I couldn't figure it out.

BenJamesAndo, 2024-11-04:
I just tried flashing it with ESPHome version 2023.10 and the problem still persists. I found this issue https://github.com/home-assistant/core/issues/66263 that relates to it. I've also tried `http://192.168.1.113/fan/pedestal_fan/turn_on?set_oscillation=true` I get 200 response but oscillation doesn't turn on.

Now I've tried adding the ESPHome device to MQTT with
```
mqtt:
  broker: ip.address
  username: username
  password: password
```

Setting oscillate through HA dashboard with MQTT fan entity doesn't work. Even sending MQTT command of 
```
Topic: dream-maker-fan/fan/pedestal_fan/oscillation/state
Payload: oscillate_on
```
doesn't work. I'm not sure if it's HA or ESPHome related but seems more ESPHome related.

Ricetato, 2024-11-16:
Also having this same issue in 2024.10.3. I tried finding when the regression happened by going back to 2024.3.2 to test some other point in time, tried going back a bit further (2023.12) but complained about ota versions and dont have a serial flasher

pethans, 2024-11-17:
Was browsing the code over coffee this morning and seem to have found the issue. Submitted PR esphome/esphome#7776

closed 2024-11-17 by jesserockz

BenJamesAndo, 2024-11-17:
@pethans thanks for aiming to get this fixed. I installed ESPHome dev and re-compiled the yaml for my Tuya fan but oscillation is still not working. Is it working for you?

BenJamesAndo, 2024-11-18:
Oh wait it is fixed, I wrongly assumed the dev ESPHome version I was using was built with this change already. From what I can see the dev docker container is built once a day. I've now just updated to the latest version and oscillation is working! Thanks so much!