esphome/issues #6167: The Bluetooth Proxy installed via web installer can't be imported to the ESPHome dashboard

finity69x2, 2024-08-17:
### The problem

I installed the ESPHome BLE proxy via the BLE proxy web page (https://esphome.io/projects/?type=bluetooth).

It installed correctly but I can't get it to import into the ESPHome Dashboard.

Reports are that there should be a discovered device with an "adopt" button but that has never showed up in my ESPHome dashboard. It's bee two days and after multiple HA restarts.

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.7.3

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

BLE proxy

### Anything in the logs that might be useful for us?

```txt
I can't get to the logs of the device since I have no way to interface with the device itself.
```

wappinghigh, 2024-09-04:
I have the same problem

geotibi, 2024-10-16:
me too. any news on this?

geotibi, 2024-10-16:
I found the issue. Just disabled "Use ping for status" and got the Adopt device in ESPhome dashboard

[image]