esphome/issues #6170: [media_player] on_play event does not fire when playing announcement

veitw, 2024-08-17:
### The problem

I use a GPIO switch to power up the amplifier of my speakers when media starts playing, and to power it down when `media_player` goes to either idle or pause.

After updating HA from 2024.7 to 2024.8 and ESPhome from 0.6.1 to 0.7.3, this no longer works, as for announcements, the `media_player.on_play` event seems to get no longer fired when playback starts on an announcement, but non-announcement media seems to play correctly.

In contrast to `media_player.on_play` not firing when playing an announcement, `media_player.on_state` **does** fire, but `media_player.is_playing` is still false in this case.

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.2

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

media_player

### Example YAML snippet

```yaml
media_player:
  - platform: i2s_audio
    id: dac
    name: "Küche Lautsprecher"
    dac_type: external
    i2s_dout_pin: GPIO14
    mode: mono
    on_play:
      - logger.log: "Playback started."
    on_pause:
      - logger.log: "Playback paused."
    on_idle:
      - logger.log: "Playback stopped."
```

### Anything in the logs that might be useful for us?

```txt
Playing an MP3 file from chime_tts' cache as media stream (not as announcement) works (on_play emits "Playback started"):

[01:27:08][D][media_player:061]: 'Küche Lautsprecher' - Setting
[01:27:08][D][media_player:068]:   Media URL: http://192.168.xxx.xxx:8123/media/local/sounds/temp/chime_tts/00ulwt3g.mp3?authSig=xxx
[01:27:08][D][main:645]: Playback started.
[01:27:14][D][main:639]: Playback stopped.

However playing an announcement via TTS does NOT work (on_play not firing, no "Playback started." logged):

[01:27:53][D][media_player:061]: 'Küche Lautsprecher' - Setting
[01:27:53][D][media_player:068]:   Media URL: http://192.168.xxx.xxx:8123/api/tts_proxy/640ab2bae07bedc4c163f679a746f7ab7fb5d1fa_de_-_google_translate.mp3
[01:27:53][D][media_player:074]:  Announcement: yes
[01:27:53][D][main:639]: Playback stopped.
```

### Additional information

My workaround is currently to use `media_player.on_state` instead and only switch off the amp if `media_player.is_idle` is true or else switch it on. But this is an incomplete workaround as `media_player.is_paused` does not exist.

n00btotal, 2024-08-22:
Having the same issue

veitw, 2024-08-23:
The problem persists on ESPhome version 2024.8.0.

guillempages, 2024-08-25:
I think there was a new `on_announcement` trigger implemented some time ago for the media player, that should trigger in the announcement case (instead of `on_play`). The documentation has not been updated yet due to some misunderstanding, but the functionality should be available. (Docs PR:  https://github.com/esphome/esphome-docs/pull/3822)

Could you try with that?

veitw, 2024-08-25:
Problem resolved, documentation just not up to date yet and not communicated as breaking change.

closed 2024-08-25 by veitw

olimexsmart, 2024-09-20:
Documentation at [](https://esphome.io/components/media_player/) still not updated with the addition of `on_announcement`

guillempages, 2024-09-21:
Fixed.

TheStigh, 2024-09-23:
Still, this trigger ` media_player.is_playing: ` doesn't seem to hold the automation, it just continues and run the next command even while announcement is spoken. There should have been a trigger also for `on_announcement`

```
  on_end:
    - wait_until:
        not:
          media_player.is_playing: onju_out
    - script.execute: reset_led
```