esphome/issues #6179: Warning/Error at log when update device from 2024.7.3 to 2024.8.0

KitHubek, 2024-08-21:
### The problem

Updating device to newest  2024.8.0 and see this error/warning at log during update process. Update done with SUCCES mark

[image]

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.2

### What platform are you using?

ESP32

### Board

esp32dev and esp32doit-devkit-v1 (wt32-eth01)

### Component causing the issue

dont know

### Anything in the logs that might be useful for us?

```txt
src/esphome/core/bytebuffer.cpp: In member function 'float esphome::ByteBuffer::get_float()':
src/esphome/core/bytebuffer.cpp:86:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   return *(float *) &value;
           ^~~~~~~~~~~~~~~~
src/esphome/core/bytebuffer.cpp: In member function 'void esphome::ByteBuffer::put_float(float)':
src/esphome/core/bytebuffer.cpp:129:61: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 void ByteBuffer::put_float(float value) { this->put_uint32(*(uint32_t *) &value); }
                                                             ^~~~~~~~~~~~~~~~~~~
```

MEKadan, 2024-08-21:
Got the same. However, from my 7 devices only on those with platform `nodemcuv2` which are 8266. On various ESP32 devices not

VlastiBroucek, 2024-08-21:
Same issue here.
board: m5stack-atom
framework_type: esp-idf

`Compiling .pioenvs/bt-proxy/src/esphome/core/component.o
src/esphome/core/bytebuffer.cpp: In member function 'float esphome::ByteBuffer::get_float()':
src/esphome/core/bytebuffer.cpp:86:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   return *(float *) &value;
           ^~~~~~~~~~~~~~~~
src/esphome/core/bytebuffer.cpp: In member function 'void esphome::ByteBuffer::put_float(float)':
src/esphome/core/bytebuffer.cpp:129:61: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 void ByteBuffer::put_float(float value) { this->put_uint32(*(uint32_t *) &value); }
                                                             ^~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/bt-proxy/src/esphome/core/component_iterator.o
`

davlaw, 2024-08-21:
Getting this with the default Bluetooth proxy code as well..

011V32, 2024-08-21:
Same for me with olimex-bluetooth-proxy and atom-bluetooth-proxy

zoomtronicOR, 2024-08-21:
Same thing here

ThiSaadeh, 2024-08-21:
Same thing occurring while updating Everything Presence Lite.

SaszkaX, 2024-08-21:
Same here. ESP8266, after update Core 2024.8.1 -> 2024.8.2 and HAOS 13.0 -> 13.1 (but there was no ESPHome update - it's still 2024.8.0), after reboot there was new update of all ESP8266 devices (GoSund P1 & GoSund SP111). Also done with SUCCESS.

VlastiBroucek, 2024-08-22:
> It's just a warning. It will still work and the warning will get fixed at some point.

Yes, it will (most probably, but not guaranteed) work, however casting errors/warnings are often exploited. So for security reasons, it would be good to have it fixed ASAP.

poundy, 2024-08-22:
For me, this error message is coming up any time I now attempt to open the logs of an updated ESPHome device, esp32 or esp8266 based does not make a difference.

Wunja, 2024-08-22:
I've experienced this problem and it did break the application - the device became unresponsive.
Sorry, I didn't capture the decode section before rolling back.

garry0garry, 2024-08-22:
https://github.com/esphome/issues/issues/6191

moravszky, 2024-08-22:
I get the same warning,
Core: 2024.8.2
HAOS: 13.1
ESPHome: 2024.8.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended

clydebarrow, 2024-08-23:
> So for security reasons, it would be good to have it fixed ASAP.

You can be assured there is no security risk - the warning is spurious, in the sense that the code is actually perfectly fine, just the linter doesn't have enough context to determine that. And that piece of code isn't at this time actually used anywhere. But it will be fixed for all the obvious reasons.

clydebarrow, 2024-08-23:
Fixed with https://github.com/esphome/issues/issues/6179

clydebarrow, 2024-08-23:
Fixed with https://github.com/esphome/esphome/pull/7316

VlastiBroucek, 2024-08-28:
This is still present in 2024.8.1

ssieb, 2024-08-28:
It's only in dev, so will be in the next full release.  I don't know if it will get pulled into a point release before that.

serlinGi, 2024-08-28:
I just updated esphome to 2024.8.1 full and this message appears when compiling/updating all my devices

ssieb, 2024-08-28:
Did you not read the two comments right before yours?

serlinGi, 2024-08-28:
> Did you not read the two comments right before yours?

sorry, i didn't think the confirmation flag would be a problem for you. I'll delete my comment right away, right now 👍

sajikur, 2024-08-28:
I've the same warning in my esp8266 gas sensor. Everything works correctly but warning is still there
Could you explain me the meaning?
thanx

ssieb, 2024-08-28:
The compiler doesn't like casting a float pointer to an int pointer like that.

sajikur, 2024-08-28:
> casting a float pointer to an int pointer

the compiler might think about doing the compiler's job without making judgments or having preferences ;)

SaszkaX, 2024-08-28:
ESPHome update 2024.8.0 -> 2024.8.1 fixed all warnings.

SaszkaX, 2024-08-28:
> Same here. After update Core 2024.8.1 -> 2024.8.2 and HAOS 13.0 -> 13.1 (but there was no ESPHome update - it's still 2024.8.0), after reboot there was new update of all ESP8266 devices (GoSund P1 & GoSund SP111), same warnings. Also update done with SUCCESS.

ESPHome update 2024.8.0 -> 2024.8.1 fixed all warnings.

garry0garry, 2024-09-03:
ESPHome 2024.8.3
ESP8266
```
Compiling .pioenvs/pzem1/src/esphome/core/bytebuffer.cpp.o
src/esphome/core/bytebuffer.cpp: In member function 'float esphome::ByteBuffer::get_float()':
src/esphome/core/bytebuffer.cpp:86:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   86 |   return *(float *) &value;
      |           ^~~~~~~~~~~~~~~~
src/esphome/core/bytebuffer.cpp: In member function 'void esphome::ByteBuffer::put_float(float)':
src/esphome/core/bytebuffer.cpp:129:61: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  129 | void ByteBuffer::put_float(float value) { this->put_uint32(*(uint32_t *) &value); }
      |                                                             ^~~~~~~~~~~~~~~~~~~
```

ssieb, 2024-09-03:
It will be fixed in the next major release.  There's no need to keep commenting about it.
I'm only leaving this open so that people can find it and hopefully not create new issues.

mawildcat, 2024-09-16:
I have the same problem with:
Core 2024.9.1
Supervisor 2024.09.1
Operating System 13.1
ESP12_Relay_x4