esphome/issues #6183: Cannot add the new update feature in the config

geotibi, 2024-08-21:
### The problem

I want to add the update new feature in the code but seems that is not working like I did below

update:
  - platform: esphome
    name: Update

It seems that the platform esphome is not the correct one. What I should use instead?

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.08.0

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

update

### Example YAML snippet

```yaml
update:
  - platform: esphome
    name: Update
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.8.0
INFO Reading configuration /config/esphome/centrala.yaml...
INFO Unable to import component esphome.update: No module named 'esphome.components.esphome.update'
Failed config

update.esphome: [source /config/esphome/centrala.yaml:147]
  
  Platform not found: 'update.esphome'.
  platform: esphome
  name: Update
```

randybb, 2024-08-21:
There is no such component, currently only https://esphome.io/components/update/http_request If it is your custom one, did you included it?

geotibi, 2024-08-21:
I'm trying to use the below component

https://esphome.io/components/update/

randybb, 2024-08-21:
This is just a core component, only platform which is possible for now is the mentioned http_request.

geotibi, 2024-08-22:
I succeed to do the configuration, the only issue that I have now is the source URL

update:
  - platform: http_request
    name: "Firmware Update"
    source: https://........
    
where I should have the path to the bin file but I dont know where these are stored in home assistant. I found only the yaml files in esphome folder under config one, but this one does not have the bin files

randybb, 2024-08-22:
You need to generate all files for every device and share them somehow. Normally this is done by a workflow, for example on github like is done for https://github.com/esphome/firmware - this feature is for device creators. Since there is no issue with the component itself and issues are not for helping with understanding of documentation I am closing this.

closed 2024-08-22 by randybb