esphome/issues #6187: Boot loop with no error after upgrade to 2024.8.0

danastasia, 2024-08-21:
### The problem

The device is stuck in a boot loop without loading the app or an apparent error.

[logs_office-door-display_run(1).txt](https://github.com/user-attachments/files/16697748/logs_office-door-display_run.1.txt)
[office-door-display.yaml.txt](https://github.com/user-attachments/files/16697806/office-door-display.yaml.txt)
[waveshare-43.yaml.txt](https://github.com/user-attachments/files/16697807/waveshare-43.yaml.txt)

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.2

### What platform are you using?

ESP32-IDF

### Board

waveshare esp32-s3-touch-lcd-4.3

### Component causing the issue

unknown

marsp88, 2024-08-21:
Same problem here for a similar configuration. I have a  ESP32-S3-4848S040 that is stuck in a boot loop.

dmptrluke, 2024-08-23:
Same thing here. I'm trying to set up and M5Stack Atom S3 and if I use the ESP-IDF platform, it just bootloops and wont start. No issue with Arduino.

clydebarrow, 2024-08-23:
Immediate fix - add this to yaml

```
external_components:
  - source: github://pr#7345
    components: [lvgl]
```

clydebarrow, 2024-08-23:
> I'm trying to set up and M5Stack Atom S3 and if I use the ESP-IDF platform, it just bootloops and wont start. No issue with Arduino.

Sounds like a different problem, this bug is related to LVGL and has nothing to do with ESP-IDF. Probably you need:
```
esphome:
  platformio_options:
    board_build.flash_mode: dio
```

danastasia, 2024-08-23:
> external_components:
>   - source: github://pr#7345
>     components: [lvgl]

Thanks this worked perfectly!

closed 2024-08-23 by danastasia