esphome/issues #6188: Component m5stack_8angle is marked FAILED

carrierone, 2024-08-21:
### The problem

**Component m5stack_8angle is marked FAILED** on startup.  This is a new module so not sure if its a bug or I configured it wrong.

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.2

### What platform are you using?

ESP32-IDF

### Board

m5stack Atom Echo with m5stack 8Angle

### Component causing the issue

components/m5stack_8angle

### Example YAML snippet

```yaml
esphome configuration:

substitutions:
  name: m5stack-atom-echo-xxxxxx
  friendly_name: M5Stack Atom Echo xxxxxx office

packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.adopted.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .lan
i2c:
  sda: 26
  scl: 32
  id: bus_a

m5stack_8angle:
    id: m5stack_8angle_base
    i2c_id: bus_a
    address: 0x43

sensor:
  - platform: m5stack_8angle
    m5stack_8angle_id: m5stack_8angle_base
    channel: 1
    name: "Knob 1"
    bit_depth: 12 bit
  - platform: m5stack_8angle
    m5stack_8angle_id: m5stack_8angle_base
    channel: 2
    name: "Knob 2"

binary_sensor:
  - platform: m5stack_8angle
    m5stack_8angle_id: m5stack_8angle_base
    name: "Switch"

light:
  - platform: m5stack_8angle
    m5stack_8angle_id: m5stack_8angle_base
    id: m8_angle_leds
    name: "Lights"
    effects:
        - addressable_lambda:
            name: "Indicate Values"
            update_interval: 200ms
            lambda: 
                    ESPHSVColor hsv;
                    hsv.value = 255;
                    hsv.saturation = 240;
                    auto parent = id(m5stack_8angle_base);
                    for (int i=0; i < 8; i++) {
                      auto kpos = parent->read_knob_pos(i);
                      if (kpos >= 0){
                        hsv.hue = kpos * 200;
                        it[i] = hsv;
                      }
                    }
                    if (parent->read_switch() > 0)
                        hsv.hue = 200;
                    else
                        hsv.hue = 0;
                    it[8] = hsv;
```

### Anything in the logs that might be useful for us?

```txt
[18:00:06][I][app:100]: ESPHome version 2024.8.0 compiled on Aug 21 2024, 17:59:31
[18:00:06][C][wifi:600]: WiFi:
[18:00:06][C][wifi:428]:   Local MAC: 10:06:xxxxx
[18:00:06][C][wifi:433]:   SSID: [redacted]
[18:00:06][C][wifi:436]:   IP Address: 192.168.70.8
[18:00:06][C][wifi:440]:   BSSID: [redacted]
[18:00:06][C][wifi:441]:   Hostname: 'm5stack-atom-echo-xxxxxx'
[18:00:06][C][wifi:443]:   Signal strength: -36 dB ▂▄▆█
[18:00:06][C][wifi:447]:   Channel: 11
[18:00:06][C][wifi:448]:   Subnet: 255.255.255.0
[18:00:06][C][wifi:449]:   Gateway: 192.168.70.1
[18:00:06][C][wifi:450]:   DNS1: 192.168.4.4
[18:00:06][C][wifi:451]:   DNS2: 0.0.0.0
[18:00:06][C][logger:185]: Logger:
[18:00:06][C][logger:186]:   Level: DEBUG
[18:00:06][C][logger:188]:   Log Baud Rate: 115200
[18:00:06][C][logger:189]:   Hardware UART: UART0
[18:00:06][C][i2c.idf:075]: I2C Bus:
[18:00:06][C][i2c.idf:076]:   SDA Pin: GPIO26
[18:00:06][C][i2c.idf:077]:   SCL Pin: GPIO32
[18:00:07][C][i2c.idf:078]:   Frequency: 50000 Hz
[18:00:07][C][i2c.idf:084]:   Recovery: bus successfully recovered
[18:00:07][I][i2c.idf:094]: Results from i2c bus scan:
[18:00:07][I][i2c.idf:100]: Found i2c device at address 0x41
[18:00:07][C][esp32_rmt_led_strip:175]: ESP32 RMT LED Strip:
[18:00:07][C][esp32_rmt_led_strip:176]:   Pin: 27
[18:00:07][C][esp32_rmt_led_strip:177]:   Channel: 0
[18:00:07][C][esp32_rmt_led_strip:202]:   RGB Order: GRB
[18:00:07][C][esp32_rmt_led_strip:203]:   Max refresh rate: 0
[18:00:07][C][esp32_rmt_led_strip:204]:   Number of LEDs: 1
[18:00:07][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Button'
[18:00:07][C][gpio.binary_sensor:016]:   Pin: GPIO39
[18:00:07][C][light:103]: Light 'M5Stack Atom Echo 168750 office'
[18:00:07][C][light:105]:   Default Transition Length: 0.0s
[18:00:07][C][light:106]:   Gamma Correct: 2.80
[18:00:07][C][light:103]: Light 'Lights'
[18:00:07][C][light:105]:   Default Transition Length: 1.0s
[18:00:07][C][light:106]:   Gamma Correct: 2.80
[18:00:07][C][template.switch:068]: Template Switch 'Use wake word'
[18:00:07][C][template.switch:091]:   Restore Mode: restore defaults to ON
[18:00:07][C][template.switch:057]:   Optimistic: YES
[18:00:07][C][template.switch:068]: Template Switch 'Use listen light'
[18:00:07][C][template.switch:091]:   Restore Mode: restore defaults to ON
[18:00:07][C][template.switch:057]:   Optimistic: YES
[18:00:07][C][template.switch:068]: Template Switch 'timer_ringing'
[18:00:07][C][template.switch:091]:   Restore Mode: always OFF
[18:00:07][C][template.switch:057]:   Optimistic: YES
[18:00:07][D][voice_assistant:504]: State changed from IDLE to START_MICROPHONE
[18:00:07][D][voice_assistant:510]: Desired state set to WAIT_FOR_VAD
[18:00:07][D][light:036]: 'M5Stack Atom Echo xxxxxx office' Setting:
[18:00:07][D][light:051]:   Brightness: 60%
[18:00:07][D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[18:00:07][D][voice_assistant:221]: Starting Microphone
[18:00:07][D][ring_buffer:024]: Created ring buffer with size 16384
[18:00:07][D][voice_assistant:504]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[18:00:07][C][safe_mode.button:024]: Safe Mode Button 'Safe Mode Boot'
[18:00:07][C][safe_mode.button:024]:   Icon: 'mdi:restart-alert'
[18:00:07][D][esp-idf:000]: I (8193) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[18:00:07][D][voice_assistant:504]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
[18:00:07][C][factory_reset.button:011]: Factory Reset Button 'Factory reset'
[18:00:07][C][factory_reset.button:011]:   Icon: 'mdi:restart-alert'
[18:00:07][D][voice_assistant:245]: Waiting for speech...
[18:00:07][D][voice_assistant:504]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
[18:00:07][C][m5stack_8angle:030]: M5STACK_8ANGLE:
[18:00:07][C][m5stack_8angle:031]:   Address: 0x43
[18:00:07][C][m5stack_8angle:032]:   Firmware version: 0 
[18:00:07][E][component:082]:   Component m5stack_8angle is marked FAILED
[18:00:07][D][voice_assistant:258]: VAD detected speech
[18:00:07][D][voice_assistant:504]: State changed from WAITING_FOR_VAD to START_PIPELINE
[18:00:07][D][voice_assistant:510]: Desired state set to STREAMING_MICROPHONE
[18:00:07][D][voice_assistant:275]: Requesting start...
[18:00:07][D][voice_assistant:504]: State changed from START_PIPELINE to STARTING_PIPELINE
[18:00:07][D][voice_assistant:525]: Client started, streaming microphone
[18:00:07][D][voice_assistant:504]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[18:00:07][D][voice_assistant:510]: Desired state set to STREAMING_MICROPHONE
[18:00:07][D][voice_assistant:627]: Event Type: 1
[18:00:07][D][voice_assistant:630]: Assist Pipeline running
```

randybb, 2024-08-22:
I2C scan says that there is a device with address 0x41. Have you changed its address from 0x43 to 0x41 as it can be changed via a register? Or is something else connected to this bus?

carrierone, 2024-08-22:
I thought maybe it was that but when I set the 8angle to 0x41 it crashed the esp. So I am pretty sure that 0x41 is the esp itself.  It does in fact see the 8angle on 0x43 in the log it just won't load the module --Sent with GMX Mail app

randybb, 2024-08-22:
Nope, there is nothing connected to the external connector. Only Atom S3 has mcp23017 (0x20) connected to 38/39.

carrierone, 2024-08-23:
So sorry!  I just realized I have the 8encoder not the 8angle, I didn't even know there was two different things that looked so similar!  Now it makes sense why the ID is different!

randybb, 2024-08-23:
yup, unfortunately they used totally different registers http://docs.m5stack.com/en/unit/8Angle / https://docs.m5stack.com/en/unit/UNIT%208EnCoder