esphome/issues #6191: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

bernardesarthur, 2024-08-22:
### The problem

I don't know what it is, but I've check my code and I don't find any problems with config.

The whole error message is below:

```
src/esphome/core/bytebuffer.cpp: In member function 'float esphome::ByteBuffer::get_float()':
src/esphome/core/bytebuffer.cpp:86:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   86 |   return *(float *) &value;
      |           ^~~~~~~~~~~~~~~~
src/esphome/core/bytebuffer.cpp: In member function 'void esphome::ByteBuffer::put_float(float)':
src/esphome/core/bytebuffer.cpp:129:61: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  129 | void ByteBuffer::put_float(float value) { this->put_uint32(*(uint32_t *) &value); }
      |                                                             ^~~~~~~~~~
```~~~~~~~~~
      
Until now I don't see any problem with my devices.

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.2

### What platform are you using?

ESP8266

### Board

Sonoff Basic

### Example YAML snippet

```yaml
esphome:
  name: sonoff-basic-nvr
  friendly_name: Sonoff Basic NVR

esp8266:
  board: esp01_1m
  restore_from_flash: True
  early_pin_init: False

#base_file_for_esphome_projects

preferences:
  flash_write_interval: 1min

logger:

api:

ota:
  - platform: esphome

mdns:
  disabled: False

wifi:
  ssid: Bernardes Casa
  password: qapsteve190
  ap: {}

captive_portal:

web_server:
  port: 80
  local: True
  auth:
    username: "admin"
    password: "<YIMI5x2"

time:
  - platform: sntp
    id: sntp_time
    timezone: America/Sao_Paulo
#    timezone: "GMT+3"
    update_interval:
      milliseconds: 0
      seconds: 60
      minutes: 0
      hours: 0
    servers:
     - a.st1.ntp.br
     - b.st1.ntp.br
     - c.st1.ntp.br

    on_time:
      - seconds: 0
        minutes: 0
        hours: 6
        days_of_week: 
          - 1  # Segunda-feira
          - 3  # Quarta-feira
          - 5  # Sexta-feira
        then:
          - switch.turn_off: tomada_do_nvr
          - delay:
              seconds: 0
              minutes: 1
              hours: 0  # Aguardar 1 minuto
          - switch.turn_on: tomada_do_nvr

button:
  - platform: factory_reset
    name: Reset
    id: Reset
    entity_category: config
    internal: False
    disabled_by_default: False

  - platform: safe_mode
    name: Safe Mode
    entity_category: config
    internal: False
    disabled_by_default: False
  
  - platform: restart
    name: Restart
    entity_category: config
    internal: False
    disabled_by_default: False

binary_sensor:
  - platform: status
    name: Status
    internal: False
    disabled_by_default: False

  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    id: button_1
    name: Button 1
    internal: False
    disabled_by_default: False
    publish_initial_state: True
    filters: 
      - delayed_on_off:
          milliseconds: 10
          seconds: 0
          minutes: 0
          hours: 0
    on_press: 
      then:
      -  switch.toggle: tomada_do_nvr

text_sensor:
  - platform: wifi_info
    ip_address: 
      name: IP Address
      entity_category: diagnostic
      internal: False
      disabled_by_default: False
    ssid: 
      name: SSID
      entity_category: diagnostic
      internal: False
      disabled_by_default: False
  - platform: version
    name: Esphome version
    internal: False
    disabled_by_default: False
    hide_timestamp: True

sensor:
  - platform: uptime
    name: Uptime Sensor
    id: uptime_sensor
    internal: False
    disabled_by_default: False
    update_interval:
      milliseconds: 0
      seconds: 60
      minutes: 0
      hours: 0
    entity_category: diagnostic
  
  - platform: wifi_signal
    name: WiFi Signal dBm
    id: wifi_signal_dbm
    internal: False
    disabled_by_default: False
    update_interval:
      milliseconds: 0
      seconds: 60
      minutes: 0
      hours: 0
    entity_category: diagnostic
    device_class: signal_strength

  - platform: copy
    name: WiFi Signal Percent
    id: wifi_signal_percent
    source_id: wifi_signal_dbm
    internal: False
    disabled_by_default: False
    unit_of_measurement: "%"
    entity_category: diagnostic
    filters: 
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);

switch:
  - platform: gpio
    pin:
      number: GPIO12
      inverted: False
    name: Tomada do NVR
    id: tomada_do_nvr
    internal: False
    disabled_by_default: False
    restore_mode: RESTORE_DEFAULT_OFF

status_led:
  pin: 
    number: GPIO13
    inverted: False
```

garry0garry, 2024-08-22:
ESPHome 2024.8.0

```
src/esphome/core/bytebuffer.cpp: In member function 'float esphome::ByteBuffer::get_float()':
src/esphome/core/bytebuffer.cpp:86:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   86 |   return *(float *) &value;
      |           ^~~~~~~~~~~~~~~~
src/esphome/core/bytebuffer.cpp: In member function 'void esphome::ByteBuffer::put_float(float)':
src/esphome/core/bytebuffer.cpp:129:61: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  129 | void ByteBuffer::put_float(float value) { this->put_uint32(*(uint32_t *) &value); }
      |                                                             ^~~~~~~~~~~~~~~~~~~
```

slongano, 2024-08-22:
This happens to me too. Snippet:
Compiling .pioenvs/interruttore-sala-piccola/src/esphome/core/bytebuffer.cpp.o
src/esphome/core/bytebuffer.cpp: In member function 'float esphome::ByteBuffer::get_float()':
src/esphome/core/bytebuffer.cpp:86:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   86 |   return *(float *) &value;
      |           ^~~~~~~~~~~~~~~~
src/esphome/core/bytebuffer.cpp: In member function 'void esphome::ByteBuffer::put_float(float)':
src/esphome/core/bytebuffer.cpp:129:61: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  129 | void ByteBuffer::put_float(float value) { this->put_uint32(*(uint32_t *) &value); }
      |                                                             ^~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/interruttore-sala-piccola/src/esphome/core/color.cpp.o

moravszky, 2024-08-22:
I get the same warning,
Core: 2024.8.2
HAOS: 13.1
ESPHome: 2024.8.0

esp32:
board: esp32dev
framework:
type: esp-idf
version: recommended

compile log:
src/esphome/core/bytebuffer.cpp: In member function 'float esphome::ByteBuffer::get_float()':
src/esphome/core/bytebuffer.cpp:86:11: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
return *(float ) &value;
^~~~~~~~~~~~~~~~
src/esphome/core/bytebuffer.cpp: In member function 'void esphome::ByteBuffer::put_float(float)':
src/esphome/core/bytebuffer.cpp:129:61: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
void ByteBuffer::put_float(float value) { this->put_uint32((uint32_t *) &value); }
^~~~~~~~~~~~~~~~~~~

jesserockz, 2024-08-22:
This is only a warning and is being fixed already.

jesserockz, 2024-08-22:
Duplicate of #6179

closed 2024-08-22 by jesserockz