esphome/issues #6193: ESP32 and W5500 - COMPLETE FAILURE OF COMPONENT

ESP32andmore, 2024-08-22:
### The problem

Using a ESP32 with the W5500 Ethernet chip, ESPHome fails to initialize device. The hardware has been verified to be fully functional with a Arduino sketch using WebServer_ESP32_W5500 lib @25mhz. 

### Which version of ESPHome has the issue?

2204.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Most recent

### What platform are you using?

ESP32

### Board

Coolboxx II, ESP-DEV-Module

### Component causing the issue

W5500

### Example YAML snippet

```yaml
substitutions:
  devicename: avfan1eth
  fdevicename: AVfan1eth
  ip: 192.168.0.201 # For static IP adressing

  ssid: !secret wifi_ssid
  password: !secret wifi_password

  alarm_temp: "50" # Threshold for alarm when implemented
  ptemp: "20"
  itemp: "0.1"

# Connect your Dallas sensors with logger: DEBUG enabled (factory flash default).
# Open the ESPHome webtool, connect to the ECB board, view the "Logs" window, hit "Reset",
# and search for Dallas scan addresses and replace the below addresses with your unique ones.
# Example Scan
#[18:16:42][D][dallas.sensor:084]:     0x8b3ca8f64935b228 4
#[18:16:42][D][dallas.sensor:084]:     0x783c0af6490d3628 2
#[18:16:42][D][dallas.sensor:084]:     0xc13c01f09642c928 1
#[18:16:42][D][dallas.sensor:084]:     0x043c6ef649772f28 3
  dallasaddress1: "0xc13c01f09642c928"
  dallasaddress2: "0x783c0af6490d3628"
  dallasaddress3: "0x043c6ef649772f28"
  dallasaddress4: "0x8b3ca8f64935b228"

#external_components:
#  - source:
#      type: git
#      url: https://github.com/ssieb/esphome
#      ref: onewire
#    components: [ one_wire, gpio ]
#    refresh: 1min

esphome:
  name: ${devicename}
  friendly_name: ${fdevicename}

  on_boot:
  - priority: 900.0
[... 254 lines cut ...]
          client.print("</html>\r\n");
          break;
        }

        if (c == '\n')
        {
          // you're starting a new line
          currentLineIsBlank = true;
        }
        else if (c != '\r')
        {
          // you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }

    // give the web browser time to receive the data
    delay(10);

    // close the connection:
    client.stop();
    Serial.println(F("Client disconnected"));
  }
}

functionpointer, 2024-09-17:
Same issue here
```
[01:03:38]Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception).
[01:03:38]Debug exception reason: Stack canary watchpoint triggered (w5500_tsk)
[01:03:38]Core  0 register dump:
[01:03:38]PC      : 0x400928a7  PS      : 0x00060836  A0      : 0x4008b30e  A1      : 0x3ffae820
WARNING Decoded 0x400928a7: xthal_save_extra_nw at /Users/igrokhotkov/e/esp32/hal/hal/state_asm.S:72
[01:03:38]A2      : 0x3ffae890  A3      : 0xb33fffff  A4      : 0x3ffaeaf0  A5      : 0x00000000
[01:03:38]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffae8f0
[01:03:38]A10     : 0x3ffae8f0  A11     : 0x3f401e42  A12     : 0x000000ff  A13     : 0x0000ff00
[01:03:38]A14     : 0x00000007  A15     : 0xff000000  SAR     : 0x00000008  EXCCAUSE: 0x00000001
[01:03:38]EXCVADDR: 0x00000000  LBEG    : 0x40089029  LEND    : 0x40089039  LCOUNT  : 0xfffffffe
[01:03:38]
[01:03:38]
[01:03:38]Backtrace: 0x400928a4:0x3ffae820 0x4008b30b:0x3ffaebf0 0x401534da:0x3ffaec80 0x400d6a85:0x3ffaecc0 0x400d6ab9:0x3ffaecf0 0x400d6b15:0x3ffaed40 0x400d6c3d:0x3ffaed70 0x400de20e:0x3ffaed90 0x4015f881:0x3ffaedd0 0x4011a85a:0x3ffaee00 0x4011a8f7:0x3ffaee30 0x400e70ba:0x3ffaeeb0 0x400e7afa:0x3ffaeee0 0x400e7c54:0x3ffaef10 0x400f5f78:0x3ffaef30 0x400f602e:0x3ffaef90 0x400f6d4b:0x3ffaefc0 0x400f61f6:0x3ffaf000
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400928a4: xthal_save_extra_nw at /Users/igrokhotkov/e/esp32/hal/hal/state_asm.S:72
WARNING Decoded 0x4008b30b: _xt_context_save at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/xtensa_context.S:159
WARNING Decoded 0x401534da: vsnprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:40
WARNING Decoded 0x400d6a85: esphome::logger::Logger::vprintf_to_buffer_(char const*, __va_list_tag) at C:\Users\Fubbel\git\gas-meter\.esphome\build\elec-gas-meter/src/esphome/components/logger/logger.h:124
WARNING Decoded 0x400d6ab9: esphome::logger::Logger::printf_to_buffer_(char const*, ...) at C:\Users\Fubbel\git\gas-meter\.esphome\build\elec-gas-meter/src/esphome/components/logger/logger.h:138
WARNING Decoded 0x400d6b15: esphome::logger::Logger::write_header_(int, char const*, int) at C:\Users\Fubbel\git\gas-meter\.esphome\build\elec-gas-meter/src/esphome/components/logger/logger.cpp:56
WARNING Decoded 0x400d6c3d: esphome::logger::Logger::log_vprintf_(int, char const*, int, char const*, __va_list_tag) at C:\Users\Fubbel\git\gas-meter\.esphome\build\elec-gas-meter/src/esphome/components/logger/logger.cpp:67
WARNING Decoded 0x400de20e: esphome::esp_idf_log_vprintf_(char const*, __va_list_tag) at C:\Users\Fubbel\git\gas-meter\.esphome\build\elec-gas-meter/src/esphome/core/log.cpp:56
WARNING Decoded 0x4015f881: esp_log_writev at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/log/log.c:200
WARNING Decoded 0x4011a85a: __wrap_esp_log_writev at /Users/ficeto/Desktop/ESP32/ESP32S2/esp32-arduino-lib-builder/components/esp-insights/components/esp_diagnostics/src/esp_diagnostics_log_hook.c:452
WARNING Decoded 0x4011a8f7: __wrap_esp_log_write at /Users/ficeto/Desktop/ESP32/ESP32S2/esp32-arduino-lib-builder/components/esp-insights/components/esp_diagnostics/src/esp_diagnostics_log_hook.c:469
WARNING Decoded 0x400e70ba: check_trans_valid at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/spi_master.c:716 (discriminator 7)
WARNING Decoded 0x400e7afa: spi_device_polling_start at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/spi_master.c:931 (discriminator 2)
WARNING Decoded 0x400e7c54: spi_device_polling_transmit at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/spi_master.c:1000
WARNING Decoded 0x400f5f78: w5500_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_eth/src/esp_eth_mac_w5500.c:96
WARNING Decoded 0x400f602e: w5500_read_buffer at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_eth/src/esp_eth_mac_w5500.c:192
WARNING Decoded 0x400f6d4b: emac_w5500_receive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_eth/src/esp_eth_mac_w5500.c:572 (discriminator 2)
WARNING Decoded 0x400f61f6: emac_w5500_task at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_eth/src/esp_eth_mac_w5500.c:339
[01:03:38]
[01:03:38]
[01:03:38]
[01:03:38]
[01:03:38]ELF file SHA256: 97715b38086b31e7
[01:03:38]
[01:03:38]E (3802) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
[01:03:38]Rebooting...
[01:03:38]ets Jun  8 2016 00:22:57
[01:03:38]
[01:03:38]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[01:03:38]configsip: 0, SPIWP:0xee
[01:03:38]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[01:03:38]mode:DIO, clock div:2
[01:03:38]load:0x3fff0030,len:1184
[01:03:38]load:0x40078000,len:13192
[01:03:38]load:0x40080400,len:3028
[01:03:38]entry 0x400805e4
[01:03:38]E (361) esp_core_dump_flash: No core d\xf5f\x9e\x81partition found!
[01:03:38]E (361) esp_core_dump_flash: No core dump partition found!
[01:03:38][I][logger:156]: Log initialized
[01:03:38][C][safe_mode:079]: There have been 5 suspected unsuccessful boot attempts
[01:03:38][D][esp32.preferences:114]: Saving 1 preferences to flash...
[01:03:38][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[01:03:38][I][app:029]: Running through setup()...
[01:03:38][I][i2c.arduino:218]: Performing I2C bus recovery
[01:03:38][C][ethernet:041]: Setting up Ethernet...
```

functionpointer, 2024-09-24:
Turns out the SPI frequency was too fast for my wiring.
Fixed by reducing it:
```yaml
ethernet:
  type: W5500
  clk_pin: 19
  mosi_pin: 26
  miso_pin: 23
  cs_pin: 27
  interrupt_pin: 22
  reset_pin: 18
  clock_speed: 8MHz
  manual_ip:
    static_ip: 10.10.11.94
    gateway: 10.10.11.1
    subnet: 255.255.255.0
```

closed 2025-02-11 by github-actions[bot]