esphome/issues #6198: Template / config loses value

yanivf38, 2024-08-23:
### The problem

I have 3 template/config numeric values in my yaml. That randomly get reset to their initial value. The reason I say randomly is because if manually reboot my node or ha the value remains but then randomly it gets reset. What is strange to me is that when the values get reset all 3 template/configs get the initial value 

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.0

### What platform are you using?

ESP8266

### Board

d1-mini

### Component causing the issue

template

### Example YAML snippet

```yaml
snippet:
--------
  - platform: template
    name: "Radar End Zone 2"
    id: radar_z2_end
    device_class: distance
    min_value: 0
    max_value: 600
    step: 1
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: 400
    icon: "mdi:arrow-collapse-right"
    entity_category: CONFIG

full package:
------------------
esphome:
  name: ${device_name}
  friendly_name: ${device_name}

esp8266:
  board: d1_mini

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: ${api_encryption_key}

ota:
  - platform: esphome
    password: ${ota_pwd}

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    password: ${ap_wifi_pwd}

captive_portal:

globals:
  - id: last_illuminance
    type: float
    restore_value: no
    initial_value: "-1"

i2c:
  sda: D2
  scl: D1
  id: bus_a

ld2410:
  uart_id: ld2410_uart
  throttle: 1500ms
  id: ld2410_comp

uart:
  id: ld2410_uart
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

binary_sensor:

  - platform: ld2410
    has_target:
      name: Presence
      id: radar_has_target
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
      
#  - platform: gpio
#    pin: D4
#    name: gpio out pin presence
#    device_class: presence

  ## Set Up Radar Zones Based On Distance
  - platform: template
    name: "Radar Zone 1 Occupancy"
    device_class: occupancy
    icon: mdi:motion-sensor
    lambda: |-
      if ((id(radar_has_target).state) && (id(radar_detection_distance).state < id(radar_z1_end).state)) {
        return true;
      } else {
        return false;
      }
  - platform: template
    name: "Radar Zone 2 Occupancy"
    device_class: occupancy
    icon: mdi:motion-sensor
    lambda: |-
      if ((id(radar_has_target).state) && ((id(radar_z1_end).state < id(radar_detection_distance).state) && (id(radar_detection_distance).state < id(radar_z2_end).state))) {
        return true;
      } else {
        return false;
      }
  - platform: template
    name: "Radar Zone 3 Occupancy"
    device_class: occupancy
    icon: mdi:motion-sensor
    lambda: |-
      if ((id(radar_has_target).state) && ((id(radar_z2_end).state < id(radar_detection_distance).state) && (id(radar_detection_distance).state < id(radar_z3_end).state))) {
        return true;
      } else {
        return false;
      }

sensor:
  - platform: bh1750
    name: "Light sensor"
    address: 0x23
    accuracy_decimals: 1
    id: bh1750_light
    update_interval: 1s
    filters:
      - lambda: !lambda |-
          if (id(last_illuminance) == x){
            return {};
          }
          if (abs(id(last_illuminance) - x) > 1){

              id(last_illuminance) = x;
              return x;
          }
          return {};  

  - platform: ld2410
    light:
      name: light
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance
      id: radar_detection_distance
    g0:
      move_energy:
        name: g0 move energy
      still_energy:
        name: g0 still energy
    g1:
      move_energy:
        name: g1 move energy
      still_energy:
        name: g1 still energy
    g2:
      move_energy:
        name: g2 move energy
      still_energy:
        name: g2 still energy
    g3:
      move_energy:
        name: g3 move energy
      still_energy:
        name: g3 still energy
    g4:
      move_energy:
        name: g4 move energy
      still_energy:
        name: g4 still energy
    g5:
      move_energy:
        name: g5 move energy
      still_energy:
        name: g5 still energy
    g6:
      move_energy:
        name: g6 move energy
      still_energy:
        name: g6 still energy
    g7:
      move_energy:
        name: g7 move energy
      still_energy:
        name: g7 still energy
    g8:
      move_energy:
        name: g8 move energy
      still_energy:
        name: g8 still energy

switch:
  - platform: ld2410
    engineering_mode:
      name: "engineering mode"
    bluetooth:
      name: "control bluetooth"

number:
  - platform: ld2410
    timeout:
      name: timeout
    light_threshold:
      name: light threshold
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold
        
  - platform: template
    name: "Radar End Zone 1"
    id: radar_z1_end
    device_class: distance
    min_value: 0
    max_value: 600
    step: 1
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: 200
    icon: "mdi:arrow-collapse-right"
    entity_category: CONFIG

  - platform: template
    name: "Radar End Zone 2"
    id: radar_z2_end
    device_class: distance
    min_value: 0
    max_value: 600
    step: 1
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: 400
    icon: "mdi:arrow-collapse-right"
    entity_category: CONFIG

  - platform: template
    name: "Radar End Zone 3"
    id: radar_z3_end
    device_class: distance
    min_value: 0
    max_value: 600
    step: 1
    update_interval: never
    optimistic: true
    restore_value: true
    initial_value: 600
    icon: "mdi:arrow-collapse-right"
    entity_category: CONFIG
    
button:
  - platform: ld2410
    factory_reset:
      name: "factory reset"
    restart:
      name: "restart"
    query_params:
      name: query params

text_sensor:
  - platform: ld2410
    version:
      name: "firmware version"
    mac_address:
      name: "mac address"

select:
  - platform: ld2410
    distance_resolution:
      name: "distance resolution"
    baud_rate:
      name: "baud rate"
    light_function:
      name: light function
#    out_pin_level:
#      name: out pin level
```

ssieb, 2024-08-24:
Could you be triggering the factory reset button somehow?

yanivf38, 2024-08-24:
Thank you @ssieb for the response. No, as the gates values are not reset.

yanivf38, 2024-08-24:
I looked at the history and it looks like the values get reset when the node becomes back online from unavailable. It's unclear to me why that wouldn't be the case if I turn the node off and back on.

ssieb, 2024-08-24:
As far as I can tell, the ld2410 values aren't saved on the ESP, so they wouldn't be affected by a factory reset.

yanivf38, 2024-08-25:
I have added the below line to the esp8266 section that might have fixed it. Need to wait to wait few days to confirm.

restore_from_flash : True

ssieb, 2024-08-25:
Yes, that is necessary.  I think without that, it will keep the values over a reboot, but not a power cycle.

closed 2025-01-07 by github-actions[bot]