esphome/issues #6202: Unable to connect to (ASUS Zen)WiFi 7 mesh

astrostl, 2024-08-24:
### The problem

I have 3x ASUS ZenWifi BE30000 WiFi 7 APs in a mesh configuration, one with an ethernet backhaul and the other with an MLO WiFi backhaul. I was unable to connect my Apollo Automation MSR-2 or MTR-1 to WiFi, and noticed that it presented a separate duplicate SSID for each AP. This persisted after resetting and reflashing them to default firmware.

To try to troubleshoot I created a guest network that only used 2.4GHz and WPA1, which still failed in mesh mode. Once I took that down to one-SSID-per-AP it finally worked, and kept working even when I upped the security to WPA2/WPA3-Personal. I'm not sure if it's "ASUS AiMesh" in particular or WiFi 7 mesh in general, but I had consistent connectivity blocks until I went with a one-SSID-per-AP model.

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.8.2

### What platform are you using?

ESP32

### Board

Apollo Automation MSR-2 and MTR-1

### Component causing the issue

WiFi module?

johansundstrom, 2025-01-03:
I have almost the same problem. I have 2 x ASUS routers configured in AI-mesh. There is definetly problems for ESP devices to connect to the network.

astrostl, 2025-01-03:
Did the workaround I use/d also work for you @johansundstrom ?