esphome/issues #6203: REST API 401 Unauthorized

MorganMLGman, 2024-08-24:
### The problem

I am trying to access REST API, but it returns 401 Unauthorized, despite that I can access webpage without any problems with the same user and password.

[image]
[image]

### Which version of ESPHome has the issue?

2024.8.0, 2024.7.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP8266

### Board

esp8285, esp32-c3-devkitm-1, esp32-s3-devkitc-1, wt32-eth01

### Component causing the issue

api/rest api

### Example YAML snippet

```yaml
web_server:
  port: 80
  local: true
  ota: true
  include_internal: True
  version: 2
  auth: 
    username: !secret web_user
    password: !secret web_password

---

web_server:
  port: 80
  local: true
  ota: false
  include_internal: True
  version: 3
  auth: 
    username: !secret web_user
    password: !secret web_password
```

### Anything in the logs that might be useful for us?

```txt
No
```

### Additional information

No

cichy1173, 2024-08-24:
I have the same problem with Gosund SP111 Smart Plug with the newest ESPHome `2024.7.3 Aug 24 2024, 15:16:30`. On older softer it worked as intended.

MorganMLGman, 2024-08-24:
I found one device esp32 s3 super mini from aliexpress that I can use api for toggling lights, but even stranger is that I can use both GET and POST to toggle lights.

web_server config from this device.
`
web_server:
  port: 80
  local: true
  ota: false
  include_internal: True
  version: 3
  auth: 
    username: !secret web_user
    password: !secret web_password
`

[image]
[image]

MorganMLGman, 2024-08-24:
Also the device that works uses the same user and password that devices that don't work.

closed 2025-01-06 by github-actions[bot]