esphome/issues #6204: LVGL idle condition issue

plplaaa2, 2024-08-25:
### The problem

I made automation in idle state according to the LVGL example, but 'mapping values are not allowed here' error occurs.

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8

### What platform are you using?

ESP32

### Board

ESP32 S3

### Component causing the issue

LVGL

### Example YAML snippet

```yaml
time:
  - platform: homeassistant
    id: time_comp
    on_time_sync:
      - script.execute: time_update
      - script.execute: date_update
    on_time:
      - minutes: '*'
        seconds: '0'
        then:
          - script.execute: time_update
          - script.execute: date_update
      - minutes: '/5'
        then:
          - if:
              condition: lvgl.is_idle
                timeout: 5min    #line 114
              then:
                - lvgl.page.show: clock_page
                - delay: 10s
                - lvgl.page.show: home_page
```

### Anything in the logs that might be useful for us?

```txt
mapping values are not allowed here
  in "/config/esphome/dash-board-test.yaml", line 114, column 24

The same results are obtained by using an example from espome.

# In some trigger:
on_...:
  then:
    - if:
        condition: lvgl.is_idle
          timeout: 5s
        then:
          - light.turn_off:
              id: display_backlight
              transition_length: 3s
```

clydebarrow, 2024-08-25:
```
            condition:
              lvgl.is_idle:
                timeout: 5s
```

clydebarrow, 2024-08-25:
Fixed with https://github.com/esphome/esphome-docs/pull/4192

closed 2024-08-25 by clydebarrow