esphome/issues #6205: 'class esphome::http_request::HttpRequestUpdate' has no member named 'get_update_available_trigger'

parkghost, 2024-08-25:
### The problem

I want esphome to automatically update the firmware when it is available, but there are compilation errors when using on_update_available

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

ESP32-C3 Mini Development Board made by Waveshare

### Component causing the issue

update

### Example YAML snippet

```yaml
update:
  - platform: http_request
    id: update_manager
    name: Firmware - Status
    source: https:/.../manifest.json
    on_update_available:
      - update.perform: update_manager
```

### Anything in the logs that might be useful for us?

```txt
esphome compile weather-display-wareshare.yaml
INFO ESPHome 2024.8.0
INFO Reading configuration weather-display-wareshare.yaml...
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing weather-display-waveshare (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing SPI
INFO Installing SPI
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- UrlEncode @ 1.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- NeoPixelBus @ 2.7.3
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- qr-code-generator-library @ 1.7.0
Compiling .pioenvs/weather-display-waveshare/src/main.cpp.o
../templates/device_standalone.yaml: In function 'void setup()':
../templates/device_standalone.yaml:686:81: error: 'class esphome::http_request::HttpRequestUpdate' has no member named 'get_update_available_trigger'; did you mean 'get_update_interval'?
*** [.pioenvs/weather-display-waveshare/src/main.cpp.o] Error 1
=============================================================== [FAILED] Took 28.14 seconds ===============================================================
```

alex-tomin, 2024-10-20:
@jesserockz seems like you are the owner of this functionality. 
I have exact same issue on the 2024.10.0 version. Any suggestions how to overcome the error?

0009281, 2024-10-30:
The same issue with the 2024.10.2 version.

alex-tomin, 2024-11-15:
Any thoughts ho to workaround this issue?

daradib, 2025-05-20:
Here is a workaround using the [interval component](https://esphome.io/components/interval.html) to check if updates are available (if you have more than one update component, you can pass the `id` to `update.is_available` and `update.perform`):

```
interval:
  - interval: 1min
    then:
      - if:
          condition:
            update.is_available
          then:
            - update.perform
```

jesserockz, 2025-05-22:
The actual issue here is that I created `on_update_available` in config, but never implemented it in c++ 🙈

jhbruhn, 2025-06-24:
I have implemented the missing trigger, you can try it out by including the PR:

```
external_components:
  - source: github://pr#9174
    components: [update, http_request]
```

closed 2025-07-07 by jesserockz