esphome/issues #6206: Unable to update ESP32 in Safe Mode via OTA with new OTA/Safe Mode segregatiion

yokokito, 2024-08-26:
### The problem

I have updated my ESPHOME implementation in a NSPANEL to the latest version 2024.8.0. I changed the YAML file to update the OTA and SAFE_MODE sections with the new approach. No more changes. It compiled perfectly and I uploaded the bin file via OTA without problem.

I try the same now, and it is not possible to update it via OTA in SafeMode.

 ERROR Connecting to 192.168.0.131:3232 failed: [WinError 10061] 

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

NS-PANEL (SONOFF)

### Example YAML snippet

```yaml
safe_mode:  
   num_attempts: 3

ota:
 - platform: esphome   
   on_begin:
     then:
      - lambda: 'id($NspanelID).goto_page("12");'   
      - lambda: |-
             id(persiana).set_update_interval(0);
             id(persiana).call_setup();
             id(termostato).set_update_interval(0);
             id(termostato).call_setup();       
   on_progress:
     then:
      - lambda: id($NspanelID).send_command_printf("firmware.progress.val=%.0f", x); 
      - lambda: id($NspanelID).set_component_text_printf("firmware.t0", "%.0f", x);       

 
 
   on_error:
     then:
       - logger.log:
           format: "OTA update error %d"
           args: ["x"]
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.8.0
INFO Reading configuration nspanel1_rec.yaml...
INFO Connecting to 192.168.0.131
ERROR Connecting to 192.168.0.131:3232 failed: [WinError 10061] No se puede establecer una conexión ya que el equipo de destino denegó expresamente dicha conexión
```

### Additional information

My previous ESPHOME version  before the upgrade to 2024.8.0 was from June (I do not remember the version).

I try to update via OTA in a Windows Terminal --> esphome run nspanel_2.yaml --device 192.168.0.131

yokokito, 2024-08-29:
In safe mode the port 3232 is closed, where in normal mode is open. I assume that this is the reason. Can somebody check if this error has been introduced recently?  Thx

ssieb, 2024-08-29:
Are you sure it's in safe mode?

yokokito, 2024-08-30:
Fully sure (as far as I can).

I've implemented a button to restart en safe mode from HA that works pretty well since I also have a watchdog mechanism that does not work in safe mode and therefore the nextion displays this fact (being in safe mode). In previous releases not problem at all to update firmware in safe mode using this mechanism.

In fact, having "num_attempts: 3" in the safe mode section, I have the same results by failing the booting more than 3 times.

This is the reason a I suspect it is a bug introduced recently because I changed nothing in my YAML, only OTA / Safe mode sections.

Thx

closed 2025-01-12 by github-actions[bot]