esphome/issues #6207: DS18B20 freaking out / continuously reporting 85°C

JiriPrchal, 2024-08-27:
### The problem

More parasitic powered sensors on bus, 1k pull up. Seems like every time after reset it publishes 85. And sometimes repeated bus reset, so one of the sensors has 85˚C all the time.
In older versions worked fine (don't know which but some from 2023).

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
one_wire:
 - platform: gpio
   pin: 27

sensor:
 - platform: dallas_temp
   address: 0x724b847d0a646128
   name: rain water temperature
   icon: mdi:thermometer-water
   update_interval: 1s
   filters:
    - sliding_window_moving_average:
         window_size: 256
         send_every: 16

 - platform: dallas_temp
   address: 0xe36168730a646128
   name: main pipe temperature
   update_interval: 1s
   filters:
    - sliding_window_moving_average:
         window_size: 256
         send_every: 16

 - platform: dallas_temp
   address: 0x2a4361700a646128
   name: tank air temperature
   id: tank_air_temperature
   update_interval: 1s
   filters:
    - sliding_window_moving_average:
         window_size: 256
         send_every: 16
```

### Additional information

Had quick look to source code and seems to me there is no first reading throw nor scratchpad 0x0c test.
Also upon datasheet for parasitic power there should be “strong pull up” during measurement what could be done by GPIO set OUTPUT an HIGH.

ThomDietrich, 2024-08-27:
This bug is also discussed here:
https://community.home-assistant.io/t/ds18b20-freaking-out-continuously-reporting-85-c/346467/20

Relevant source that discusses this "feature" and presets a technical mitigation: https://github.com/cpetrich/counterfeit_DS18B20#solution-to-the-85-c-problem

ssieb, 2024-08-27:
Parasite power is not currently supported.  But it works for other people if you sequence the sensors.  Disable the sensor updates and use an `interval:` to update the sensors individually with at least 1s delay between them.

ssieb, 2024-08-27:
Oh, I just noticed what's in your yaml.  You can't update them that fast.  Each sensor needs 1s to sample and you can't do anything else on the bus during that time if you're using parasite power.  The 85C is happening because the sensor is resetting before esphome reads the sampled value.

JiriPrchal, 2024-08-28:
> Parasite power is not currently supported. 

Not mentioned anywhere.

> But it works for other people if you sequence the sensors. Disable the sensor updates and use an `interval:` to update the sensors individually with at least 1s delay between them.

Could you be more specific, please.

>Oh, I just noticed what's in your yaml. You can't update them that fast. Each sensor needs 1s to sample and you can't do anything else on the bus during that time if you're using parasite power. The 85C is happening because the sensor is resetting before esphome reads the sampled value.

But exactly this worked last year. May be the sequencing was done by dallas component, because the update interval was defined there, now it's at individual sensors.

ssieb, 2024-08-28:
`update_interval: never` on the sensors.

```yaml
interval:
    - interval: 3s
    then:
      - component.update: sensor1
      - delay: 1s
      - component.update: sensor2
      - delay: 1s
      - component.update: sensor3
```
If you really have to sample them every second, then either give them power or put them on separate pins.

ssieb, 2024-08-28:
When parasite power is supported, it will be doing exactly this, but automatically, so it still won't be any faster than that.

JiriPrchal, 2024-08-28:
> `never`

This helps, but the first reading is still 85. Thanks

> When parasite power is supported, it will be doing exactly this, but automatically, so it still won't be any faster than that.

Yes, I understand why it's impossible to start conversion at one and communicate with others.

ssieb, 2024-08-28:
The first reading of each or just the very first reading.  You could put a `delay: 1s` first as well if it's the second case.

JiriPrchal, 2024-08-28:
> The first reading of each or just the very first reading. You could put a `delay: 1s` first as well if it's the second case.

The very first.

ThomDietrich, 2024-08-28:
Hello, just to add my two cents here: I might have misunderstood the topic of this ticket but I am interested specifically in the 85°C reading.
This is my configuration:

```yaml
sensor:
 - platform: dallas_temp
   ...
   filters:
    - filter_out: 85
```

According to the link provided above the wrongful 85°C reading can be identified and correctly ignored by leveraging the scratchpad register. Sounds like a worthy feature request.

https://github.com/cpetrich/counterfeit_DS18B20#solution-to-the-85-c-problem

JiriPrchal, 2024-08-28:
> Hello, just to add my two cents here: I might have misunderstood the topic of this ticket but I am interested specifically in the 85°C reading. This is my configuration:
> 
> ```yaml
> sensor:
>  - platform: dallas_temp
>    ...
>    filters:
>     - filter_out: 85
> ```
> 
> According to the link provided above the wrongful 85°C reading can be identified and correctly ignored by leveraging the scratchpad register. Sounds like a worthy feature request, or bug mitigation if we want to be picky.
> 
> https://github.com/cpetrich/counterfeit_DS18B20#solution-to-the-85-c-problem

And what if your measured temp would be 85?

ThomDietrich, 2024-08-28:
If measured temperature is exactly 85.0000 then this value is ignored. Chances are high reading will be 85.1 or 84.9 a minute later. Nothing to be scared of.
But more importantly, the linked article specifically mentions that there is a technical method to classify those readings and filter them correctly. This should imho be a code improvement.

ThomDietrich, 2024-12-28:
Is there no interest to implement this fix? @ssieb I can see that you did the one wire platform refactor. Previous authors were @leejoow and @Benout3, as well as Otto Winter himself.

ssieb, 2024-12-28:
I do plan to add parasitic support, but I haven't had a chance to get to it.
I won't add that 85 rejection based on an undocumented reserved data field.

JiriPrchal, 2024-12-29:
> I do plan to add parasitic support, but I haven't had a chance to get to it. 

I did it, but it's in early state, as a part of MAX31888 driver...
During Jan I'll push it.

ThomDietrich, 2024-12-30:
> I won't add that 85 rejection based on an undocumented reserved data field.

That's a fair point. There seems to be another theory how to correctly avoid these wrongful readings, described e.g. here: https://www.espboards.dev/sensors/ds18b20/ --> "DS18B20 Dallas Temperature Sensor Troubleshooting"

Is this applicable here?

ssieb, 2024-12-30:
That's just a description of the normal reading process which is what we do.

ThomDietrich, 2024-12-30:
Understood, great. With that out of the way, do you see any additional improvement that could be implemented to resolve this issue? Seems like it's a pretty common and well understood issue. Why not implement the fix or workaround (however we want to call it) right in esphome?

ssieb, 2024-12-30:
The only cause I see is with parasitic powered sensors and that's planned to be added.

nomcode, 2025-01-07:
If it helps for testing I'm keen to get onboard.
I've got a bunch of genuine (they had better be, Dallas sent me them!) Dallas DS18B20 and DS18S20 which are wired parasitic mode plus some genuine DS18S20PAR sensors on shelf.
They all worked well under tasmota but I'm migrating everything over to esphome hence finding this thread.

I've tried a few of the things above like interrupt but actually they read better without.
There are 8 in this particular chain weirdly some in the middle (a mixture) read better than ones at the ends. The cable is differing lengths between each. only 3 sensors are ever read reliably. 
Another string with 4 sensors but on a much longer cable, approx 2m, only one of the senors doesnt read, the one nearest to the ESP.
Unfortunately the power pin was cropped on all of them for these and several other chains.
ESP8266 or ESP32 same results really.

Max 3 sensors being read. Coincidence ?
OP only had 3 sensors connected.

ssieb, 2026-09-02:
@JiriPrchal can you test it?

JiriPrchal, 2026-09-03:
Sorry for the delay.
As far as I can test it (3 sensors): it WORKS. 
```
 - platform: dallas_temp
   update_interval: 3s
```
```
[13:50:32.470][D][dallas.temp.sensor:054]: 'tank air temperature': Got Temperature=18.750000°C
[13:50:32.998][D][dallas.temp.sensor:054]: 'main pipe temperature': Got Temperature=18.875000°C
[13:50:33.600][D][dallas.temp.sensor:054]: 'rain water temperature': Got Temperature=16.812500°C
```
Thanx @ssieb 👍

closed 2026-09-08 by jesserockz