esphome/issues #6212: PlatformIO CI fails with "fatal error: ../../lv_conf.h: No such file or directory"

ivankravets, 2024-08-29:
### The problem

See https://github.com/platformio/platformio-core/actions/runs/10611496736/job/29411142101

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
https://github.com/platformio/platformio-core/blob/develop/.github/workflows/projects.yml#L20
```

### Anything in the logs that might be useful for us?

```txt
https://github.com/platformio/platformio-core/actions/runs/10611496736/job/29411142101
```

jesserockz, 2024-08-29:
ESPHome does not (and not expected to) compile using the root `platformio.ini` project file. 
ESPHome always generates an entirely new project in a build folder and builds that only with the components that the project needs.

The specific issue you are seeing above is because that `lv_conf.h` file is generated by the python parts of ESPHome during this process.

ivankravets, 2024-08-30:
Thanks! We temporarily disabled ESPHome from CI https://github.com/platformio/platformio-core/commit/b9d27240b5a7aa0bdd22d285d416ad1a8009ef08

Could we ask you to use `pio upgrade --dev` for ESPHome's CI? In 99.99% of our development branch is stable. Nevertheless, sometime the issue can happen and it would be great if we fix it before make a stable release.

closed 2025-01-12 by github-actions[bot]