esphome/issues #6213: BMP280 is found and configured but not reporting temperature or pressure

Dave-Jones-1963, 2024-08-29:
### The problem

I have connected a BMP280 module to an NodeMCU Lua Lolin V3 Module ESP8266 ESP-12F WIFI Wifi Development Board and can see it over i2c but don't seem to get any temperature or pressure data back - Before getting esphome on the microcontroller, i used Arduino IDE to check everything was working and could see data coming back with a simple bmp280 test script so I know the two devices are working correctly!

### Which version of ESPHome has the issue?

2024.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core 2024.7.4 - Supervisor 2024.08.0 - Operating System 13.1

### What platform are you using?

ESP8266

### Board

nodemcu v3

### Component causing the issue

bmp280 i2c

### Example YAML snippet

```yaml
substitutions:
  name: av-cupboard-sensor
  friendly_name: av-cupboard-sensor

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  project:
    name: esphome.web
    version: dev

esp8266:
  board: nodemcuv2
  framework: 
    version: recommended

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted" 

# Allow Over-The-Air updates
ota:
  platform: esphome
  password: !secret ota_av_cupboard_sensor

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Set up a wifi access point
  ap: {}
  

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp8266.yaml@main
  import_full_config: true

# To have a "next url" for improv serial
web_server:

i2c:
  sda: GPIO04
  scl: GPIO05
  scan: true
  id: bus_a

sensor:
  - platform: bmp280
    i2c_id: bus_a
    address: 0x76
    temperature:
      name: "AV cupboard temperature"
      id: bmp280_temperature
      oversampling: 16x
    pressure:
      name: "AV cupboard pressure"
      id: bmp280_pressure
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
Logs seem great except no temp or pressure data - I expect I am doing something wrong but cannot see it
INFO ESPHome 2024.8.0
INFO Reading configuration /config/esphome/av-cupboard-sensor.yaml...
INFO Starting log output from 10.88.1.229 using esphome API
INFO Successfully connected to av-cupboard-sensor @ 10.88.1.229 in 0.004s
INFO Successful handshake with av-cupboard-sensor @ 10.88.1.229 in 2.383s
[13:52:25][I][app:100]: ESPHome version 2024.8.0 compiled on Aug 29 2024, 13:41:31
[13:52:25][I][app:102]: Project esphome.web version dev
[13:52:25][C][wifi:600]: WiFi:
[13:52:25][C][wifi:428]:   Local MAC: C8:C9:A3:26:12:15
[13:52:25][C][wifi:433]:   SSID: 'IoT'[redacted]
[13:52:25][C][wifi:436]:   IP Address: 10.88.1.229
[13:52:25][C][wifi:439]:   BSSID: 62:22:32:3B:4C:89[redacted]
[13:52:25][C][wifi:441]:   Hostname: 'av-cupboard-sensor'
[13:52:25][C][wifi:443]:   Signal strength: -67 dB ▂▄▆█
[13:52:25][C][wifi:447]:   Channel: 11
[13:52:25][C][wifi:448]:   Subnet: 255.255.255.0
[13:52:25][C][wifi:449]:   Gateway: 10.88.1.1
[13:52:25][C][wifi:450]:   DNS1: 192.168.3.252
[13:52:25][C][wifi:451]:   DNS2: 192.168.3.252
[13:52:25][C][logger:185]: Logger:
[13:52:25][C][logger:186]:   Level: DEBUG
[13:52:25][C][logger:188]:   Log Baud Rate: 115200
[13:52:25][C][logger:189]:   Hardware UART: UART0
[13:52:25][C][i2c.arduino:071]: I2C Bus:
[13:52:25][C][i2c.arduino:072]:   SDA Pin: GPIO4
[13:52:25][C][i2c.arduino:073]:   SCL Pin: GPIO5
[13:52:25][C][i2c.arduino:074]:   Frequency: 50000 Hz
[13:52:25][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[13:52:25][I][i2c.arduino:096]: Results from i2c bus scan:
[13:52:25][I][i2c.arduino:102]: Found i2c device at address 0x76
[13:52:25][C][bmp280.sensor:124]: BMP280:
[13:52:25][C][bmp280.sensor:125]:   Address: 0x76
[13:52:25][C][bmp280.sensor:137]:   IIR Filter: OFF
[13:52:25][C][bmp280.sensor:138]:   Update Interval: 60.0s
[13:52:25][C][bmp280.sensor:140]:   Temperature 'AV cupboard temperature'
[13:52:25][C][bmp280.sensor:140]:     Device Class: 'temperature'
[13:52:25][C][bmp280.sensor:140]:     State Class: 'measurement'
[13:52:25][C][bmp280.sensor:140]:     Unit of Measurement: '°C'
[13:52:25][C][bmp280.sensor:140]:     Accuracy Decimals: 1
[13:52:25][C][bmp280.sensor:141]:     Oversampling: 16x
[13:52:25][C][bmp280.sensor:142]:   Pressure 'AV cupboard pressure'
[13:52:25][C][bmp280.sensor:142]:     Device Class: 'pressure'
[13:52:25][C][bmp280.sensor:142]:     State Class: 'measurement'
[13:52:25][C][bmp280.sensor:142]:     Unit of Measurement: 'hPa'
[13:52:25][C][bmp280.sensor:142]:     Accuracy Decimals: 1
[13:52:25][C][bmp280.sensor:143]:     Oversampling: 16x
[13:52:25][C][captive_portal:088]: Captive Portal:
[13:52:25][C][web_server:145]: Web Server:
[13:52:25][C][web_server:146]:   Address: av-cupboard-sensor.local:80
[13:52:25][C][mdns:116]: mDNS:
[13:52:25][C][mdns:117]:   Hostname: av-cupboard-sensor
[13:52:25][C][esphome.ota:073]: Over-The-Air updates:
[13:52:25][C][esphome.ota:074]:   Address: av-cupboard-sensor.local:8266
[13:52:25][C][esphome.ota:075]:   Version: 2
[13:52:25][C][esphome.ota:078]:   Password configured
[13:52:25][C][safe_mode:018]: Safe Mode:
[13:52:25][C][safe_mode:019]:   Boot considered successful after 60 seconds
[13:52:25][C][safe_mode:021]:   Invoke after 10 boot attempts
[13:52:25][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[13:52:25][C][api:139]: API Server:
[13:52:25][C][api:140]:   Address: av-cupboard-sensor.local:6053
[13:52:25][C][api:142]:   Using noise encryption: YES
[13:52:25][C][improv_serial:032]: Improv Serial:
```

### Additional information

Micro controller is an NodeMCU Lua Lolin V3 Module ESP8266 ESP-12F WIFI Wifi Development Board.
https://www.az-delivery.uk/products/nodemcu-lolin-v3-modul-mit-esp8266

Vjee1991, 2024-08-29:
try changing the adress to 0x77

Dave-Jones-1963, 2024-08-30:
> try changing the adress to 0x77

Strangely enough as the i2c device is found at 0x76 and identified - when changing to 0x77 it fails completely!

Dave-Jones-1963, 2024-08-30:
> > try changing the adress to 0x77
> 
> Strangely enough as the i2c device is found at 0x76 and identified - when changing to 0x77 it fails completely!

Even more strangely, when flashing back to 0x76 it has suddenly started working! That really is strange - the only thing I changed was address 0x76 to 0x77 and then back again!

Vjee1991, 2024-08-30:
> Even more strangely, when flashing back to 0x76 it has suddenly started working! That really is strange - the only thing I changed was address 0x76 to 0x77 and then back again!

Yeah that was my second option to change back. I had that too when it suddenly didn't work anymore and tried to change it back.

Dave-Jones-1963, 2024-09-25:
This is really annoying! For some reason when updating to the latest version of esphome, my bmp280_12c device has stopped reporting temperature and pressure - even reverting to previous version did not bring it back. The NodeMCU LoLin microcontroller can still see the bmp280 but there are NO results coming back - if I connect the BMP280 to an Arduino and run the BMP280 test script it works fine! What has changed?

Dave-Jones-1963, 2024-09-26:
Still no idea why it has suddenly stopped giving any pressure or temperature readings even when using an Arduino uno I can see readings. The Esphome code for the device sees the bmp280 but no readings!

ssieb, 2024-09-27:
Set the log level to verbose and see if there's anything.
Getting serial logs from boot would be good.

Dave-Jones-1963, 2024-09-28:
What does this tell me exactly? Do I try to change the i2c pins and reconnect to see if it is a GPIO issue?

[18:26:57][C][logger:185]: Logger:
[18:26:57][C][logger:186]:   Level: VERY_VERBOSE
[18:26:57][C][logger:188]:   Log Baud Rate: 115200
[18:26:57][C][logger:189]:   Hardware UART: UART0
[18:26:57][C][i2c.arduino:071]: I2C Bus:
[18:26:57][C][i2c.arduino:072]:   SDA Pin: GPIO4
[18:26:57][C][i2c.arduino:073]:   SCL Pin: GPIO5
[18:26:57][C][i2c.arduino:074]:   Frequency: 50000 Hz
[18:26:57][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[18:26:57][I][i2c.arduino:096]: Results from i2c bus scan:
[18:26:57][I][i2c.arduino:102]: Found i2c device at address 0x76
[18:26:57][C][bmp280_i2c.sensor:022]:   Address: 0x76
[18:26:57][C][bmp280.sensor:132]: BMP280:
[18:26:57][C][bmp280.sensor:144]:   IIR Filter: OFF
[18:26:57][C][bmp280.sensor:145]:   Update Interval: 60.0s
[18:26:57][C][bmp280.sensor:147]:   Temperature 'AV cupboard temperature'
[18:26:57][C][bmp280.sensor:147]:     Device Class: 'temperature'
[18:26:57][C][bmp280.sensor:147]:     State Class: 'measurement'
[18:26:57][C][bmp280.sensor:147]:     Unit of Measurement: '°C'
[18:26:57][C][bmp280.sensor:147]:     Accuracy Decimals: 1
[18:26:57][C][bmp280.sensor:148]:     Oversampling: 16x
[18:26:57][C][bmp280.sensor:149]:   Pressure 'AV cupboard pressure'
[18:26:57][C][bmp280.sensor:149]:     Device Class: 'pressure'
[18:26:57][C][bmp280.sensor:149]:     State Class: 'measurement'
[18:26:57][C][bmp280.sensor:149]:     Unit of Measurement: 'hPa'
[18:26:57][C][bmp280.sensor:149]:     Accuracy Decimals: 1
[18:26:57][C][bmp280.sensor:150]:     Oversampling: 16x
[18:26:57][C][captive_portal:089]: Captive Portal:
[18:26:57][C][web_server:145]: Web Server:
[18:26:57][C][web_server:146]:   Address: av-cupboard-temperature-sensor.local:21080
[18:26:57][C][mdns:116]: mDNS:
[18:26:57][C][mdns:117]:   Hostname: av-cupboard-temperature-sensor
[18:26:57][V][mdns:118]:   Services:
[18:26:57][V][mdns:120]:   - _esphomelib, _tcp, 6053
[18:26:57][V][mdns:122]:     TXT: friendly_name = av-cupboard-sensor
[18:26:57][V][mdns:122]:     TXT: version = 2024.9.1
[18:26:57][V][mdns:122]:     TXT: mac = c8c9a3261215
[18:26:57][V][mdns:122]:     TXT: platform = ESP8266
[18:26:57][V][mdns:122]:     TXT: board = nodemcuv2
[18:26:57][V][mdns:122]:     TXT: network = wifi
[18:26:57][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[18:26:57][V][mdns:122]:     TXT: project_name = esphome.web
[18:26:57][V][mdns:122]:     TXT: project_version = dev
[18:26:57][V][mdns:120]:   - _http, _tcp, 21080
[18:26:57][C][esphome.ota:073]: Over-The-Air updates:
[18:26:57][C][esphome.ota:074]:   Address: av-cupboard-temperature-sensor.local:8266
[18:26:57][C][esphome.ota:075]:   Version: 2
[18:26:57][C][esphome.ota:078]:   Password configured
[18:26:58][C][safe_mode:018]: Safe Mode:
[18:26:58][C][safe_mode:019]:   Boot considered successful after 60 seconds
[18:26:58][C][safe_mode:021]:   Invoke after 10 boot attempts
[18:26:58][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[18:26:58][C][api:139]: API Server:
[18:26:58][C][api:140]:   Address: av-cupboard-temperature-sensor.local:6053
[18:26:58][C][api:142]:   Using noise encryption: YES
[18:27:05][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=21394 (now=31394)
[18:27:15][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=31394 (now=41395)
[18:27:17][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=4294950473 (now=43177)
[18:27:17][VV][scheduler:225]: Running interval 'update' with interval=60000 last_execution=4294950729 (now=43433)
[18:27:17][V][bmp280.sensor:158]: Sending conversion request...
[18:27:17][VV][i2c.arduino:176]: 0x76 TX F4B5
[18:27:17][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[18:27:25][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=41394 (now=51399)
[18:27:33][VV][api.service:697]: on_ping_request: PingRequest {}
[18:27:33][VV][api.service:043]: send_ping_response: PingResponse {}
[18:27:35][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=51394 (now=61397)
[18:27:38][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[18:27:45][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=61394 (now=71394)
[... 1 lines cut ...]
[18:27:57][VV][api.connection:132]: Sending keepalive PING...
[18:27:57][VV][api.service:037]: send_ping_request: PingRequest {}
[18:27:57][VV][api.service:706]: on_ping_response: PingResponse {}
[18:28:05][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=81394 (now=91394)
[18:28:13][VV][api.service:697]: on_ping_request: PingRequest {}
[18:28:13][VV][api.service:043]: send_ping_response: PingResponse {}
[18:28:15][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=91394 (now=101394)
[18:28:17][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=43177 (now=103177)
[18:28:17][VV][scheduler:225]: Running interval 'update' with interval=60000 last_execution=43433 (now=103438)
[18:28:17][V][bmp280.sensor:158]: Sending conversion request...
[18:28:17][VV][i2c.arduino:176]: 0x76 TX F4B5
[18:28:17][VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[18:28:25][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=101394 (now=111394)
[... cut ...]
[18:28:57][VV][api.service:037]: send_ping_request: PingRequest {}
[18:28:57][VV][api.service:706]: on_ping_response: PingResponse {}
[18:29:05][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=141394 (now=151394)
[18:29:15][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=151394 (now=161394)
[18:29:17][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=103177 (now=163177)
[18:29:17][VV][scheduler:225]: Running interval 'update' with interval=60000 last_execution=103433 (now=163435)

ssieb, 2024-09-29:
The sensor isn't acknowledging the i2c request.  Can you get serial logs from boot to see the bme setup?

ssieb, 2024-09-29:
Plug it into your computer using the USB port, then use either the web installer or a program like putty to watch the logs.  Baud rate 115200.  Press the reset button on the board to start from boot.

Dave-Jones-1963, 2024-09-30:
SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
[I][logger:034]: Log initialized
[C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[I][i2c.arduino:218]: Performing I2C bus recovery
[V][i2c.arduino:042]: Scanning i2c bus for active devices...
[VV][i2c.arduino:176]: 0x08 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x09 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x0A TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x0B TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x0C TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x0D TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x0E TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x0F TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x10 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x11 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x12 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x13 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x14 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x15 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x16 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x17 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x18 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x19 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x1A TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x1B TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x1C TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x1D TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x1E TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x1F TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x20 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x21 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x22 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x23 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x24 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x25 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x26 TX
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][i2c.arduino:176]: 0x27 TX
[... 643 lines cut ...]
[String] 'id: 215107 ... 15106)': Reallocating large String(142 -> 146 bytes)
                                                                             [V][bmp280.sensor:158]: Sending conversion request...
[VV][i2c.arduino:176]: 0x76 TX F4B5
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][scheduler:225]: Running interval '' with interval=60000 last_execution=158279 (now=218279)
[VV][scheduler:225]: Running interval '' with interval=10000 last_execution=210550 (now=220550)
[VV][scheduler:225]: Running interval '' with interval=10000 last_execution=220550 (now=230550)
[VV][api.service:697]: on_ping_request: PingRequest {}
[VV][api.service:043]: send_ping_response: PingResponse {}
[VV][scheduler:225]: Running interval '' with interval=10000 last_execution=230550 (now=240552)
[VV][scheduler:225]: Running interval '' with interval=10000 last_execution=240550 (now=250550)
[VV][scheduler:225]: Running interval '' with interval=10000 last_execution=250550 (now=260550)
[VV][scheduler:225]: Running interval '' with interval=10000 last_execution=260550 (now=270553)
[VV][api.service:697]: on_ping_request: PingRequest {}
[VV][api.service:043]: send_ping_response: PingResponse {}
[VV][scheduler:225]: Running interval 'update' with interval=60000 last_execution=215106 (now=275111)
[String] 'id: 275112 ... 75111)': Reallocating large String(142 -> 146 bytes)
                                                                             [V][bmp280.sensor:158]: Sending conversion request...
[VV][i2c.arduino:176]: 0x76 TX F4B5
[VV][i2c.arduino:202]: TX failed: not acknowledged: 2
[VV][scheduler:225]: Running interval '' with interval=60000 last_execution=218279 (now=278279)
[VV][scheduler:225]: Running interval '' with interval=10000 last_execution=270550 (now=280550)
[VV][scheduler:225]: Running interval '' with interval=10000 last_execution=280550 (now=290550)
[VV][scheduler:225]: Running interval '' with interval=10000 last_execution=290550 (now=300550)
`

ssieb, 2024-10-04:
What is the BMP280 module you have?

Dave-Jones-1963, 2024-10-04:
One for AZ-delivery it works fine using script on an Arduino Uno r3

ssieb, 2024-10-04:
I think you have to connect the SDO pin to either GND or VCC.  I can't tell if there's a pullup on board, but the datasheet specifically says it can't be floating.  There's even a comment on the AZ-Delivery site for the module (translated):
> Setup took a while
> Now the breakout board works very well on a Lolin32 Lite (SDA pin 16, SCL pin 4). However, to fix the address 0x76, the SDO pin on the breakout board side must be connected to ground, otherwise nothing happens and the sensor is not found. It's a shame that this isn't mentioned in the description, it would have saved me a lot of time.