esphome/issues #6216: ESPHome Dashboard handles name_add_mac_suffix badly

hickey, 2024-08-31:
### The problem

The dashboard does not handle the name of the end device correctly when `esphome.name_add_mac_suffix` is set to `true`. As a result it is required to specify `wifi.use_address` in order to communicate with the end device. 

The dashboard does not actually calculate a hostname and uses what has been specified as the `esphome.name` attribute. This causes a discontinutity when `esphome.name_add_mac_suffix` is specified in the configuration YAML as after installation, the device will advertise itself with the MAC suffix yet the dashboard will continue to use what is specified in `esphome.name`. 

This causes another problem where a device gets represented as two tiles as demonstrated in the following screen snapshot.

[image]

I think the problem exists because the device is being discovered using the MAC suffix, but the tile for the actual device (Spot enclosure in the above snapshot) actually has the `esphome.name` set to `fan_control` and then uses the `wifi.use_address` to specify the hostname to connect to the device. 

I think the proper way to handle this is for the dashboard to calculate the hostname by using the `wifi.use_address` if it is specified and when it is not the hostname is calculated by starting with the `esphome.name` value and then check for `esphome.name_add_mac_suffix` to be `true` and append the MAC suffix to the name value. There is the situation where a developer specifies `esphome.name` with the MAC suffix (incorrectly) and also specifies `esphome.name_add_mac_suffix` as `true`. This in actuallity should be treated as an error condition with a warning message on the tile of the device, but it can easily be remedied in code by eliminating a second MAC suffix when the hostname is calculated. 

This issue was encountered when trying to develop a configuration to be used as a factory install. The example YAML below is the base configuration that is being used during the development. Ideally if the hostname is calculated as described above then `wifi.use_address` would not need to be specified and it would not cause problems when the setting is not removed for the release of the image. Imagine a number of end devices created with the same image announcing the same hostname. Not exactly the user experience one desires :-)

For completeness, the following is the MDNS announcement from the end device.

[image]

### Which version of ESPHome has the issue?

2024.7.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32-d1

### Component causing the issue

ESPHome core configuration

### Example YAML snippet

```yaml
substitutions:
  name: fan_control
  friendly_name: Spot enclosure

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: true
  project:
    name: hickey.fan_control
    version: 0.1.0

mdns:
  disabled: false

wifi:
  use_address: fan_control-689314
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
Nothing specific. Everything is operating correctly, just the dashboard calculates the wrong hostname.
```

ssieb, 2024-08-31:
The dashboard doesn't know the mac address.  If you have a device in the dashboard, you should not have the mac suffix enabled.  That's why the dashboard changes the setting to false when you adopt.

hickey, 2024-09-01:
Doesn't the dashboard learn of devices through mDNS? It is in the mDNS announcement. So the dashboard does have access to the MAC address and can be stored for future use if it is not discovered when the dashboard is loaded. 

In addition, the MAC suffix is known at some point as the configuration YAML generally has the MAC suffix as part of the filename. So that is another source for the suffix. Although I do have one configuration YAML that was created with out a suffix, but I am not sure how I adopted that device. 

Another source for the MAC suffix is when the device is adopted. You have it right in the tile to be adopted (`fan_control-689314` in my screen shot above). This is probably related to my first point.