esphome/issues #6220: Failed to load packages. m5stack/m5stack-atom-lite.yaml does not exist in repository

aselling, 2024-09-02:
### The problem

After updating my Atom M5Stack Lite that I use for bluetooth proxies with Home Assistant late in august all of my M5Atom went offline.

I have reinstalled them through: https://esphome.io/projects/index.html
They show up in Home Assistant and I have adopted them and they are online.

But now they won't compile

Error message says:
INFO ESPHome 2024.8.1
INFO Reading configuration /config/esphome/m5stack-atom-lite-8538f8.yaml...
Failed config

packages: [source /config/esphome/m5stack-atom-lite-8538f8.yaml:5]
  
  Failed to load packages. m5stack/m5stack-atom-lite.yaml does not exist in repository.
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/m5stack/m5stack-atom-lite.yaml@main

Here is the Yaml-code: 
substitutions:
  name: m5stack-atom-lite-8538f8
  friendly_name: Bluetooth Proxy garage
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/m5stack/m5stack-atom-lite.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  static_ip: 192.168.1.232
  gateway: 192.168.1.1
  subnet: 255.255.255.0
  dns1: 192.168.1.1

### Which version of ESPHome has the issue?

2024.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.8.3

### What platform are you using?

ESP32

### Board

m5stack-atom-lite

ssieb, 2024-09-02:
It works for me, but your wifi config isn't valid.

aselling, 2024-09-02:
Yes missing manual_ip:

But when installing I get this:
during RTL pass: vartrack
/data/cache/platformio/packages/framework-espidf/components/libsodium/libsodium/src/libsodium/crypto_generichash/blake2b/ref/blake2b-compress-ref.c: In function 'blake2b_compress_ref':
/data/cache/platformio/packages/framework-espidf/components/libsodium/libsodium/src/libsodium/crypto_generichash/blake2b/ref/blake2b-compress-ref.c:93:1: internal compiler error: Segmentation fault
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
*** [.pioenvs/m5stack-atom-lite-8538fc/libsodium/libsodium/src/libsodium/crypto_generichash/blake2b/ref/blake2b-compress-ref.o] Error 1
========================= [FAILED] Took 124.83 seconds =========================

ssieb, 2024-09-02:
That's a completely different issue from the original one.  What are you running HA on?

aselling, 2024-09-02:
Inte Mac mini and from Mac OS virtualbox openSuse

ssieb, 2024-09-02:
That's probably a virtualbox issue.  Make sure you have the latest version.

aselling, 2024-09-02:
Ok thanks. Will close this

closed 2024-09-02 by aselling