esphome/issues #6222: i2s_audio_speaker not playing all audio data

esand, 2024-09-03:
### The problem

I am trying to use the speaker component to play() raw audio. I have audio encoded as signed (little endian) 16-bit, 16kHz mono and then converted to hex using `xxd`. The audio sample is 1.633s long, but it doesn't play all the way through.

I have tried various audio encodings, and out of the ones that sound like they play correctly, none of them seem to play all the way through.

I am using an external DAC (MAX98357A) and I am uncertain if this might be causing the issue. I have not tested with internal DAC.

### Which version of ESPHome has the issue?

2024.8.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

ESP32-CH340 type C

### Component causing the issue

i2s_audio_speaker

### Example YAML snippet

```yaml
logger:
  level: VERY_VERBOSE

esphome:
  name: $name
  name_add_mac_suffix: true
  friendly_name: $display_name
  project:
    name: $project_name
    version: $project_version

  includes:
    - audio/test_sound.h

  on_boot:
    priority: -100
    then:
      - delay: 3s
      - lambda: |-
          id(i2s_speaker).play(test_sound_pcm, test_sound_pcm_len);

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: recommended

i2s_audio:
  i2s_lrclk_pin: GPIO22
  i2s_bclk_pin: GPIO23

speaker:
  - platform: i2s_audio
    id: i2s_speaker
    dac_type: external
    i2s_dout_pin:
      number: GPIO21
    mode: mono
```

### Anything in the logs that might be useful for us?

```txt
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '0', to_send_length '1024', remaining '116416'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '1024', to_send_length '1024', remaining '115392'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '2048', to_send_length '1024', remaining '114368'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '3072', to_send_length '1024', remaining '113344'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '4096', to_send_length '1024', remaining '112320'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '5120', to_send_length '1024', remaining '111296'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '6144', to_send_length '1024', remaining '110272'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '7168', to_send_length '1024', remaining '109248'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '8192', to_send_length '1024', remaining '108224'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '9216', to_send_length '1024', remaining '107200'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '10240', to_send_length '1024', remaining '106176'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '11264', to_send_length '1024', remaining '105152'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '12288', to_send_length '1024', remaining '104128'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '13312', to_send_length '1024', remaining '103104'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '14336', to_send_length '1024', remaining '102080'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '15360', to_send_length '1024', remaining '101056'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '16384', to_send_length '1024', remaining '100032'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '17408', to_send_length '1024', remaining '99008'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '18432', to_send_length '1024', remaining '97984'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '19456', to_send_length '1024', remaining '96960'
[08:55:32][D][i2s_audio.speaker:269]: DEBUG_i2s: index '20480', to_send_length '1024', remaining '95936'
[08:55:32][D][i2s_audio.speaker:272]: DEBUG_i2s: exited early?
```

### Additional information

I added additional debug logging to the i2s_audio_speaker.cpp file to see what might be causing the issue. I've never used C++ before so I'm pretty limited in my abilities to fix/debug in depth.

Here's where I added the extra debug lines in i2s_audio_speaker.cpp:
```c
...
  while (remaining > 0) {
    DataEvent event;
    event.stop = false;
    size_t to_send_length = std::min(remaining, BUFFER_SIZE);
    event.len = to_send_length;
    ESP_LOGD(TAG, "DEBUG_i2s: index '%d', to_send_length '%d', remaining '%d'", index, to_send_length, remaining);
    memcpy(event.data, data + index, to_send_length);
    if (xQueueSend(this->buffer_queue_, &event, 1000) != pdTRUE) {
      ESP_LOGD(TAG, "DEBUG_i2s: exited early?");
      return index;
    }
    remaining -= to_send_length;
    index += to_send_length;
  }
  ESP_LOGD(TAG, "DEBUG_i2s: all done (index '%d')", index);
  return index;
}
```

esand, 2024-09-03:
To note for those paying close attention, the `xQueueSend()` call has the ticks set to 1000 in what I showed as a test. I was testing to see if waiting longer would allow the queue to clear and get the sound to complete but it had no impact. Original value of 0, 10 and 1000 had no apparent difference.

esand, 2024-09-07:
It seems the issue is indirectly with `BUFFER_COUNT` which is set to `20`; the buffer is being filled and not emptied.

I tested with setting `BUFFER_COUNT` to `100` and it allowed my audio clip to play completely, however with a buffer size of 100 it's really pushing the limits of available memory, so I wouldn't consider this a viable solution.

esand, 2024-09-07:
Setting `xQueueSend()` to `portMAX_DELAY` to have it wait indefinitely until the queue can take more items doesn't seem to work; no audio plays. I don't know enough about the queue system to know why it wouldn't be playing right away, or how to trigger that.

ricardo-manriquez, 2024-10-10:
Hello Eli!
I was trying to do the same, have a speaker component play preloaded sounds on ESPHome.

I saw your other issues as well and I found a hacky solution, the problem stems from the way the FreeRTOS sound player task is created. This means that when the play function runs, it runs the start method but this doesn't _directly_ start the player, so they are run sequentially and the player never has the chance to consume the data in the buffer. Getting the play function stuck and quitting because of the buffer being full.
What I did which I think can be improved on is instead of running start, I ran start_ which is the function that actually creates the task. 
Finally I added a taskYIELD() so that it can yield when the buffer is full, with this the buffer has the opportunity to empty itself and more data can be loaded. the actual source code is as follows:
```
#include <freertos/task.h>

...

size_t I2SAudioSpeaker::play(const uint8_t *data, size_t length) {
  if (this->is_failed()) {
    ESP_LOGE(TAG, "Cannot play audio, speaker failed to setup");
    return 0;
  }
  if (this->state_ != speaker::STATE_RUNNING && this->state_ != speaker::STATE_STARTING) {
    // Call start_() directly so that the task is actually allocated
    this->state_ = speaker::STATE_STARTING; 
    this->start_();
  }
  size_t remaining = length;
  size_t index = 0;
  while (remaining > 0) {
    DataEvent event;
    event.stop = false;
    size_t to_send_length = std::min(remaining, BUFFER_SIZE);
    event.len = to_send_length;
    memcpy(event.data, data + index, to_send_length);
    if (xQueueSend(this->buffer_queue_, &event, 0) != pdTRUE) {
      taskYIELD(); // if buffer is full yield so that the player can consume the data
    } else { // do not change the index if the sending failed
      remaining -= to_send_length;
      index += to_send_length;
    }
  }
  return index;
}
```

esand, 2024-10-10:
That's awesome - did you happen to get this code working in the latest version of esphome (2024.9.2)? I just updated and grabbed the i2s_audio component from that version and replaced the play() method with yours but my ESP32 crashes when it tries to play the audio.

I'm wondering if because of the taskYIELD() call, if we have to reset this watchdog timer that kills long running tasks (in my case, I'm ok that it's blocking - for now 😄). It is slightly odd 

```
[12:54:12][C][i2s_audio:028]: Setting up I2S Audio...
[12:54:12][VV][scheduler:032]: set_timeout(name='', timeout=3000)
[12:54:12][C][i2s_audio.speaker:019]: Setting up I2S Audio Speaker...
[12:54:12][I][app:062]: setup() finished successfully!
[12:54:12][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=4294878782 (now=66)
[12:54:12][I][app:100]: ESPHome version 2024.9.2 compiled on Oct 10 2024, 12:47:19
[12:54:12][I][app:102]: Project esand.testdevice version 0.3
[12:54:12][C][logger:185]: Logger:
[12:54:12][C][logger:186]:   Level: VERY_VERBOSE
[12:54:12][C][logger:188]:   Log Baud Rate: 115200
[12:54:12][C][logger:189]:   Hardware UART: UART0
[12:54:15][VV][scheduler:226]: Running timeout '' with interval=3000 last_execution=45 (now=3049)
[12:54:20]E (16298) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[12:54:20]E (16298) task_wdt:  - loopTask (CPU 1)
[12:54:20]E (16298) task_wdt: Tasks currently running:
[12:54:20]E (16298) task_wdt: CPU 0: IDLE
[12:54:20]E (16298) task_wdt: CPU 1: loopTask
[12:54:20]E (16298) task_wdt: Aborting.
[12:54:20]
[12:54:20]abort() was called at PC 0x400df415 on core 0
[12:54:20]
[12:54:20]
[12:54:20]Backtrace:0x40083591:0x3ffbe7ac |<-CORRUPTED
```

However, I also can't seem to get any audio playing with just a stock 2024.9.2 version either. At the moment, I'm not sure if from my testing I've blown my DAC or speaker. I'll have to do a few more tests on my end to figure that out.

esand, 2024-10-10:
Ok @ricardo-manriquez, I got your code working (had a bad setting in my yaml making the audio not play - totally my fault) and was able to eliminate the crashing by adding a call to `arch_feed_wdt();` right before the `taskYIELD();` call. This resets the watchdog and lets the function play the entire audio clip through. You get a warning afterwards though that the call took a long time, but for now it works, and I don't care 😂
Thank you for your suggestion!

I'm leaving the ticket open however, for hopefully obvious reasons.

ricardo-manriquez, 2024-10-11:
Ok I found the function in the voice assistant, they are calling speaker->play repeatedly with at most 4096 bytes of data so the buffer never fills up, but there is no way we can do that in our use case of playing audio clips, I would consider this a bug, because the start function is blocking when it shouldn't. One workaround would be to first send a small amount of data, then the full audio file, this would prime the start function, but a proper solution would be that the play function doesn't block without starting playback.
On another note, maybe my yield is just unnecessary, because the RTOS should take care of it as well, it requires more testing, but it seems like a reasonable solution that shouldn't break other things.

esand, 2024-10-11:
> Do you know of any other components that use the speaker?

It seems it's really only used by people creating assistants and having to relay microphone stuff to the speaker or something. The i2s speaker component seems rather basic - and the only reason I'm even using it is because it's the only module that plays local music files; media_player has more functions but needs a URL.

I'm happy with it working the way it is right now with your fix. I wouldn't spend too much time fiddling with the i2s stuff (unless you want to help contribute to the other work people are doing) because it seems the assistant uptick has sparked some smart people to rethink media_player, speaker and i2s in general and make a bunch of improvements.

mateuszlukasik-PL, 2024-12-01:
Problem with playing full audio still exists in version 2024.11.1, but code of I2S Audio Speaker::play function is different, it has changed, and Ricardo fix can't be applied. But I think there is new problem. I have yaml file like this :

`button:
  - platform: template
    name: Dzwonek
    icon: "mdi:bell"
    on_press:
      then:
        - lambda: |-
            id(i2s_speaker).play(doorbell_sound, doorbell_sound_len);
        - lambda: |-
            char timestamp_string[18];
            time_t current_time = id(ha_time).now().timestamp;
            strftime(timestamp_string, sizeof(timestamp_string), "%H:%M, %d.%m.%Y", localtime(&current_time));
            id(last_pressed).publish_state(timestamp_string);
    id: virtual_button

# Sensor daty ostatniego naciśnięcia dzwonka
text_sensor:
  - platform: template
    name: "Ostatnio przyciśnięty"
    icon: "mdi:history"
    update_interval: 60s
    id: last_pressed`

When I press the button it should play sound from PROGMEM, and it play's (still it doesn't play all the way through) but I have to double-click button. One click doesn't play anything. 

I've done some simple debug and modified "play" function like this...

`size_t I2SAudioSpeaker::play(const uint8_t *data, size_t length, TickType_t ticks_to_wait) {
  if (this->is_failed()) {
    ESP_LOGE(TAG, "Cannot play audio, speaker failed to setup");
    return 0;
  }
  if (this->state_ != speaker::STATE_RUNNING && this->state_ != speaker::STATE_STARTING) {
    this->start();
  }

  ESP_LOGD(TAG, "DEBUG : state - '%d'", this->state_);

  size_t bytes_written = 0;
  if ((this->state_ == speaker::STATE_RUNNING) && (this->audio_ring_buffer_.use_count() == 1)) {
    // Only one owner of the ring buffer (the speaker task), so the ring buffer is allocated and no other components are
    // attempting to write to it.

    // Temporarily share ownership of the ring buffer so it won't be deallocated while writing
    std::shared_ptr<RingBuffer> temp_ring_buffer = this->audio_ring_buffer_;
    bytes_written = temp_ring_buffer->write_without_replacement((void *) data, length, ticks_to_wait);
  }

  ESP_LOGD(TAG, "DEBUG : bytes_writen - '%d'", bytes_written);

  return bytes_written;
}`

result of one-click in LOG is...

`[14:29:16][D][button:010]: 'Dzwonek' Pressed.
[14:29:16][D][i2s_audio.speaker:201]: DEBUG : state - '0'
[14:29:16][D][i2s_audio.speaker:213]: DEBUG : bytes_writen - '0'
[14:29:16][D][i2s_audio.speaker:108]: Starting Speaker
[14:29:16][D][i2s_audio.speaker:113]: Started Speaker
[14:29:17][D][i2s_audio.speaker:120]: Stopping Speaker
[14:29:17][D][i2s_audio.speaker:126]: Stopped Speaker`

and result of double-click in LOG is...

`[14:09:16][D][button:010]: 'Dzwonek' Pressed.
[14:09:16][D][i2s_audio.speaker:201]: DEBUG : state - '0'
[14:09:16][D][i2s_audio.speaker:213]: DEBUG : bytes_writen - '0'
[14:09:16][D][i2s_audio.speaker:108]: Starting Speaker
[14:09:16][D][i2s_audio.speaker:113]: Started Speaker
[14:09:16][D][button:010]: 'Dzwonek' Pressed.
[14:09:16][D][i2s_audio.speaker:201]: DEBUG : state - '2'
[14:09:16][D][i2s_audio.speaker:213]: DEBUG : bytes_writen - '16000'
[14:09:17][D][i2s_audio.speaker:120]: Stopping Speaker
[14:09:17][D][i2s_audio.speaker:126]: Stopped Speaker`

sound plays but ony about half of it's lenght

esand, 2024-12-21:
It's not working on stock 2024.12.2 either. It seems that it almost crashes/hangs in playing the audio. It plays about less than a second of my audio (not even the beginning of it - like it's missing a bit of the beginning and playing a brief bit shortly after), then just sits there and the speaker never goes back to idle - it just sits hung on playing.

Here's the yaml for my test device that I'm using to play the audio:

```yaml
  on_boot:
    priority: -100
    then:
      - delay: 5s
      - speaker.volume_set: 0.9
      - lambda: |-
          id(i2s_speaker).play(long_sound_pcm, long_sound_pcm_len);
      - wait_until:
          not:
            speaker.is_playing:
      - delay: 1s
      - lambda: |-
          id(i2s_speaker).play(long_sound_pcm, long_sound_pcm_len);
```
The 2nd play call was in case it needed to be called a 2nd time (like @mateuszlukasik-PL seemed to be stating), but it never even got to the 2nd call - just hung on the first one.

```
[15:00:49][C][logger:185]: Logger:
[15:00:49][C][logger:186]:   Level: VERY_VERBOSE
[15:00:49][C][logger:188]:   Log Baud Rate: 115200
[15:00:49][C][logger:189]:   Hardware UART: UART0
[15:00:53][VV][scheduler:226]: Running timeout '' with interval=5000 last_execution=43 (now=5047)
[15:00:53][VV][scheduler:032]: set_timeout(name='', timeout=1000)
[15:00:53][D][i2s_audio.speaker:111]: Starting Speaker
[15:00:53][D][i2s_audio.speaker:116]: Started Speaker
[15:00:54][VV][scheduler:226]: Running timeout '' with interval=1000 last_execution=5048 (now=6052)
[15:01:31][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=4294949491 (now=42195)
[15:02:31][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=42195 (now=102195)
[15:03:31][VV][scheduler:226]: Running interval '' with interval=60000 last_execution=102195 (now=162195)
```

I also tried what @kahrendt mentioned over in https://github.com/esphome/issues/issues/6497#issuecomment-2509441488 which was calling start(), then a 50ms delay before calling play(), but it had no change to the outcome.

Also, I noticed that you get compile errors if you use the esp-idf framework; it throws warnings that the i2s speaker component is using a legacy i2s driver.

kahrendt, 2024-12-24:
The play function tries to send all the audio to the speaker in a single pass. If the speaker's buffer can't store the entire audio clip, then it will be cutoff. A semi-fix is to increase the speaker's buffer duration by adding ``buffer_duration: 1000ms`` to the speaker config. Note that this will use more memory, so if your device doesn't have PSRAM, you may run out of space.

The long term fix is to use the new speaker media player (not finished yet) instead of directly playing audio with the speaker component, as it properly handles when not all the data can fit in the speaker's buffer.