esphome/issues #6227: bme68x_bsec2_i2c: Temperature readings too high and can't calibrate

zeirix, 2024-09-04:
### The problem

Hello everybody,

I've tried the bme68x_bsec2_i2c sensor but after two days reverted back to my existing bme680 sensor module, because the temperature reading was one to four degrees too high and after a short period in the state of `Stabilizing` it went into `Uncertain`. I'm using the BME680.

I've tried to bring some amplitude into the `gas_resistance` reading by opening the window and holding it close for some time to help the algorithm to calibrate but nothing changed.

Here's an example of the temperature:
[image]

Switching between LP and ULP mode changed the temperature behavior.
The rel. humidity reading was also about 5% lower than expected with the new module.

### Which version of ESPHome has the issue?

2024.8.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

bme68x_bsec2_i2c

### Example YAML snippet

```yaml
bme68x_bsec2_i2c:
  address: 0x77
  model: bme680
  operating_age: 28d
  sample_rate: LP
  supply_voltage: 3.3V
  state_save_interval: 24h

[...]

  - platform: bme68x_bsec2
    temperature:
      name: "Temperatur"
      accuracy_decimals: 1
      state_topic: /esp32-buero/tempin
    pressure:
      name: "Barometer"
      accuracy_decimals: 1
      state_topic: /esp32-buero/pres
    humidity:
      name: "rel. Luftfeuchte"
      accuracy_decimals: 1
      state_topic: /esp32-buero/humin
    gas_resistance:
      name: "BME68x Widerstand"
      state_topic: /esp32-buero/gasin-res
    iaq:
      name: "BME68x IAQ"
      state_topic: /esp32-buero/gasin-iaq
    iaq_static:
      name: "BME68x IAQ static"
      state_topic: /esp32-buero/gasin-iaq-static
    co2_equivalent:
      name: "BME68x CO2 Equivalent"
      state_topic: /esp32-buero/gasin-eco2
    breath_voc_equivalent:
      name: "BME68x Breath VOC Equivalent"
      state_topic: /esp32-buero/gasin-bvoc
```

plplaaa2, 2024-09-19:
same here

[image]

적용 = apply
자체 offset 적용 = Apply own offset
복귀 = remove BSEC2

DerGuenni, 2024-09-23:
Same here. Also, when applying a manual offset via "temperature_offset" configuration it bumps up the humidity by nearly 10 %.

Using it with RP2040.

luar123, 2024-11-07:
You can try the PR with:
```yaml
external_components:
  - source: github://pr#7728
    components: [ bme68x_bsec2 ]
```

closed 2024-11-13 by kbx81