esphome/issues #6229: cp15barrier_handler

stanislavdavid, 2024-09-06:
### The problem

I have syslog full of this messages:
every 16 seconds

```
2024-09-06T15:02:51.883842+00:00 homeassistant kernel: [ 1918.516010] cp15barrier_handler: 98 callbacks suppressed
2024-09-06T15:02:51.884002+00:00 homeassistant kernel: [ 1918.516032] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf7149c88
2024-09-06T15:02:51.884029+00:00 homeassistant kernel: [ 1918.516069] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf7149ca4
2024-09-06T15:02:51.884044+00:00 homeassistant kernel: [ 1918.516115] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf71645d0
2024-09-06T15:02:51.884056+00:00 homeassistant kernel: [ 1918.516136] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf71645ec
2024-09-06T15:02:51.884069+00:00 homeassistant kernel: [ 1918.516269] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf71645d0
2024-09-06T15:02:51.884081+00:00 homeassistant kernel: [ 1918.516297] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf71645ec
2024-09-06T15:02:51.884093+00:00 homeassistant kernel: [ 1918.516324] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf71645d0
2024-09-06T15:02:51.884103+00:00 homeassistant kernel: [ 1918.516345] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf71645ec
2024-09-06T15:02:51.884114+00:00 homeassistant kernel: [ 1918.516418] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf71645d0
2024-09-06T15:02:51.884125+00:00 homeassistant kernel: [ 1918.516444] "esphome" (4477) uses deprecated CP15 Barrier instruction at 0xf71645ec

```

Does anybody know, what is the problem behind? 

### Which version of ESPHome has the issue?

 2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.0

### What platform are you using?

ESP32

### Board

Orange pi 3B

cfiske, 2024-11-01:
I'm experiencing this as well, in my case on an Orange Pi 5 running Debian 11 + supervised HA install, esphome version 2024.10.2
I found this: https://www.tal.org/tutorials/booting-64-bit-kernel-raspberry-pi-3
which notes that there are some arm64 kernel issues, and offers a way to silence the warnings.

dothebart, 2024-11-22:
so basically native 64 bit ARM docker containers are missing?
(obviosly got flooded by those too)