esphome/issues #6230: MQTT discovery is false still tries to subscribe to topics

martijnvwezel, 2024-09-06:
### The problem

I want to connect with mqtt broker that does not allow unspecified topics. If discovery_retain: false and discovery are on false still the topic is being tried to subscribe too.

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Not in HA an issue

### What platform are you using?

ESP32-IDF

### Board

esp32-c3 seeeduino

### Example YAML snippet

```yaml
mqtt:
  discovery: false
  discovery_retain: false
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  client_id: !secret mqtt_client_id
  idf_send_async: true
  port: 8883
  skip_cert_cn_check: true  
  log_topic: 
    topic: "v1/projects/2344324366342263/measurements"
    level: ERROR
  keepalive: 10s            # Keep alive interval in seconds
  topic_prefix: "v1/projects/2344324366342263/measurements"
  discovery_prefix: null  
  birth_message:
  will_message:
```

ssieb, 2024-09-06:
Which topic is it subscribing to?  Or do you mean publishing?

martijnvwezel, 2024-09-09:
It tries to publish the following topics

`
Esphome/ping/test_dev
Esphome/discover
#SYS/#
#`

martijnvwezel, 2024-09-09:
ahhh `discover_ip` I though this ment ip address but it actually true or false. Think this discovery_ip is a bit confusion. Disabled it and now the messages come in

closed 2024-09-09 by martijnvwezel

ssieb, 2024-09-09:
> discover_ip (Optional, boolean): If Home Assistant automatic device discovery should be enabled. Defaults to true.

It clearly says it's a boolean with a default of `true`.  It's for letting HA and the dashboard discover which IP address the device is at.