esphome/issues #6231: Voice Assistant with ESP3S3, no speaker sound

aviitzhaki, 2024-09-07:
### The problem

I am using esp32-s3-devkitc-1 connected to INMP441 Microphone and MAX98357A Audio Amplifier.
I am using the esp-idf to implement voice assistant with micro wake word.
Th emicrophone is workinf fine (responding to the wake word... see example.
I have no sound on the speaker. I checked the I2S signals and see the LRCLK and BCLK but no data,
I also see on the log file that there is an issue with the speaker task.
I have audio pipeline set correctly.

Did someone encountered an issue with the speaker?

### Which version of ESPHome has the issue?

ESPHome 2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.1

### What platform are you using?

ESP32-IDF

### Board

esp32-s3-devkitc-1

### Component causing the issue

I2S speaker

### Example YAML snippet

```yaml
esphome:
  name: voiceassist
  friendly_name: VoiceAssist
  platformio_options:
    board_build.flash_mode: dio
  on_boot:
    - light.turn_on:
        id: led_ww
        blue: 100%
        brightness: 60%
        effect: fast pulse

esp32:
  board:   esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB  
  framework:
    type: esp-idf
    version: 4.4.6
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_AUDIO_BOARD_CUSTOM: "y"
   
psram:
  mode: octal # Please change this to quad for N8R2 and octal for N16R8
  speed: 80MHz

# Enable Home Assistant API
api:
  encryption:
[... 292 lines cut ...]
[20:39:31][D][voice_assistant:719]: Assist Pipeline ended
[20:39:31][D][voice_assistant:504]: State changed from AWAITING_RESPONSE to IDLE
[20:39:31][D][voice_assistant:510]: Desired state set to IDLE
[20:39:31][D][esp-idf:000][speaker_task]: I (5962148) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=8

[20:39:31][D][light:036]: 'On board light' Setting:
[20:39:31][D][light:047]: State: OFF
[20:39:31][D][light:085]: Transition length: 1.0s
[20:39:31][D][light:036]: 'Led Strip' Setting:
[20:39:31][D][light:047]: State: OFF
[20:39:31][D][light:085]: Transition length: 1.0s
[20:39:31][D][micro_wake_word:399]: Resetting buffers and probabilities
[20:39:31][D][micro_wake_word:195]: State changed from IDLE to START_MICROPHONE
[20:39:31][D][micro_wake_word:107]: Starting Microphone
[20:39:31][D][micro_wake_word:195]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[20:39:31][D][i2s_audio.speaker:211]: Starting I2S Audio Speaker
[20:39:31][D][esp-idf:000]: I (5962206) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[20:39:31][D][micro_wake_word:195]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
[20:39:31][D][i2s_audio.speaker:214]: Started I2S Audio Speaker
[20:39:31][D][esp-idf:000][speaker_task]: I (5962251) I2S: DMA queue destroyed

[20:39:31][D][i2s_audio.speaker:218]: Stopping I2S Audio Speaker
[20:39:31][D][i2s_audio.speaker:230]: Stopped I2S Audio Speaker
```

Coldness00, 2024-09-11:
Hello,
I had same issue, I fixed it by creating a media_player entity:
```
media_player:
  - platform: i2s_audio
    id: speaki2s
    i2s_audio_id: i2s_out
    name: ESPHome VA Media Player
    dac_type: external
    i2s_dout_pin: GPIO16
    mode: mono
    on_pause:
      - media_player.stop
```
      

Then in voice assistant, I do not define a speaker.
But I tell him to use the media player

```
voice_assistant:
  id: va
  microphone: mici2s
  use_wake_word: true
  noise_suppression_level: 4
  auto_gain: 31dBFS
  volume_multiplier: 8.0
  on_tts_end:
        - media_player.play_media:
            id: speaki2s
            media_url: !lambda 'return x;'
```

aviitzhaki, 2024-09-11:
Thanks 

media_player entity needs arduino platform 
micro_wake_word needs esp-idf platform

How can I resolve this?

Coldness00, 2024-09-11:
I use push to talk :/
Maybe try with speaker if it also works

aviitzhaki, 2024-09-15:
Thank you @Coldness00 

the issue was related to network setting 

The solution:
1. turn off Local network
2. set local network to https://homeassistant.localdomain

00xBAD, 2024-11-20:
this fixed my issue:

https://community.home-assistant.io/t/voice-assistant-no-sound/749258/3