esphome/issues #6236: Boot sequence hangs until MQTT client component connects

Alphaemef, 2024-09-11:
### The problem

If MQTT client is configured but can not connect to the server, components that load with a boot priority lower than the MQTT Component. 

I.e:
If both API and MQTT are connected, and MQTT disconnects API continues to work until the reboot teimout. At which point the ESP will startup without API. Once the MQTT connects, the API will connect. 

If API setup_priority is changed to 201, this alleviates the issue. Other components with lower priority do not load either, i.e modbus

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

esp32

### Component causing the issue

MQTT Client Component

### Example YAML snippet

```yaml
mqtt:
  broker: ${broker}
  port: ${port}
  username: ${name}
  password: ${pass}
  discovery_unique_id_generator: mac
  id: mqtt_client
  skip_cert_cn_check: false
  idf_send_async: false
  certificate_authority:  | 
    -----BEGIN CERTIFICATE-----
    -----END CERTIFICATE-----
```

victorclaessen, 2024-09-17:
I figured out later my problem in #4875 partly had the same cause. (MQTT could not connect because I had a wrong configuration of the wireguard component. Therefore the API port also did not open.)

One would really hope that these things would become independent.

Alphaemef, 2024-09-17:
> I figured out later that my problem in #4875 partly had the same cause. (MQTT could not connect because I had a wrong configuration of the wireguard component. Therefore the API port also did not open.)
> 
> One would really hope that these things would become independent.

yes, it shouldn't wait for "ok" from MQTT before proceeding with loading components with a lower priority.

victorclaessen, 2024-09-20:
I think your suggestion to increase the priority makes sense. If [this line](https://github.com/esphome/esphome/blob/8e5d7337c8e9490c901cd17f59b72ed300f775b1/esphome/components/mqtt/mqtt_client.cpp#L372) is changed to 
`float MQTTClientComponent::get_setup_priority() const { return setup_priority::AFTER_WIFI + 1; }`
`                                                                                         ----`
then the problem is likely to be solved.

ssieb, 2024-09-20:
How would that help?  That would only make things worse if anything.

victorclaessen, 2024-09-21:
Could you explain why? 
Our assumption is that this would make the mqtt component start *after* the api component. Is that not correct?

victorclaessen, 2024-09-21:
Ah, right, it should be ` - 1`. Good catch.

Alphaemef, 2024-09-25:
> Ah, right, it should be ` - 1`. Good catch.

While you are right, this doesn't actually solve the issue. 
Say some other component is loaded at priority -10, if MQTT doesn't connect immediately anything downstream does not work, which makes no sense since API for instance could be running, og if its an operation that doesn't even need MQTT / API.

victorclaessen, 2024-09-26:
You're right; but it would at least allow the API to start (I suppose), so you could interact with the device remotely?

victorclaessen, 2025-01-25:
It may be stale, but I, personally, would still be interested in having this fixed.