esphome/issues #6237: Native API, Service documentation missing

coxtor, 2024-09-11:
### The problem

Hi, not really a real issue but I was absolutely not able to find the services documentation in the native api component. Is it missing or am I searching wrong? Ultimately I knew this feature existed and figured out how to use it using discussion forum templates, but that shouldn't really be necessary.  The search field in the documentation also yields no results. 
If I'm wrong please point me to the documentation and close this issue =) . 

Thanks! 

### Which version of ESPHome has the issue?

irellevant 

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

ssieb, 2024-09-11:
They're called "actions" now.
https://esphome.io/components/api#user-defined-actions
`services:` still works, but the documentation should probably make it clearer that it was changed.

closed 2024-09-11 by ssieb

jesserockz, 2024-09-11:
@coxtor Yes it it the same. They are "device actions" now. That is exactly what is linked in the above comment

ssieb, 2024-09-11:
https://github.com/esphome/esphome-docs/pull/4244