esphome/issues #6238: Unable to flash ESP32-POE

MaxDucy, 2024-09-12:
### The problem

I'm trying to flash Olimex ESP32-POE to use as an IR Remote Climate. And I keep getting this log when I flashed with ESPHome Web

```invalid header: 0x20747375
invalid header: 0x20747375
invalid header: 0x20747375
invalid header: 0x20747375
invalid header: 0x20747375
invalid header: 0x20747375
invalid header: 0x20747375
invalid header: 0x20747375
ets Jul 29 2019 12:21:46

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.4

### What platform are you using?

ESP32

### Board

Olimex ESP32-POE

### Example YAML snippet

```yaml
esphome:
  name: daikin-server-room
  friendly_name: Daikin IR Server Room

esp32:
  board: esp32-poe
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "something"

ota:
    platform: esphome
    password: "somethingsomething"

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
  manual_ip:
    static_ip: 1xx.xx.x.xxx
    gateway: 1xx.xx.x.xxx
    subnet: xxx.xxx.xxx.0

web_server:
  
remote_transmitter:
  pin: GPIO2
  carrier_duty_percent: 50%

climate:
  - platform: daikin_arc
    name: "Daikin Server Room AC"
```

ssieb, 2024-09-12:
What file are you flashing?

MaxDucy, 2024-09-13:
I downloaded .bin manually in OTA format. I tried to download the factory format but nothing happen on both Chrome and Firefox.

nielsnl68, 2024-09-13:
Do you have the Arduino IDE installed? 
When so make, you could try this, it helped me a couple of times out.

Make in the arduino IDE a simple sketch like a blinking led. Try to install it and see what happens. Try to lower the upload speed to 115400 to see if it uploads at that speed. When it does succeed i was able to use the esphome web tool to upload the firmware again.

I hope this helps you.

ssieb, 2024-09-13:
That factory file is what you need.  What do you mean that nothing happened?

MaxDucy, 2024-09-18:
I think I found the problem. I forgot to enable https on my HA that's why when I try to flash ESP32 through ESPHome web and nothing happened. After I enabled https I was able to flash ESP. Now it's working.

Thank you.

closed 2024-09-18 by MaxDucy