esphome/issues #6240: UART does not get all bytes?

f3nix, 2024-09-12:
### The problem

Hi.

I'm converting a tuya device to esphome.
I stumbled accross a problem. I think not all bytes are read.

Here is a log output from my serial sniffer (received bytes):
```
55 aa 03 07 00 08 12 02 00 04 00 00 00 00 29
55 aa 03 07 00 08 13 02 00 04 00 00 00 00 2a
55 aa 03 07 00 08 14 02 00 04 00 00 00 02 2d
55 aa 03 07 00 08 7a 02 00 04 00 00 01 38 ca
55 aa 03 07 00 08 7b 02 00 04 00 00 00 00 92
55 aa 03 07 00 08 85 02 00 04 00 00 00 00 9c
55 aa 03 07 00 08 86 02 00 04 00 0f 42 3f 2d
55 aa 03 07 00 08 87 02 00 04 00 00 00 20 be
55 aa 03 07 00 08 67 02 00 04 00 00 00 00 7e
55 aa 03 07 00 08 69 02 00 04 00 00 02 f5 77
55 aa 03 24 00 00 26
55 aa 03 07 00 08 69 02 00 04 00 00 02 f5 77
55 aa 03 24
00 00 26
```

But I only get these bytes in esphome:
```
55:AA:03:07:00:08:12:02:00:04:00:00:00:00:29:
55:AA:03:07:00:08:13:02:00:04:00:00:00:00:2A:
55:AA:03:07:00:08:14:02:00:04:00:00:00:02:2D:
55:AA:03:07:00:08:7A:02:00:04:00:00:01:38:CA:
55:AA:03:07:00:

                                 0F:42:3F:2D:
55:AA:03:07:00:08:87:02:00:04
```

```
[20:39:01][D][uart_debug:114]: <<< 55:AA:03:07:00:08:12:02:00:04:00:00:00:00:29:55:AA:03:07:00:08:13:02:00:04:00:00:00:00:2A:55:AA:03:07:00:08:14:02:00:04:00:00:00:02:2D:55:AA:03:07:00:08:7A:02:00:04:00:00:01:38:CA:55:AA:03:07:00:0F:42:3F:2D:55:AA:03:07:00:08:87:02:00:04
```

Esphome config (partial):
```
bk72xx:
  board: cb2s

logger:
  baud_rate: 0
  tx_buffer_size: 1024
  level: VERBOSE

uart:
  tx_pin: TX1
  rx_pin: RX1
  baud_rate: 115200
  rx_buffer_size: 1024
  debug:
    direction: BOTH
    dummy_receiver: false
```

Do you have any idea how to debug it further?
When I receive a shorter answer from tuya mcu evertyhing works as expected.

Cheers,
Mateusz

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

BK72XX

### Board

cb2s

### Anything in the logs that might be useful for us?

```txt
[20:39:01][D][uart_debug:114]: <<< 55:AA:03:07:00:08:12:02:00:04:00:00:00:00:29:55:AA:03:07:00:08:13:02:00:04:00:00:00:00:2A:55:AA:03:07:00:08:14:02:00:04:00:00:00:02:2D:55:AA:03:07:00:08:7A:02:00:04:00:00:01:38:CA:55:AA:03:07:00:0F:42:3F:2D:55:AA:03:07:00:08:87:02:00:04
```

ssieb, 2024-09-12:
Why are you setting the tx buffer on the logger when you've disabled serial logging?

But after checking, it looks like buffer sizes aren't supported at all on libretiny.

f3nix, 2024-09-12:
tx buffer is a remnant of me playing with some options.

I get this in the log:
```
[06:35:00][C][uart.lt:101]: UART Bus:
[06:35:00][C][uart.lt:102]:   Type: hardware
[06:35:00][C][uart.lt:104]:   Port number: 1
[06:35:00][C][uart.lt:106]:   TX Pin: 11
[06:35:00][C][uart.lt:107]:   RX Pin: 10
[06:35:00][C][uart.lt:109]:   RX Buffer Size: 1024
[06:35:00][C][uart.lt:111]:   Baud Rate: 115200 baud
[06:35:00][C][uart.lt:112]:   Data Bits: 8
[06:35:00][C][uart.lt:113]:   Parity: NONE
[06:35:00][C][uart.lt:114]:   Stop bits: 1
```

After changing to this:
```
bk72xx:
  board: cb2s
  framework:
    version: dev
    loglevel: debug
```

So it seems that the rx buffer size is changing. Am I missing something?

Cheers,
Mateusz

ssieb, 2024-09-12:
That's misleading.  It prints it in the log, but doesn't actually use it.

f3nix, 2024-09-13:
@kuba2k2 Care to weigh in?

kuba2k2, 2024-09-13:
The RX buffer is limited to 64 bytes by default, and there is currently no easy way to change it. So when the device sends more data than ESPHome is able to process in real-time, bytes could be dropped.

You can try modifying the file:

`.platformio/packages/framework-arduino-api/RingBuffer.h`

and change `SERIAL_BUFFER_SIZE` to something more than 64.

f3nix, 2024-09-16:
Hi.
To answer my own question.

* Fork https://github.com/libretiny-eu/ArduinoCore-API.git
* Make necessary changes
* Add
```
platformio_options:
      platform_packages:
      - framework-arduino-api @ PUT_GIT_REPOSITORY_HERE
```
to esphome section.

@ssieb Any idea how to clean the package repository? It does not get deleted with "clean build files". Nor does it update when new commits are added remotely.

Cheers,
Mateusz

ssieb, 2024-09-16:
Answer from discord:
> If you don't change the version / url in platformio it just keeps using the files that are already downloaded
> You have to delete them from the global platformio packages folder or put the git hash on the package line itself

f3nix, 2024-09-17:
@ssieb Thanks! :)

hn, 2025-01-09:
I have two Tuya devices which do not work correctly with a UART buffer of 64 bytes. Increasing it to 256 bytes (ESPHome default) helps.

For the impatient, the fix can simply be applied as follows:

```
esphome:
  platformio_options:
    platform_packages:
      - framework-arduino-api @ https://github.com/hn/ArduinoCore-API#RingBufferSize256
```

Alternatively, `RingBufferSize512` for 512 bytes size can be used. For one of my chatty TuyaMCUs this seems to be necessary, only with 512 bytes there are no serial CRC errors.

Together with https://github.com/libretiny-eu/libretiny/issues/154 (RingBuffer is not interrupt-safe, worked around by https://github.com/hn/ArduinoCore-API/commit/9c92661fdfb312d8a3a924b958c46ac398d4b657), this confirms to me that the Arduino RingBuffer is not suited well as a component for the LibreTiny UART @kuba2k2.

kuba2k2, 2025-01-09:
Feel free to PR the necessary changes to libretiny/ArduinoCore-API (including the "ringbuffer workaround", perhaps).

But for changing the RingBuffer's size, I think there could just be a typedef, since it's a template class:
```cpp
typedef RingBufferN<256> RingBuffer256;
```
added to LibreTiny in the main `Serial.h` header.

dshcherb, 2025-01-15:
Had a similar issue with [Tongou TO-Q-SYS-JWT](https://devices.esphome.io/devices/Tongou-TO-Q-SYS-JWT-power-meter#uart-buffer-size-adjustment) which needs 300+ bytes (I could only tell what's wrong by recording RX with a logical analyzer).

I'm going to test `LT_SERIAL_BUFFER_SIZE` and see if that does it instead of using a patched `ArduinoCore-API`.

Thanks for looking into making a proper fix!

dshcherb, 2025-01-15:
Tested on Tongou TO-Q-SYS-JWT (CBU) with Libretiny 1.8.0 - it seems to work and all data points are seen as expected.

```
esphome:
  name: to-q-sys-jwt
  name_add_mac_suffix: true
  # Commented out the patched ArduinoCore-API
  #  platform_packages:
  #    - framework-arduino-api @ https://github.com/dshcherb/ArduinoCore-API/archive/refs/tags/2025.1.4.zip

bk72xx:
  board: cbu
  framework:
    options:
      LT_SERIAL_BUFFER_SIZE: 512
    # The registry has 1.7.0 only at the time of writing: https://registry.platformio.org/platforms/kuba2k2/libretiny/versions
    # so build from source.
    version: "1.8.0"
    source: "https://github.com/libretiny-eu/libretiny"
```