esphome/issues #6242: ili9xxx broken on ESP32-S3-BOX since 2024.9.0b1

Kannix2005, 2024-09-13:
### The problem

Since the newest beta update the ili9xxx component is broken.

I first got the error
`
display.ili9xxx: [source /data/packages/dec89df9/esp32-s3-box-3/esp32-s3-box-3.yaml:688]
  
  'invert_colors' is a required option for [display.ili9xxx].`

Then i added this parameter, but now i only get a blank screen.

Nothing else was changed, so my guess is, that some bug has been introduced with the latest changes.

### Which version of ESPHome has the issue?

2024.9.0b1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32s3box

### Component causing the issue

ili9xxx

### Example YAML snippet

```yaml
display:
  - platform: ili9xxx
    id: s3_box_lcd
    model: S3BOX
    data_rate: 40MHz
    invert_colors: true
    cs_pin: 5
    dc_pin: 4
    reset_pin:
      number: 48
      inverted: true
    update_interval: never
```

### Anything in the logs that might be useful for us?

```txt
Everything boots up correctly. Even the touch screen works.
```

Kannix2005, 2024-09-13:
NVM my S3 Box is defective

closed 2024-09-13 by Kannix2005

dj-sf, 2024-09-18:
@Kannix2005 what did the issue end up being with your S3 Box? im having the same thing happen

ssieb, 2024-09-19:
https://github.com/esphome/wake-word-voice-assistants/issues/18

ssieb, 2024-09-19:
@dj-sf the compile error or the screen not working?

dj-sf, 2024-09-19:
Oh the compile error. However I was able to find a fix -- just needed to update some yaml files in the firmware config.  PR is in review now.  Thx though!

Kannix2005, 2024-09-19:
If anybody of you knows how to debug/recover the display, i´d be glad to hear. I don´t really want to trash this unit.

ssieb, 2024-09-19:
Come ask on the esphome discord server.