esphome/issues #6243: Vector data gets altered on script.execute

escoand, 2024-09-13:
### The problem

I've some strange problems with `script.execute` which sometimes alters the data of a given vector.

Im not that deep into C/C++ to know if it's my fault or a wrong behaviour.
But in the most cases it's working without any problems.

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

lolin32_lite

### Component causing the issue

script

### Example YAML snippet

```yaml
- id: info
    parameters:
      cmd: uint32_t
      value: float
    then:
      - lambda: |-
          int16_t raw = value;
          std::vector<uint8_t> data = {(uint8_t)(raw >> 8), (uint8_t)raw, 0};
          // debug
          if(cmd == 0x2E3E0215 || cmd == 0x3E2E0215 || cmd == 0x2F3F0215 || cmd == 0x3F2F0215) {
            char str[512] = {};
            sprintf(str, "info cmd=0x%08X pre=%f post=%04X data=%02X:%02X:%02X", cmd, value, raw, data[0], data[1], data[2]);
            id(debug)->publish_state(str);
          }
          id(write_msg)->execute($bsb_general, 0x00, 0x02, cmd, data);
  - id: write_msg
    mode: queued
    parameters:
      src: uint8_t
      dst: uint8_t
      type: uint8_t
      cmd: uint32_t
      data: uint8_t[]
    then:
      - lambda: |-
          // debug
          if(cmd == 0x2E3E0215 || cmd == 0x3E2E0215 || cmd == 0x2F3F0215 || cmd == 0x3F2F0215) {
            char str[512] = {};
            sprintf(str, "write_msg cmd=0x%08X data=", cmd);
            for(size_t i = 0; i < data.size(); i++) {
              sprintf(str + strlen(str), "%02X:", data[i]);
            }
            str[strlen(str) - 1] = '\0';
            id(debug)->publish_state(str);
          }
          // ...
```

### Anything in the logs that might be useful for us?

```txt
# from HomeAssistant
# as you can see the info script shows the correct data but write_message something different
# but all the primitive variables are unchanged
Debug wechselte zu info cmd=0x2F3F0215 pre=1209.599976 post=04B9 data=04:B9:00
10:59:23 - Vor 28 Minuten
Debug wechselte zu write_msg cmd=0x2F3F0215 data=D4:7E:FB
10:59:23 - Vor 28 Minuten

# here a good run:
Debug wechselte zu info cmd=0x2F3F0215 pre=1203.199951 post=04B3 data=04:B3:00
10:59:39 - Vor 30 Minuten
Debug wechselte zu write_msg cmd=0x2F3F0215 data=04:B3:00
10:59:39 - Vor 30 Minuten
```

ssieb, 2024-09-13:
You're passing a vector to an array parameter.  I'm not sure how that works at all.  And you need to make sure it's copied, not passed by reference since it's a local variable.
github issues are not for support requests.  Come ask on the esphome discord server.

closed 2024-09-13 by ssieb

escoand, 2024-09-13:
[image]

Is it me (together with the docs) or is it you being wrong at this point?

ssieb, 2024-09-13:
Ah, ok.  I missed that it converts the argument types, so that's why it works.  Still, this is not the right place.