esphome/issues #6244: ESP32-C3 (Shelly 1 Mini G3) not receiving OTA update

carlo161, 2024-09-13:
### The problem

Some months ago (esphome docker based version 2024.3.1) I installed a firmware in my device.
Yesterday I updated to the newest version (2024.8.3) and I tried to update it without success.
After a lot of tests I was able to install Tasmota via serial interface on it (tasmota32c3 version 14.2) and via OTA I installed the bin generated vie ESPHome Dashboard.
Now the device is online, I can read the logs but when I try to change the firmware I received the same error.

Installation log:
```
INFO ESPHome 2024.8.3
INFO Reading configuration /config/study-light.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing study-light (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Reading CMake configuration...
No dependencies
RAM:   [=         ]   8.5% (used 27876 bytes from 327680 bytes)
Flash: [=====     ]  45.1% (used 827078 bytes from 1835008 bytes)
========================= [SUCCESS] Took 9.47 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.16.29
INFO Uploading /config/.esphome/build/study-light/.pioenvs/study-light/firmware.bin (831888 bytes)
ERROR Error binary size: Unknown error from ESP
```

I cannot understand if I am doing something wrong, I also tried to return to the older version of esphome but the behavior is the same. I tried also to cut the yaml but it does not change anything.
What could be the issue?

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Shelly 1 Mini G3

Jason2866, 2024-09-13:
Actual Tasmota can not be converted via OTA to esphome. Updates will not work. Just what you see. You have to flash esphome via serial.

carlo161, 2024-09-13:
I tested it and it works fine, thanks.
So the only way now is flashing via the serial interface? Or it is viable to use a Tasmota version < 12?
Do you know if is possible to upload the esphome firmware via OTA in the web interface of the standard Shelly?

Jason2866, 2024-09-13:
Yes, only serial flashing is the possible way to get esphome on Shelly devices.

closed 2024-09-14 by carlo161

aceindy, 2024-11-13:
It works OTA also:
https://community.home-assistant.io/t/esphome-ota-shelly-1-pm-mini-g3-guide/794350

aceindy, 2024-12-30:
Until this [PR](https://github.com/esphome/esphome/pull/5535) is merged, the only way to update is by serial wire.
(and if you had it flashed before this PR, you're still required to flash by serial before OTA will work again :?