esphome/issues #6247: Cover opens/closes every reboot.

AlexMKX, 2024-09-16:
### The problem

While using either feedback or timebased cover to control the relays, cover opens and closes every reboot, which causes : 
1.  Security issues (e.g. someone can get in).
2. Possibility to close someone outside.

### Which version of ESPHome has the issue?

ESPHome version 2024.8.3 compiled on Sep 15 2024, 17:25:37

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.1

### What platform are you using?

ESP32

### Board

es32a08

### Component causing the issue

Cover

### Example YAML snippet

```yaml
cover:
  - platform: feedback
    id: gate_control
    name: "Gate Cover"
    device_class: gate
    open_duration: 30s
    close_duration: 30s
    has_built_in_endstop: True
    open_endstop: in_7
    close_endstop: in_6
    stop_action: 
      then:
        - delay: 500ms
    open_action:
      - switch.turn_on: relay_7
      - delay: 500ms
      - switch.turn_off: relay_7
    close_action:
      - switch.turn_on: relay_8
      - delay: 500ms
      - switch.turn_off: relay_8
```

ssieb, 2024-09-16:
Does it do that without the cover?

AlexMKX, 2024-09-16:
> Does it do that without the cover?

it is not. When I comment out the cover - everything works as it should - e.g. doing nothing after reboot.

ssieb, 2024-09-16:
Can you get serial logs from boot?

AlexMKX, 2024-09-16:
[logs_gate_run.txt](https://github.com/user-attachments/fil
[es32a08.yaml.txt](https://github.com/user-attachments/files/17010309/es32a08.yaml.txt)
[qdy30a.yaml.txt](https://github.com/user-attachments/files/17010310/qdy30a.yaml.txt)
es/17010276/logs_gate_run.txt)
[gate.yaml.txt](https://github.com/user-attachments/files/17010307/gate.yaml.txt)

Definetely yes.
Kindly find the log attached and full config for this node.

AlexMKX, 2024-09-16:
It seems I've found the root cause:
```yaml
  - platform: template
    id: gate_button
    name: Gate Button 

    lambda: |-
      return (id(in_8).state);

    on_click: 
      then:
        - logger.log: "On Click"
        - cover.open: gate_control
        - delay: 30s
        - cover.close: gate_control
        
```

When commenting this - no open/close on reboot.

ssieb, 2024-09-16:
Right, I didn't think it could be the cover.  That wasn't **serial** logs, but you found the problem anyway.
Btw, the open and close commands don't wait for it to finish, which it looks like you're expecting there.

closed 2024-09-16 by ssieb