esphome/issues #6248: Build esp32-c6 with esp32_ble_tracker and bluetooth_proxy

straccio, 2024-09-16:
### The problem

Unable to build with errors
```
In file included from src/esphome/components/esp32_ble/ble.cpp:6:
src/esphome/components/esp32_ble/const_esp32c6.h:62:1: error: 'const esp_bt_controller_config_t' has no non-static data member named 'enable_uart_hci'
   62 | };
      | ^
Compiling .pioenvs/camerettasensor-5cf4ec/src/esphome/components/esp32_ble_client/ble_characteristic.cpp.o
*** [.pioenvs/camerettasensor-5cf4ec/src/esphome/components/esp32_ble/ble.cpp.o] Error 1
```

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.1

### What platform are you using?

ESP32

### Board

Waveshare ESP32-C6 

### Component causing the issue

esp32_ble

### Example YAML snippet

```yaml
esphome:
  name: bletracker-1
  friendly_name: bletracker-1

esp32:
  board: esp32-c6-devkitm-1
  variant: esp32c6
  flash_size: 4MB
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.8.1
    sdkconfig_options:
        CONFIG_OPENTHREAD_ENABLED: n
        CONFIG_ENABLE_WIFI_STATION: y
        CONFIG_USE_MINIMAL_MDNS: n
        CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

esp32_ble_tracker:
  scan_parameters:
    active: false

bluetooth_proxy:
  active: false
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/esp32_ble/ble.cpp:6:
src/esphome/components/esp32_ble/const_esp32c6.h:62:1: error: 'const esp_bt_controller_config_t' has no non-static data member named 'enable_uart_hci'
   62 | };
      | ^
Compiling .pioenvs/camerettasensor-5cf4ec/src/esphome/components/esp32_ble_client/ble_characteristic.cpp.o
*** [.pioenvs/camerettasensor-5cf4ec/src/esphome/components/esp32_ble/ble.cpp.o] Error 1
```

### Additional information

I have commented these rows from esphome/components/esp32_ble/const_esp32c6.h in order to build and run correctly on this board.
```
pragma once

#ifdef USE_ESP32_VARIANT_ESP32C6

#include <esp_bt.h>

namespace esphome {
namespace esp32_ble {

static const esp_bt_controller_config_t BT_CONTROLLER_CONFIG = {
    ....
    .enable_qa_test = RUN_QA_TEST,
    .enable_bqb_test = RUN_BQB_TEST,
    // .enable_uart_hci = HCI_UART_EN,
    // .ble_hci_uart_port = DEFAULT_BT_LE_HCI_UART_PORT,
    // .ble_hci_uart_baud = DEFAULT_BT_LE_HCI_UART_BAUD,
    // .ble_hci_uart_data_bits = DEFAULT_BT_LE_HCI_UART_DATA_BITS,
    // .ble_hci_uart_stop_bits = DEFAULT_BT_LE_HCI_UART_STOP_BITS,
    // .ble_hci_uart_flow_ctrl = DEFAULT_BT_LE_HCI_UART_FLOW_CTRL,
    // .ble_hci_uart_uart_parity = DEFAULT_BT_LE_HCI_UART_PARITY,
    .enable_tx_cca = DEFAULT_BT_LE_TX_CCA_ENABLED,
    .cca_rssi_thresh = 256 - DEFAULT_BT_LE_CCA_RSSI_THRESH,
    ...
};

}  // namespace esp32_ble
}  // namespace esphome

#endif  // USE_ESP32_VARIANT_ESP32C6
```

randybb, 2024-09-16:
As was mentioned multiple times, if you are overriding version, you are on your own. Currently esphome supports only IDF 4.4, so C6 is not officially supported as it needs IDF 5.1, even there were some changes made to support it.

ESP-IDF 5.1 support will come, maybe this year https://github.com/esphome/esphome/pull/7419, but the way to 5.3 is very long.

If you want to bring such support and fix it, fell free to help with PR's

rvalitov, 2024-10-10:
@straccio Thank you! Great work! I also had to modify the code the same way to make it work and made a PR so other owners of ESP32-C6 won't face this issue anymore.

straccio, 2024-10-10:
Thank to you, 
i have abandoned the ESP32-C6 because the i2c does not work and i don't have time to work on it!

rvalitov, 2024-10-10:
That's strange. I don't need I2c yet but I saw plenty examples of it working with ESP32-C6. Even in Zigbee mode.

straccio, 2024-10-10:
With Arduino 3.0.x i2c works, but not in esphome with latest idf that have new i2c implementation.
The old one is named legacy and does not work OTB with esphome.

closed 2024-10-17 by jesserockz