esphome/issues #6249: rtttl.stop command no longer stops rtttl playing

galvheim, 2024-09-16:
### The problem

When starting to play a rtttl melody it has, since many versions ago, not been possible to stop the ongoing tune by calling the rtttl.stop command. 
It simply doesnt stop the tune. For a while it was possible to send a mute rtttl.play command to make a melody stop that way, but now even that is not possible, so please make the rtttl.stop command work as normal again. 

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.1

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

RTTTL

### Additional information

I am just purely sending a service xall in automations:
rtttl.stop while it is playing. Worked before. Like 6-8 months ago.

ssieb, 2024-09-16:
yaml? logs?

galvheim, 2024-09-17:
Updated case with code I try to send and the log of the actual ESP-device.

nielsnl68, 2024-09-17:
Please show the whole RTTTL setup, please. So we can test it ourselfs.

galvheim, 2024-09-17:
This is the whole YAML.
It is playing back fine using this string:
`      - data:
          song_str: >-
            rtttl.play:
            annoying:'d=4,o=5,b=112:8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7,8e7'
        action: esphome.steinar_detektor_play_rtttl
        enabled: true`
        
ESP YAML-Code:
        
`esphome:
  name: "steinar-detektor"
  friendly_name: ESP32 - KattAlarm
  on_boot: 
      then:
      - light.turn_on:
          id: status_led_rgb
          brightness: 50%
          green: 50%
          red: 0%
          blue: 0%

esp32:
  board: esp32dev
  framework:
    type: arduino
    version: latest

# Enable logging
logger:

# Enable Home Assistant API
api:
  services:
    - service: play_rtttl
      variables:
        song_str: string
      then:
        - rtttl.play:
            rtttl: !lambda 'return song_str;'

ota:
  platform: esphome

web_server:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

captive_portal:

button:
  # Restart the ESP
  - platform: restart
    name: "Restart - KattAlarm"
switch:
  - platform: gpio
    pin: 33
    inverted: true
    id: relay
    name: "Lyd-Modul"
    icon: "hass:alert"
    on_turn_on:
    - delay: 50ms
    - switch.turn_off: relay

binary_sensor:
  - platform: gpio
    pin: 32
    name: "PIR Sensor"
    device_class: motion
    filters:
      - delayed_on: 3s
      - delayed_off: 8s    

light:
  - platform: neopixelbus
    variant: WS2812
    pin: 14
    num_leds: 1
    id: status_led_rgb
    type: RGB
    name: "Status LED"
    icon: "mdi:led-outline"

    effects:
      - lambda:
          name: Randon Colours
          update_interval: .5s
          lambda: |-
            auto call = id(status_led_rgb).turn_on();
            call.set_transition_length(500);
            call.set_rgb(random(10)*.1, random(10)*.1, random(10)*.1);
            call.perform();
      - lambda:
          name: Randon Colours Slow
          update_interval: 2s
          lambda: |-
            auto call = id(status_led_rgb).turn_on();
            call.set_transition_length(1000);
            call.set_rgb(random(10)*.1, random(10)*.1, random(10)*.1);
            call.perform();
      - pulse:
          name: "Fast Pulse"
          transition_length: 0.1s
          update_interval: 0.2s
          min_brightness: 10%
          max_brightness: 100%
      - pulse:
          name: "Slow Pulse"
          # transition_length: 1s      # defaults to 1s
          update_interval: 1.5s
          min_brightness: 10%
          max_brightness: 100%
      - flicker:
          name: Flicker Effect With Custom Values
          alpha: 95%
          intensity: 3%
      - addressable_twinkle:
          name: Twinkle Effect With Custom Values
          twinkle_probability: 5%
          progress_interval: 4ms
output:

  - platform: ledc
    pin: 19
    id: rtttl_out

rtttl:
  output: rtttl_out
  on_finished_playback:
    - logger.log: 'Song ended!'`

nielsnl68, 2024-09-17:
anyway could you test this out for me?

```yaml
logger:
  level: verbose
```
and play the music and call the rtttl.stop event.
and show the log result 
thanks

galvheim, 2024-09-17:
> To make sure i understand your issue correctly. When you press the `rtttl.stop` the last playing note is playing for every, is it not?
> 
> I never tried the stop option so it slipped my intention.

It behaved like that when the stop-command first started to have issues. But I was able to mitigate it by sending an rtttl.play and then a mute note. 

I will try what you ask.

galvheim, 2024-09-17:
Here I tell it to play. Two seconds later I send the stop-action, but as you can see it totally ignores it. Its like it is blocked from doing anything else while doing the playback. 

```
[18:22:21][D][rtttl:058]: Playing song rtttl.play
[18:22:21][D][rtttl:388]: State changed from STATE_STOPPED to STATE_RUNNING
[18:22:21][V][ledc.output:055]: Calculating resolution bit-depth for frequency 2637.000000
[18:22:21][V][ledc.output:060]: Resolution calculated as 14
[18:22:21][V][ledc.output:204]: Using Arduino API - Trying to define channel, frequency and bit depth...
[18:22:21][V][ledc.output:211]: Trying initialize channel 0 with frequency 2637.0 and bit depth of 14...
[18:22:21][V][ledc.output:216]: Configured frequency: 2637 with bit depth: 14
[18:22:21][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:21][V][ledc.output:119]: Setting duty: 9830 on channel 0
[18:22:21][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:21][V][ledc.output:055]: Calculating resolution bit-depth for frequency 2637.000000
[18:22:21][V][ledc.output:060]: Resolution calculated as 14
[18:22:21][V][ledc.output:204]: Using Arduino API - Trying to define channel, frequency and bit depth...
[18:22:21][V][ledc.output:211]: Trying initialize channel 0 with frequency 2637.0 and bit depth of 14...
[18:22:21][V][ledc.output:216]: Configured frequency: 2637 with bit depth: 14
[18:22:21][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:21][V][ledc.output:119]: Setting duty: 9830 on channel 0
[18:22:21][W][component:237]: Component rtttl took a long time for an operation (59 ms).
[18:22:21][W][component:238]: Components should block for at most 30 ms.
[18:22:21][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:21][V][ledc.output:055]: Calculating resolution bit-depth for frequency 2637.000000
[18:22:21][V][ledc.output:060]: Resolution calculated as 14
[18:22:21][V][ledc.output:204]: Using Arduino API - Trying to define channel, frequency and bit depth...
[18:22:21][V][ledc.output:211]: Trying initialize channel 0 with frequency 2637.0 and bit depth of 14...
[18:22:21][V][ledc.output:216]: Configured frequency: 2637 with bit depth: 14
[18:22:21][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:21][V][ledc.output:119]: Setting duty: 9830 on channel 0
[18:22:21][W][component:237]: Component rtttl took a long time for an operation (67 ms).
[18:22:21][W][component:238]: Components should block for at most 30 ms.
[18:22:22][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:22][V][ledc.output:055]: Calculating resolution bit-depth for frequency 2637.000000
[18:22:22][V][ledc.output:060]: Resolution calculated as 14
[18:22:22][V][ledc.output:204]: Using Arduino API - Trying to define channel, frequency and bit depth...
[18:22:22][V][ledc.output:211]: Trying initialize channel 0 with frequency 2637.0 and bit depth of 14...
[18:22:22][V][ledc.output:216]: Configured frequency: 2637 with bit depth: 14
[18:22:22][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:22][V][ledc.output:119]: Setting duty: 9830 on channel 0
[18:22:22][W][component:237]: Component rtttl took a long time for an operation (65 ms).
[18:22:22][W][component:238]: Components should block for at most 30 ms.
[18:22:22][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:22][V][ledc.output:055]: Calculating resolution bit-depth for frequency 2637.000000
[18:22:22][V][ledc.output:060]: Resolution calculated as 14
[18:22:22][V][ledc.output:204]: Using Arduino API - Trying to define channel, frequency and bit depth...
[18:22:22][V][ledc.output:211]: Trying initialize channel 0 with frequency 2637.0 and bit depth of 14...
[18:22:22][V][ledc.output:216]: Configured frequency: 2637 with bit depth: 14
[18:22:22][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:22][V][ledc.output:119]: Setting duty: 9830 on channel 0
[18:22:22][W][component:237]: Component rtttl took a long time for an operation (57 ms).
[18:22:22][W][component:238]: Components should block for at most 30 ms.
[18:22:22][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:22][V][ledc.output:055]: Calculating resolution bit-depth for frequency 2637.000000
[18:22:22][V][ledc.output:060]: Resolution calculated as 14
[18:22:22][V][ledc.output:204]: Using Arduino API - Trying to define channel, frequency and bit depth...
[18:22:22][V][ledc.output:211]: Trying initialize channel 0 with frequency 2637.0 and bit depth of 14...
[18:22:22][V][ledc.output:216]: Configured frequency: 2637 with bit depth: 14
[18:22:22][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:22][V][ledc.output:119]: Setting duty: 9830 on channel 0
[18:22:22][W][component:237]: Component rtttl took a long time for an operation (65 ms).
[18:22:22][W][component:238]: Components should block for at most 30 ms.
[18:22:23][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:23][V][ledc.output:055]: Calculating resolution bit-depth for frequency 2637.000000
[18:22:23][V][ledc.output:060]: Resolution calculated as 14
[18:22:23][V][ledc.output:204]: Using Arduino API - Trying to define channel, frequency and bit depth...
[18:22:23][V][ledc.output:211]: Trying initialize channel 0 with frequency 2637.0 and bit depth of 14...
[18:22:23][V][ledc.output:216]: Configured frequency: 2637 with bit depth: 14
[18:22:23][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:23][V][ledc.output:119]: Setting duty: 9830 on channel 0
[... 529 lines cut ...]
[... cut ...]
[18:22:40][V][ledc.output:119]: Setting duty: 9830 on channel 0
[18:22:40][W][component:237]: Component rtttl took a long time for an operation (66 ms).
[18:22:41][W][component:238]: Components should block for at most 30 ms.
[18:22:41][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:41][V][ledc.output:119]: Setting duty: 0 on channel 0
[18:22:41][D][rtttl:388]: State changed from STATE_RUNNING to STATE_STOPPED
[18:22:41][D][main:179]: Song ended!
[18:22:41][D][rtttl:364]: Playback finished
```

nielsnl68, 2024-09-17:
at which time did you executed the rtttl.stop?

galvheim, 2024-09-18:
> at which time did you executed the rtttl.stop?

As I wrote, about 2 seconds later. After this I tried again another time and sent the stop-command about 5 times throughout the playback, but nothing shows in the logs. So it totally igneores it, until it is finished, then it will come with the red error saying missing ":" as you see above. 
But to be clear: Stop-command works with your setup?

ssieb, 2024-09-18:
How are you executing the stop?  There's nothing in that config that would do that.

galvheim, 2024-09-18:
> How are you executing the stop? There's nothing in that config that would do that.

You wouldn't execute stop in the config. I execute it both as an action in automations and as events in developertools, as described in current documentation:

```
action: esphome.steinar_detektor_play_rtttl
data:
  song_str: rtttl.stop
```

randybb, 2024-09-18:
so basically executing
```yaml
api:
  services:
    - service: play_rtttl
      variables:
        song_str: string
      then:
        - rtttl.play:
            rtttl: !lambda 'return "rtttl.stop";'
```
This was working? It totally doesn't make sense.

ssieb, 2024-09-18:
Invalid data.  I guess that would stop it.
But that is definitely not described in the documentation.  You have to use the stop action.
https://esphome.io/components/rtttl#rtttl-stop-action

closed 2024-09-18 by ssieb

galvheim, 2024-09-18:
How is this hard to understand? What Mr. Randybb is coming with is definently not what I am doing. 
I start a melody using the rtttl.play and then the song. Easy. 
I then execute the rtttl.stop action using the code:
```
action: esphome.steinar_detektor_play_rtttl
data:
  song_str: rtttl.stop
  ```
That does not work, and it should not be difficult to understand it. Whatever Randybb here comes with, is not what I describe. 
What is unclear on your part?

galvheim, 2024-09-18:
> so basically executing
> 
> ```yaml
> api:
>   services:
>     - service: play_rtttl
>       variables:
>         song_str: string
>       then:
>         - rtttl.play:
>             rtttl: !lambda 'return "rtttl.stop";'
> ```
> 
> This was working? It totally doesn't make sense.

[image]

randybb, 2024-09-18:
You are sending "rtttl.stop" as song_str instead of a valid RTTTL string!

Basically you need to create another action and execute it instead of sending a wrong string.

```yaml
api:
  services:
    - service: stop_rtttl
      then:
        - rtttl.stop
```

galvheim, 2024-09-18:
> You are sending "rtttl.stop" as song_str instead of a valid RTTTL string!
> 
> Basically you need to create another action and execute it instead of sending a wrong string.
> 
> ```yaml
> api:
>   services:
>     - service: stop_rtttl
>       then:
>         - rtttl.stop
> ```
Cool! But how are mere mortals going to know this? I have only followed the documentation and from what I can see there is no mentioning or examples of this code. Can the devs, see that this is a problem? 
Why was it not necessary before? My code has stayed untouched since I first installed it 1 year ago.

galvheim, 2024-09-18:
I regard myself as a very regular user, and this is NOT obvious at all, until one really think about it. Why not show the whole code in the example as in:
```
api:
  services:
    - service: play_rtttl
      variables:
        song_str: string
      then:
        - rtttl.play:
            rtttl: !lambda 'return song_str;'
    - service: stop_rtttl
      then:
        - rtttl.stop
```
Then nobody will encounter this issue, because this is definently not clear enough, and you are the very first in all forums ever to mention this. So not even Mr. ssieb, was not able to identify this. So if he can not see it, how is the average user who wants to use this fuction supposed to know? Maybe try listen to the users, as you might see yourself blind on code. But most others not hav time to go very deep into this, and mostly hope to get example code that works. 

I don't know what crashed, but certainly my ESP did not crash since it is holding other states and doing other things besides playing rtttl, and all this it was holding. But yes, it might have crashed the rtttl.

galvheim, 2024-09-18:
Now it seems to work. So thank you for finally. As the first person in the world in many forums, you where able to guide me in the right direction.

[image]

randybb, 2024-09-18:
He was not able to spot it, because you did not provided your config and once you provided it with some logs you did not provided the way how you are stopping it. Thus he mentioned there is nothing related to stop in your config. Once you provided all details it was pretty clear where the issue is :) I was just quicker...

But agree - When I started with esphome a very long time I never needed any help as the documentation (I am not ESPHome dev, just a standard user). Even I thought it is perfect and clear some people still had problems - so there might be a space for improvements.