esphome/issues #6250: The Direction of Currents

KaitsuN6, 2024-09-16:
### The problem

Hello,

I live in Finland with 3x230V el.power.I am using the 6-channel Energy Meter with ESPHome(dev).

My board is a vanilla 6 Channel Board 1.4V rev1. I am measuring 3 phase Solar power on ic1 phase_a - phase_c. Ic2 phase_4 - phase_6 are on the main Grid leads. My CTs are all "20A/25mA Micro Current Transformer YHDC SCT-006".

My problem is, that my Current values are always positive no matter which way I turn my CTs. I really need to know the real current direction because I can not rely on any of the board entities (power, power_factor etc.) showing correct direction of the current. The given solution(s) like "...the solution is to turn around the current transformer on the wire..." do not make any difference.

Also a peculiar thing is, that when I was using the card calculated power, I noticed that ic1 phase_a and ic2 phase_a were opposite directions to their companions on same ic (CT opposite way). It was easy to note this on the Solar Power, as the power goes only one way. After that the Grid Power anomaly was easy to detect.

I bought this board as I thought I could do a total Realtime Energy Display. But to my big disappointment, when this did not work, reading the fine print told me to buy an add-on board to be able to do this. Then I would have a main board with 6 used channels and an add-in with no connected CTs. Now if I would buy this add-on separately, the freight cost is bigger than the value of the new card.

Will provide additional data if asked.

### Which version of ESPHome has the issue?

2024.10.0-dev20240916

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.1

### What platform are you using?

ESP32

### Board

6 Channel Board 1.4V rev1

### Component causing the issue

platform: atm90e32

### Anything in the logs that might be useful for us?

```txt
[W][atm90e32:418]: SPI power factor read error.
```

### Additional information

All said in the description...

randybb, 2024-09-16:
You don't need two transformers. Just one for each phase and then use `forward_active_energy:` and `reverse_active_energy:` for measuring direction of current https://esphome.io/components/sensor/atm90e32.html

KaitsuN6, 2024-09-17:
> You don't need two transformers. Just one for each phase and then use `forward_active_energy:` and `reverse_active_energy:` for measuring direction of current https://esphome.io/components/sensor/atm90e32.html

Hello, I don't see a connection to my problem:
- "...my Current values are always positive no matter which way I turn my CTs..."

Also "...I can not rely on any of the board entities (power, power_factor etc.)..."the etc. including **forward_active_energy** and **reverse_active_energy**. See picture (Verkko = Grid):
In "Verkko LX Watts (Calc)" i used the corresponding power_factor in the calculation.

[image]

According to my "_Acrel DTSD1352-CT/C three phases high precision smart meter_", that I have connected to the grid main leads, for the time of the picture all Watts were positive.

randybb, 2024-09-17:
atm90e32 is reading current flowing through CT, that value is stored in a register and this esphome components reads it . As it is alternating current its direction is always bidirectional and CT reads basically just an absolute (thus positive) value. If it is forward or reverse power is detected by its DSP (based on direction of a wave), thus there are another two registers for active energy, which is again the component showing as it is. This component is not doing any calculations, just reading what it is related registers. This is how the atm90e32 works.

If you think it is not reading data from registers correctly, then we need a proof the value in that register is different than esphome is reading. Then this is a valid issue.
If some other power meter is interpreting data differently, it is not a problem of this component.

KaitsuN6, 2024-09-18:
I have never seen an atm90e32 before in my life.
I am here asking for help, not a doctorate debate or prooving anything.
See enclosed pictures: Grid_Fed shows zero between 4 PM and about 8 PM. Grid_Reverse shows only zero.

As an authority on atm90e32 you sure have some testing code to give me, so I can run it in ESPhome and publish/send to you.

[image]
[image]

CircuitSetup, 2024-09-18:
@KaitsuN6 The CircuitSetup 6 Channel Meter uses 2 atm90e32 chips, each set up to read voltage and current on a single phase. To read the 3rd phase properly, you would need an add-on board.

If you have 3 phase power, you need an AC transformer for each of the 3 phases to measure their voltage. Without this, the meter cannot determine what direction current is flowing, and is likely reporting a very low power factor and power output.  More details on how to set that up is [located here.](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/tree/master?tab=readme-ov-file#measuring-3-phase-electricity)

However, you can read current only and calculate the power based on the voltage of a single phase. This won't give you the direction, though. Here is an [example of a config that calculates the power. ](https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter/blob/master/Software/ESPHome/6chan_energy_meter_v1.2.yaml)

FYI, you would only use `forward_active_energy` and `reverse_active_energy` for getting accumulated energy over a specific period of time. These will not be correct if the voltage and current on the corresponding phases are not being measured.

descipher, 2024-09-18:
@KaitsuN6 Thanks for reporting a possibly anomaly. The description @randybb has given you is accurate. We would need to see the YAML at this point to rule it out as a source of the issue. The component will publish forward and reverse energy as a float directly from the atm90e32. There is a store for cumulative reverse and forward energy that are stored as absolute values so it is a different path where we need to see the YAML.

KaitsuN6, 2024-09-20:
> @KaitsuN6 The CircuitSetup 6 Channel Meter uses 2 atm90e32 chips, each set up to read voltage and current on a single phase. To read the 3rd phase properly, you would need an add-on board.
>
> If you [...]
I needed a solution to measure my solar power(one direction of power) and my grid power (incoming and outgoing power). I thought I found it:
[image]
I understand that "_...Energy Meter..._" is something else than "_Measure current and calculate/speculate the rest_" (this is my opinion not a statement from anybody else).
Also reading the Readme file:
https://github.com/CircuitSetup/Expandable-6-Channel-ESP32-Energy-Meter
told me that this was the solution (fragments of the text displayed here):
 "_The Expandable 6 Channel ESP32 Energy Meter uses 2 ATM90E32AS ICs. Each IC has 3 voltage channels and 3 current channels._"..."_v1.4 removed JP8-JP11, and has voltage channels connected internally on the pcb._"

> FYI, you would only use `forward_active_energy` and `reverse_active_energy` for getting accumulated energy over a specific period of time. These will not be correct if the voltage and current on the corresponding phases are not being measured.
I used those only for the display.

For now, in the Verkko(Grid) power calculation/lambda I use **power_factor** to conclude if power is positive or negative.  I had to guess a limit value so the calculation would have right polarity (by experience without any Solar Power included. See line 406 in YAML I'll send to  @descipher soon. 
For Solar Power it is easy to use channel 1 **power_factor** as the direction of Power does not change. 

I am not going to buy any add-on boards, as I understand I would need more than one add-on board ($$$$$$$$$). I'll cut my losses with and try to manage with my current acquisition. 
There are easier, one stop solution, meters available else. Just that I need to dive into the worm can of MQTT in HA.

Thank You for trying to help me.

KaitsuN6, 2024-09-20:
> @KaitsuN6 Thanks for reporting a possibly anomaly. The description @randybb has given you is accurate. We would need to see the YAML at this point to rule it out as a source of the issue. The [...]

Hello! I understood Your request for the YAML. The rest is in a light blue haze floating before my eyes. :)
So, here is my YAML attached (as a text file because "We don’t support that file type.").

Thank You for trying to help me.
[kaitsun6_configuration.txt](https://github.com/user-attachments/files/17072283/kaitsun6_configuration.txt)

descipher, 2024-09-20:
@KaitsuN6 I reviewed the YAML, there are no issues I can see. @CircuitSetup has noted that you need to have a reference voltage for each of the phases A,B and C. The 6 channel board physically has two AC reference voltage inputs. In order to do the third without hacking the board you would need two 6 channel boards in total to provision a connection of the 3rd phase AC voltage reference input. Without that reference voltage the ATM90E32 IC will not have the 3rd phase information to calculate the directional flow. Then all three voltages can be fed to the ATM90E32 IC. 

The site documents this:

_European single phase 240V 50Hz (must provide [AC-AC transformer](https://learn.openenergymonitor.org/electricity-monitoring/voltage-sensing/different-acac-power-adapters) 9V or 12V with at least 500mA output)
3 phase - It is recommended to measure all 3 voltages with 3 voltage transformers. This can be done by using a main board with 1 add-on board (more information below). A single meter can be used to measure 3-phase, but power (wattage) will not be calculated correctly. Power can be calculated in software, but the [power factor](https://en.wikipedia.org/wiki/Power_factor) will have to be estimated ((voltage*current)*power_factor))._

KaitsuN6, 2024-09-22:
> @KaitsuN6 I reviewed the YAML, there are no issues I can see. 

Hello. So do you have any clue why **reversive_active_energy** is showing only 0, even if there should be readings.

CircuitSetup, 2024-09-22:
> > @KaitsuN6 I reviewed the YAML, there are no issues I can see.
> 
> Hello. So do you have any clue why **reversive_active_energy** is showing only 0, even if there should be readings.

The meter can't determine direction without measuring the voltage (with a voltage transformer) of the same phase the current transformer is measuring. 

The phases have to match, otherwise the energy readings will be wrong. Every phase is 120 degrees out from eachother in 3-phase systems. So, if you're measuring current on phase 3, but voltage from phase 1, the readings will be 240 degrees out from eachother, which will cause power factor, and energy (and phase angle) to be wrong. The meter is expecting those sine waves to be in phase with eachother, in other words.

descipher, 2024-09-22:
@KaitsuN6 

The ATM90E32 will calculate the "Active" energy in any direction based on an instantaneous point in time on each phase angle position for three value inputs which are phase current, voltage and angle. Normally with three phases you have 120 degrees of difference to calculate from, in this case the difference between two of the phases is always 0 degrees since they are the same voltage and current input references and that's why it calculates 0 active power on the L2 phase. It sees the same phase angle, voltage and current on that phase.

Does that makes sense?

closed 2025-02-04 by github-actions[bot]