esphome/issues #6251: 2024.8.3 breaks when upgrading from 2024.7.3 in Manual Install

curt7000, 2024-09-17:
### The problem

Running ESPHome just fine manually for years, upgrading via
`pip3.11 install -U esphome`

Following the latest pull, restarting via systemctl via
`sudo systemctl start esphome.service`

After installing, Dashboard will not load, receive the following when verifying service status;
`    Process: 3240061 ExecStart=/usr/local/bin/esphome /home/homeassistant/.homeassistant/esphome/config dashboard (code=exited, status=2)`

Rolling back to version 2024.7.3 resolves problem
`pip3.11 install -U --force-reinstall -v "esphome==2024.7.3"`

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.9.2

### What platform are you using?

ESP8266

### Board

N/A

### Component causing the issue

N/A

### Anything in the logs that might be useful for us?

```txt
After installing, Dashboard will not load, receive the following when verifying service status;
`    Process: 3240061 ExecStart=/usr/local/bin/esphome /home/homeassistant/.homeassistant/esphome/config dashboard (code=exited, status=2)`
```

### Additional information

Service Parameters are:

`[Unit]
Description=esphome
After=network.target

[Service]
Type=simple
User=homeassistant
ExecStart=/usr/local/bin/esphome /home/homeassistant/.homeassistant/esphome/config dashboard
RestartSec=30
Restart=on-failure

[Install]
WantedBy=multi-user.target`

ssieb, 2024-09-17:
You're using pip as root??  That's a great way to break your OS.
You'll need to find out more info about why it's not starting.

curt7000, 2024-09-17:
> You're using pip as root?? That's a great way to break your OS. You'll need to find out more info about why it's not starting.

Yes, unfortunately, necessary for my setup.  I've had it running this way for better part of 5+yrs, no issues.

curt7000, 2024-09-17:
Solved it!

When running 2024.7.3, reviewing the logs, I see;

Sep 17 03:02:52  systemd[1]: Started esphome.
Sep 17 03:02:53  esphome[631]: 2024-09-17 03:02:53,005 WARNING Calling ESPHome with the configuration before the command is deprecated and will be removed in the future.
Sep 17 03:02:53 esphome[631]: 2024-09-17 03:02:53,005 WARNING Please instead use:
**Sep 17 03:02:53 esphome[631]: 2024-09-17 03:02:53,005 WARNING    esphome dashboard /home/homeassistant/.homeassistant/esphome/config**

Changed my service parameters from;

[Service]
Type=simple
User=homeassistant
ExecStart=/usr/local/bin/esphome /home/homeassistant/.homeassistant/esphome/config dashboard
RestartSec=30
Restart=on-failure

to;

[Service]
Type=simple
User=homeassistant
**ExecStart=/usr/local/bin/esphome dashboard /home/homeassistant/.homeassistant/esphome/config**
RestartSec=30
Restart=on-failure

Now 2024.8.3 is working fine!  I see this was deprecated, but probably should have been noted in Breaking Changes for 2024.8.

closed 2024-09-17 by curt7000

ssieb, 2024-09-17:
https://esphome.io/changelog/2024.8.0.html#breaking-changes

curt7000, 2024-09-17:
> https://esphome.io/changelog/2024.8.0.html#breaking-changes The description could be more clear, but that command order has been deprecated for a very long time.

Thanks for highlighting this.  Agreed.  I read the changelog when this first occurred and this was not clear.