esphome/issues #6252: [SPI] On ESP32-S2 wrong naming of SPI devices

shibida, 2024-09-17:
### The problem

In the SPI component :
- when try to use "interface: spi2", it is translate as SPI device, instead of FSPI.
- when try to use "interface: spi3", it is translate as FSPI device, instead of HSPI.

This cause the problem of do not be able to use the real spi3 device on the ESP32-S2 boards.

### Which version of ESPHome has the issue?

2024.10.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.8

### What platform are you using?

ESP32

### Board

wemos S2 mini

### Component causing the issue

SPI

### Example YAML snippet

```yaml
spi:
  - id: spi_bus3
    clk_pin: GPIO39
    mosi_pin: GPIO37
    miso_pin: GPIO40
    interface: spi3
```

jesserockz, 2024-09-19:
Try using esp-idf as the framework instead of arduino

shibida, 2024-09-20:
I didn't try that cause, I made a custom sensor integrate an other library (based on arduino)
It was more simple to me to patch spi component to make it work for my case.

But it is to trace the fault.

closed 2025-02-03 by github-actions[bot]