esphome/issues #6253: Incorrect Data from MODBUS MAX485

ryjogo, 2024-09-17:
### The problem

I've had this working - then randomly it stopped:

```yaml
  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Battery SOH"
    register_type: read
    address: 1622
    register_count: 1
    value_type: U_WORD
    device_class: battery
    unit_of_measurement: "%"
    accuracy_decimals: 1
    filters:
      - multiply: 0.01
    icon: "mdi:battery-heart"
```

The register when queries using another application and a USB to Modbus adapter gives the correct reading. 

It should be 99 (%) however the value ESPHome gets 65535

[image]

Can this be a bug? I've tried to play around with the data types, but this didn't work. 

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.1

### What platform are you using?

ESP32

### Board

ESP32-C3 Super Mini 

### Component causing the issue

MAX485

### Example YAML snippet

```yaml
substitutions:
  name: "esphome-solplanet-modbus"
  friendly_name: ESPHome Solplanet Modbus

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.8.3
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0
  level: VERY_VERBOSE

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
# improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32c3.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

# Setup the UART connection
uart:
  id: mod_uart
  tx_pin:
    number: GPIO21
  rx_pin: 
    number: GPIO20

  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1

modbus:
  id: modbus_connection
  uart_id: mod_uart
  flow_control_pin: GPIO07

modbus_controller:
- id: modbus_device
  address: 0x3
  modbus_id: modbus_connection
  update_interval: 10s
  setup_priority: -10
  command_throttle: 2ms

sensor:
  - platform: modbus_controller
    modbus_controller_id: modbus_device
    name: "Battery SOH"
    register_type: read
    address: 1622
    register_count: 1
    value_type: U_WORD
    device_class: battery
    unit_of_measurement: "%"
    accuracy_decimals: 1
    filters:
      - multiply: 0.01
    icon: "mdi:battery-heart"
```

### Anything in the logs that might be useful for us?

```txt
[16:12:52][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=54232 (now=64232)
[16:12:53][VV][api.connection:132]: Sending keepalive PING...
[16:12:53][VV][api.service:037]: send_ping_request: PingRequest {}
[16:12:53][VV][api.service:683]: on_ping_response: PingResponse {}
[16:12:53][VV][api.connection:132]: Sending keepalive PING...
[16:12:53][VV][api.service:037]: send_ping_request: PingRequest {}
[16:12:53][VV][api.service:683]: on_ping_response: PingResponse {}
[16:13:00][VV][scheduler:226]: Running interval 'update' with interval=10000 last_execution=62224 (now=72224)
[16:13:00][V][modbus_controller:228]: Updating modbus component
[16:13:00][VV][modbus_controller:232]: Updating range 0x656
[16:13:00][V][modbus_controller:195]: Range : 656 Size: 1 (4) skip: 0
[16:13:00][V][modbus_controller:044]: Sending next modbus command to device 3 register 0x656 count 1
[16:13:00][VV][uart.arduino_esp32:194]:     Flushing...
[16:13:00][V][modbus:210]: Modbus write: 03.04.06.56.00.01.D0.B0 (8)
[16:13:00][V][modbus_controller:561]: Command sent 4 0x656 1
[16:13:00][V][modbus:042]: Modbus received Byte  3 (0X3)
[16:13:00][V][modbus:042]: Modbus received Byte  4 (0X4)
[16:13:00][V][modbus:042]: Modbus received Byte  2 (0X2)
[16:13:00][V][modbus:042]: Modbus received Byte  255 (0Xff)
[16:13:00][V][modbus:042]: Modbus received Byte  255 (0Xff)
[16:13:00][V][modbus:042]: Modbus received Byte  193 (0Xc1)
[16:13:00][V][modbus:042]: Modbus received Byte  64 (0X40)
[16:13:00][V][modbus_controller:079]: Modbus response queued
[16:13:00][V][modbus_controller:087]: Process modbus response for address 0x656 size: 2
[16:13:00][V][modbus_controller:168]: data for register address : 0x656 : 
[16:13:00][D][modbus_controller.sensor:025]: Sensor new state: 65535.00
[16:13:00][V][sensor:043]: 'Battery SOH': Received new state 65535.000000
[16:13:00][VV][sensor.filter:014]: Filter(0x3fca587c)::input(65535.000000)
[16:13:00][VV][sensor.filter:021]: Filter(0x3fca587c)::output(655.349976) -> SENSOR
[16:13:00][D][sensor:094]: 'Battery SOH': Sending state 655.34998 % with 1 decimals of accuracy
[16:13:00][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 2263547299
  state: 655.35
  missing_state: NO
}
```

jesserockz, 2024-09-19:
If the log shows the data coming in as 0xff 0xff then it might be a hardware issue.

ryjogo, 2024-09-19:
Thanks for some direction, I’ll try swap some components out - not sure what though it’s a very simple setup but I could try a new board perhaps

ryjogo, 2024-09-21:
I've been testing a little with the help from @ssieb and we concluded that i need to filter out NaN values, probably due to the inverter not _having_ any values here. I'll try with 

```
    filters:
      - filter_out: 0xffff
```

closed 2024-09-21 by ryjogo