esphome/issues #6255: Ethernet Info Text Sensor reports wrong MAC address

victorclaessen, 2024-09-17:
### The problem

I have several esp32 boards with ethernet on board:
* [Lilygo T-Internet-POE](https://www.lilygo.cc/products/t-internet-poe)
* [Kincony A16](https://www.kincony.com/arduino-esp32-16-channel-relay-module-2.html)
* [Kincony A868-AM](https://www.kincony.com/esp32-io-expansion-board.html)
* WT32-ETH01

I use the Ethernet Info Text Sensor on each of them.

```
text_sensor:
  - platform: ethernet_info
    mac_address:
      name: "${hostname} MAC address"
```

The problem is: the Ethernet Info Text Sensor reports the wrong MAC address for each of them!

One example, the text sensor (Lilygo POE board) reports MAC B4:8A:0A:9E:A9:**28**, but with my network hardware I can see that the real MAC address that is used is B4:8A:0A:9E:A9:**2B**. _(Note that one is number eighth, and the other is letter B.)_

Interestingly enough, for all my devices (25+) the MAC address is off by the value of 3, i.e. the MAC address reported by the text sensor is always 3 lower than the real MAC address. It's therefore easy to find the real address, but I think nevertheless this bug could be fixed.

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Lilygo T-Internet-POE

### What platform are you using?

ESP32

### Board

several ESP32 boards with ethernet onboard

### Component causing the issue

ethernet_info

victorclaessen, 2024-09-17:
I wonder if this is related to #5167.

randybb, 2024-09-17:
yes, it is the same problem. checking mine device - reported MAC in HA is E8:9F:6D:D2:EC:60, the real one visible on my switch is E8:9F:6D:D2:EC:63
Duplicate of https://github.com/esphome/issues/issues/5167

closed 2024-09-17 by randybb

victorclaessen, 2024-09-17:
I am confused. Why call it the "ethernet info text sensor" if it doesn't actually report the "ethernet info" (but the "wifi info")?

randybb, 2024-09-18:
Because nobody noticed that issue before.

victorclaessen, 2024-09-18:
Ah, ok. But you agree that, in principe, it should be fixed?