esphome/issues #6257: ESP32 S3 Box 3 cannot install 2024.9.0 due to missing `invert_colors` option

dj-sf, 2024-09-18:
### The problem

Upon attempting installation of ESPHome 2024.9.0 on my ESP32 S3 Box 3 I get the following error.

```
display.ili9xxx: [source /data/packages/dec89df9/esp32-s3-box-3/esp32-s3-box-3.yaml:688]
  
  'invert_colors' is a required option for [display.ili9xxx].
  platform: ili9xxx
  id: s3_box_lcd
  model: S3BOX
  data_rate: 40MHz
  cs_pin: 5
  dc_pin: 4
  reset_pin: 
    number: 48
    inverted: True
  update_interval: never
  pages: 
    - id: idle_page
==== [ERROR] /config/esphome/esp32-s3-box-3-059d58.yaml ====
```

Impact: I cannot install `2024.9.0`

### Which version of ESPHome has the issue?

2024.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.2

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
substitutions:
  name: esp32-s3-box-3-059d58
  friendly_name: ESP32 S3 Box 3 059d58
packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: ***REDACTED***

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
Full log:

INFO ESPHome 2024.9.0
INFO Reading configuration /config/esphome/esp32-s3-box-3-059d58.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

display.ili9xxx: [source /data/packages/dec89df9/esp32-s3-box-3/esp32-s3-box-3.yaml:688]
  
  'invert_colors' is a required option for [display.ili9xxx].
  platform: ili9xxx
  id: s3_box_lcd
  model: S3BOX
  data_rate: 40MHz
  cs_pin: 5
  dc_pin: 4
  reset_pin: 
    number: 48
    inverted: True
  update_interval: never
  pages: 
    - id: idle_page
```
```

dj-sf, 2024-09-18:
seems related to this PR: https://github.com/esphome/esphome/pull/7292

dj-sf, 2024-09-18:
also, link to the yaml file where this needs to be added:

https://github.com/esphome/wake-word-voice-assistants/blob/79ec1c60f4f36ae18b2a5a3765c02e1ca2a4bfab/esp32-s3-box-3/esp32-s3-box-3.yaml#L688

ssieb, 2024-09-18:
Duplicate of https://github.com/esphome/wake-word-voice-assistants/issues/18

closed 2024-09-18 by ssieb