esphome/issues #6258: ESPHome device working but not appearing in ESPHome dashboard in HA

Stooovie, 2024-09-19:
### The problem

I had to reinstall the M5 Atom Echo. Prior to that, I removed it from both ESPHome addon and HA Settings/Devices.

I used the official way to flash it, it asked me to add to Home Assistant, I did that, it works fine. I can control HA via voice with this device.

But it doesn’t appear in the ESPHome dashboard (my other ESPhome device does, and works fine). It does appear in HA Settings/Devices.

Is this how it’s supposed to be? I rebooted the HA VM between uninstalling and reinstalling, and after the reinstall, no dice.

ESPHome has debug logging enabled but doesn’t show anything else than

2024-09-18 21:13:16,729 INFO 304 GET /devices (0.0.0.0) 1.74ms
2024-09-18 21:13:22,733 INFO 304 GET /devices (0.0.0.0) 2.71ms
2024-09-18 21:13:28,729 INFO 304 GET /devices (0.0.0.0) 2.37ms
2024-09-18 21:13:34,755 INFO 304 GET /devices (0.0.0.0) 4.26ms
2024-09-18 21:13:40,730 INFO 304 GET /devices (0.0.0.0) 3.56ms
2024-09-18 21:13:45,745 INFO 304 GET /devices (0.0.0.0) 2.11ms
2024-09-18 21:13:50,736 INFO 304 GET /devices (0.0.0.0) 4.48ms
2024-09-18 21:13:56,724 INFO 304 GET /devices (0.0.0.0) 1.08ms
Any ideas?? Thanks

### Which version of ESPHome has the issue?

2024.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.2

### What platform are you using?

ESP32

ssieb, 2024-09-20:
If MDNS is working, the dashboard should show it as adoptable.

Stooovie, 2024-09-20:
I used the [official HA site](https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/) to install, have no other apparent issues with mDNS and so far everything has appeared in the Esphome dashboard, except this.

closed 2025-02-03 by github-actions[bot]