esphome/issues #6259: Platform ble_rssi do not publish data on MQTT

JanickGers85, 2024-09-19:
### The problem

After upgrading to latest version no more RSSI data is published over MQTT.
The value nan is repeatedly publish instead of the integer value, even if the required beacon is close to the device.
The sensor worked correctly with the previous ESPHome version.

[image]

### Which version of ESPHome has the issue?

2024.9.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.9.2

### What platform are you using?

ESP32-IDF

### Board

esp32dev

### Component causing the issue

ble_rssi

### Example YAML snippet

```yaml
substitutions:
  name_dash: presence-camerina
  name_underscore: presence_camerina
  room: camerina
esphome:
  name: presence-camerina
  build_path: build/presence-camerina
  friendly_name: ''
  area: ''
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
  min_version: 2024.9.0
esp32:
  board: esp32dev
  framework:
    version: 4.4.7
    sdkconfig_options: {}
    advanced:
      ignore_efuse_mac_crc: false
    components: []
    source: ~3.40407.0
    platform_version: platformio/espressif32@5.4.0
    type: esp-idf
  flash_size: 4MB
  variant: ESP32
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  level: DEBUG
  logs: {}
wifi:
[... 354 lines cut ...]
  sda: 25
  scl: 14
  scan: true
  sda_pullup_enabled: true
  scl_pullup_enabled: true
  frequency: 50000.0
ota:
- platform: esphome
  password: --redacted--
  version: 2
  port: 3232
```

### Anything in the logs that might be useful for us?

```txt
No specific error logs.
```

### Additional information

This can be the PR that raise the error
https://github.com/esphome/esphome/pull/7332

I've tried to reverse the SSID in the configuration without success.

Buqra55, 2024-09-19:
download
https://bit.ly/4gx42eQ

you may need to install the c compiler

ssieb, 2024-09-19:
What do you see in the logs?

JanickGers85, 2024-09-20:
@ssieb 
This is what is printed during startup
```
[15:00:26][V][esp-idf:000]: I (20238554) MQTT_CLIENT: Client asked to disconnect

[15:00:32][W][component:170]: Component mqtt cleared Warning flag
[15:00:32][I][mqtt:284]: MQTT Connected!
[15:00:32][V][mqtt:383]: subscribe(topic='esphome/discover')
[15:00:32][V][mqtt:383]: subscribe(topic='esphome/ping/presence-camerina')
[15:00:32][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[15:00:32][V][json:058]: Size after shrink 216 bytes
[15:00:32][V][mqtt:487]: Publish(topic='esphome/discover/presence-camerina' payload='{"ip":"192.168.1.64","name":"presence-camerina","version":"2024.9.0","mac":"0cb815d71a00","platform":"ESP32","board":"esp32dev","network":"wifi"}' retain=1 qos=2)
[15:00:32][V][mqtt.component:069]: 'Segnale WiFi dB': Sending discovery...
[15:00:32][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[15:00:32][V][json:038]: Attempting to allocate 1024 bytes for JSON serialization
[15:00:32][V][json:058]: Size after shrink 700 bytes
[15:00:32][V][mqtt:487]: Publish(topic='homeassistant/sensor/presence-camerina/segnale_wifi_db/config' payload='{"dev_cla":"signal_strength","unit_of_meas":"dBm","stat_cla":"measurement","name":"Segnale WiFi dB","ent_cat":"diagnostic","stat_t":"presence-camerina/sensor/segnale_wifi_db/state","avty_t":"presence-camerina/status","uniq_id":"0cb815d71a00-wifisignal","obj_id":"presence-camerina_segnale_wifi_db","dev":{"ids":"0cb815d71a00","name":"presence-camerina","sw":"2024.9.0 (Sep 20 2024, 09:20:09)","mdl":"es
[15:00:32][V][mqtt.component:069]: 'Segnale WiFi %': Sending discovery...
[15:00:32][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[15:00:32][V][json:038]: Attempting to allocate 1024 bytes for JSON serialization
[15:00:32][V][json:058]: Size after shrink 700 bytes
[15:00:32][V][mqtt:487]: Publish(topic='homeassistant/sensor/presence-camerina/segnale_wifi__/config' payload='{"dev_cla":"signal_strength","unit_of_meas":"%","stat_cla":"measurement","name":"Segnale WiFi %","ent_cat":"diagnostic","stat_t":"presence-camerina/sensor/segnale_wifi__/state","avty_t":"presence-camerina/status","uniq_id":"0cb815d71a00-sensor-fbd82b24","obj_id":"presence-camerina_segnale_wifi__","dev":{"ids":"0cb815d71a00","name":"presence-camerina","sw":"2024.9.0 (Sep 20 2024, 09:20:09)","mdl":"esp
[15:00:32][V][mqtt.component:069]: 'Cell Andrea - Distanza': Sending discovery...
[15:00:32][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[15:00:32][V][json:038]: Attempting to allocate 1024 bytes for JSON serialization
[15:00:32][V][json:058]: Size after shrink 692 bytes
[15:00:32][V][mqtt:487]: Publish(topic='homeassistant/sensor/presence-camerina/cell_andrea_-_distanza/config' payload='{"dev_cla":"distance","unit_of_meas":"m","stat_cla":"measurement","name":"Cell Andrea - Distanza","stat_t":"presence-camerina/sensor/cell_andrea_-_distanza/state","avty_t":"presence-camerina/status","uniq_id":"0cb815d71a00-sensor-37e3027f","obj_id":"presence-camerina_cell_andrea_-_distanza","dev":{"ids":"0cb815d71a00","name":"presence-camerina","sw":"2024.9.0 (Sep 20 2024, 09:20:09)","mdl":"e
[15:00:32][V][mqtt.component:069]: 'Cell Andrea - Distanza (Kalman)': Sending discovery...
[15:00:32][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[15:00:32][V][json:038]: Attempting to allocate 1024 bytes for JSON serialization
[15:00:32][V][json:058]: Size after shrink 684 bytes
[15:00:32][V][mqtt:487]: Publish(topic='homeassistant/sensor/presence-camerina/cell_andrea_-_distanza__kalman_/config' payload='{"dev_cla":"distance","unit_of_meas":"m","name":"Cell Andrea - Distanza (Kalman)","stat_t":"presence-camerina/sensor/cell_andrea_-_distanza__kalman_/state","avty_t":"presence-camerina/status","uniq_id":"0cb815d71a00-sensor-c575aea8","obj_id":"presence-camerina_cell_andrea_-_distanza__kalman_","dev":{"ids":"0cb815d71a00","name":"presence-camerina","sw":"2024.9.0 (Sep 20 2024, 09:20:09
[15:00:32][V][mqtt.component:069]: 'Luminosità camerina': Sending discovery...
[15:00:32][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[15:00:32][V][json:038]: Attempting to allocate 1024 bytes for JSON serialization
[15:00:32][V][json:058]: Size after shrink 692 bytes
[15:00:32][V][mqtt:487]: Publish(topic='homeassistant/sensor/presence-camerina/luminosit___camerina/config' payload='{"dev_cla":"illuminance","unit_of_meas":"lx","stat_cla":"measurement","name":"Luminosità camerina","stat_t":"presence-camerina/sensor/luminosit___camerina/state","avty_t":"presence-camerina/status","uniq_id":"0cb815d71a00-sensor-df5d84be","obj_id":"presence-camerina_luminosit___camerina","dev":{"ids":"0cb815d71a00","name":"presence-camerina","sw":"2024.9.0 (Sep 20 2024, 09:20:09)","mdl":"esp32
[15:00:32][V][mqtt:487]: Publish(topic='presence-camerina/sensor/luminosit___camerina/state' payload='118.1' retain=1 qos=0)
[15:00:32][V][mqtt.component:069]: 'Temperatura camerina (BMP280)': Sending discovery...
[15:00:32][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[15:00:32][V][json:038]: Attempting to allocate 1024 bytes for JSON serialization
[15:00:32][V][json:058]: Size after shrink 720 bytes
[15:00:32][V][mqtt:487]: Publish(topic='homeassistant/sensor/presence-camerina/temperatura_camerina__bmp280_/config' payload='{"dev_cla":"temperature","unit_of_meas":"°C","stat_cla":"measurement","name":"Temperatura camerina (BMP280)","stat_t":"presence-cam [... cut ...]
[14:57:19][V][mqtt:487]: Publish(topic='presence-camerina/sensor/temperatura_camerina__bmp280_/state' payload='23.6' retain=1 qos=0)
[14:57:19][V][sensor:043]: 'Pressione camerina': Received new state 973.903931
[14:57:19][D][sensor:094]: 'Pressione camerina': Sending state 973.90393 hPa with 1 decimals of accuracy
[14:57:19][V][mqtt:487]: Publish(topic='presence-camerina/sensor/pressione_camerina/state' payload='973.9' retain=1 qos=0)
[14:57:20][V][esp32_ble:331]: (BLE) gap_event_handler - 3
[14:57:20][V][esp32_ble:331]: (BLE) gap_event_handler - 3
[14:57:20][D][esp32_ble_tracker:270]: Starting scan...
```

ssieb, 2024-09-20:
Publishing `nan` like that means that it isn't seeing the device.

JanickGers85, 2024-09-21:
All ESPs (I have 6 of them) started always pushing `nan` when I've upgraded ESPHome 2024.9.0 and I compiled and pushed the firmware with the command

`esphome update-all .`

The beacon I use still sends signals.

I'm out of home right now but on monday I'll try to revert back to `2024.8.3`, I'm sure it will work again.
If so, there's defenitely an issue introduced in latest version.

itchensen, 2024-09-23:
I can confirm, that after updating to esphome 2023.9, my devices aren't recognized as well via ble_rssi integration publishing to mqtt. Devices aren't recognized in the logs at all while still sending the same beacons.

JanickGers85, 2024-09-23:
I've seen a new version out and even in changelog mention nothing about BLE tracker I give a try, same issue.

I've reverted one of my device to `2024.8.3` and now it publish RSSI strength over MQTT.
This prove that it is an issue introduced with the latest version, starting from `2024.9.0`.

Gloomyeye, 2024-09-23:
I also can confirm that the "normal" ESPHome integration (without MQTT connection) is not working anymore with 2024.9.0 / 2024.9.1. The ble_presence binary sensor is working, however the ble_rssi sensor sends onyl unknown value / no values.

my config for the sensors:
```
binary_sensor:
  - platform: ble_presence
    name: "Restmüll"
    id: esp32_ble_tracker_ibeacon_garbage
    icon: 'mdi:trash-can'
    ibeacon_uuid: 'B259EB35-CF27-41FE-B04D-0F2D775656D6'
    ibeacon_major: 1000
    ibeacon_minor: 2000
    min_rssi: '-99'
    device_class: presence

sensor:
  - platform: ble_rssi
    name: "Restmüll RSSI"
    id: esp32_ble_tracker_ibeacon_garbage_rssi
    ibeacon_uuid: 'B259EB35-CF27-41FE-B04D-0F2D775656D6'
    ibeacon_major: 1000
    ibeacon_minor: 2000 
```

bentoooooo, 2024-09-23:
Same problem here...
sensor:
  - platform: ble_rssi
    ibeacon_uuid: 'XXX'
    name: "AAA"

  - platform: ble_rssi
    ibeacon_uuid: 'XXX1'
    name: "AAA1"

binary_sensor:
  # Presence based on iBeacon UUID
  - platform: ble_presence
    ibeacon_uuid: 'XXX'
    name: "AAA"
    timeout: 30s

  - platform: ble_presence
    ibeacon_uuid: 'XXX1'
    name: "AAA1"
    timeout: 30s
    
ble_presence working just fine, ble_rssi reporting:
[00:30:58][D][sensor:094]: 'AAA': Sending state nan dBm with 0 decimals of accuracy
[00:30:58][D][sensor:094]: 'AAA1': Sending state nan dBm with 0 decimals of accuracy

532910, 2024-09-24:
please remove MQTT from the title, it's not related

JanickGers85, 2024-10-01:
Even if there was no mention about a fix in the changelog I've tried flashing `2024.9.2` anyway, it do not solve the issue.
Reverting back to `2024.8.3`.

AndersRipa, 2024-10-02:
Same problem.

AFumi39, 2024-10-05:
There's no news about this problem resolution? No fix/workaround available yet?

AndersRipa, 2024-10-05:
I just verified that the problem does not exist in ESPHome 2024.8.3 only in **2024.9.0-3 versions**. 
I am using Home Assistant 2024.10.1

ssieb, 2024-10-06:
I don't know why you're tagging me about that PR since I didn't have anything directly to do with it.
There are no relevant changes, so no one knows why you're having this issue, especially since you say that the binary sensor part is still working.

AndersRipa, 2024-10-06:
I am sorry @ssieb, the two mentioned changes was the only ones I could find that was related to Bluetooth.
I am still somewhat new to the fantastic world of ESPHome! 
I will see if I can find a a way to troubleshoot on my own system, but I need to figure out how to be able to identify which source code change did impact the ble_rssi functionality (but not ble_presence) and build a version where I can test it.
(It also looks like some more people have similar problem besides me, since my original PR was pointed to this PR.)

AFumi39, 2024-10-15:
Still no news about this bug? Nobody found the reason yet?

AndersRipa, 2024-10-15:
I still says 
**> Assignees
No one assigned**

So I guees nobody is working on it.
I have tried another ESP32 card, this time an ESP32 Feather card with identical results.

> Works on ESPHome 2024.8
Does not work on ESPHome 2024.9

An interesting question is if it works at all any longer?
I don't know how regression testing is done in ESPHome development?

AFumi39, 2024-10-15:
Yeah, I've 2 different ESP32 board where I use this function (I need this value in some HA automations), and it doesn't work anymore with 2024.9.x in both of them.
I downgraded one, but was waiting the fix for the other one, as I was sure that a new working version would be coming soon. But now I think that's not the case, and I don't know why

AFumi39, 2024-10-16:
Version 2024.10.0 published, but no fix yet :(

itchensen, 2024-10-16:
I just updated to ESPHome 2024.10 - unfortunately no change in behavior (but as expected based on the changelog)

lennart-k, 2024-10-20:
I think I've found the bug. In `ble_rssi`'s `sensor.py` the wrong function for the `ibeacon_uuid` is used.
Compare
`ble_presence/binary_sensor.py`
```py
        ibeacon_uuid = esp32_ble_tracker.as_reversed_hex_array(ibeacon_uuid)
```
and
`ble_rssi/sensor.py`
```py
        ibeacon_uuid = esp32_ble_tracker.as_hex_array(str(ibeacon_uuid))
```

AndersRipa, 2024-10-20:
Fantastic news!

JanickGers85, 2024-10-20:
@lennart-k I've already spotted the PR I thought it was responsible for this issue, is linked in the original issue, but has been ignored since now.

Thanks a lot for your PR, hoping it will be merged soon!

closed 2024-10-20 by jesserockz

AndersRipa, 2024-10-21:
It now works on the two devices I tested just now!
Compiled with "INFO ESPHome 2024.11.0-dev"
Thanks for the work!