esphome/issues #6260: Since upgrade to 2024.9.0 Dallas temp sensors not working

deanhalllincoln, 2024-09-19:
### The problem

Hi I have one esp32 device with two dallas temp sensors connected to it. All has been working fine for weeks. Today upgraded to 2024.9.0 and both sensors stopped sending data. I have tried to update the config on the ESP32 devce and the logs are below. There is a message in the logs when updating the yaml config to the ESP32 device that says devices not found. No changes to hardware or wiring have happened so this makes no sense.
Included yaml file as well.
Any help would be appreciated.

### Which version of ESPHome has the issue?

2024.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.2

### What platform are you using?

ESP32

### Component causing the issue

Dallar temp sensors

### Example YAML snippet

```yaml
substitutions:
  name: esphome-web-51f44c
  friendly_name: HP-FR-temp-sensors

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32dev
  framework:
    type: arduino

one_wire:
  - platform: gpio
    pin: GPIO25

sensor:
  - platform: dallas_temp
    address: 0x3a0000003858b428
    name: Flow
    update_interval: 10s
  - platform: dallas_temp
    address: 0x2000000037443528
    name: Return
    update_interval: 10s    

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome
[... 152 lines cut ...]
[13:17:42][W][dallas.temp.sensor:139]: 'Flow' - Scratch pad checksum invalid!
[13:17:42][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[13:17:42][D][sensor:094]: 'Flow': Sending state nan °C with 1 decimals of accuracy
[13:17:50][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[13:17:50][W][dallas.temp.sensor:139]: 'Return' - Scratch pad checksum invalid!
[13:17:50][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[13:17:50][D][sensor:094]: 'Return': Sending state nan °C with 1 decimals of accuracy
[13:17:52][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[13:17:52][W][dallas.temp.sensor:139]: 'Flow' - Scratch pad checksum invalid!
[13:17:52][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[13:17:52][D][sensor:094]: 'Flow': Sending state nan °C with 1 decimals of accuracy
[13:18:00][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[13:18:01][W][dallas.temp.sensor:139]: 'Return' - Scratch pad checksum invalid!
[13:18:01][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[13:18:01][D][sensor:094]: 'Return': Sending state nan °C with 1 decimals of accuracy
[13:18:02][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[13:18:02][W][dallas.temp.sensor:139]: 'Flow' - Scratch pad checksum invalid!
[13:18:02][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[13:18:02][D][sensor:094]: 'Flow': Sending state nan °C with 1 decimals of accuracy
[13:18:10][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[13:18:10][W][dallas.temp.sensor:139]: 'Return' - Scratch pad checksum invalid!
[13:18:10][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[13:18:10][D][sensor:094]: 'Return': Sending state nan °C with 1 decimals of accuracy
[13:18:12][W][component:170]: Component dallas_temp.sensor cleared Warning flag
```

ssieb, 2024-09-20:
More sensors with a surprising number of 0s.  Are you using parasite power?  Can you get serial logs from boot?

ssieb, 2024-09-20:
I asked you for more info on discord, but you didn't reply.  What was the previous esphome version installed on this device?  There haven't been any 1-wire changes for quite a while.

deanhalllincoln, 2024-09-20:
Hi Ssieb thanks for the reply very much appreciated. 
I was running 2024.8.3 ESPhome prior to upgrading and all was running ok. Once upgraded to 2024.9.0 the two Dallas sensors stopped working.
Do you mean by "Parasitic power" that I am running both Dallas devices from the power supply to the ESP32 board? Then yes I am and its been working fine for weeks. My power supply arrangement is a mini USB cable from a USB 13A plug.

Ok here are the steps I have taken :-
a) Upgraded HA ESPHome to 2024.9.0 from 2024.8.3
b) For full info there was an upgrade to HA Core 2024.9.2 either just before or just after I carnt remeber.
c) Every thing in HA was working ok after the install except the one ESPHome device I have with two Dallas temp sensors attached.
d) The temp from the two sensors had stopped reporting.
e) I tried installing the YAML file to the ESP32 device while running 2024.9.0 - No differance still not working.
f) I did notice a line in the logs "[13:17:05][W][gpio.one_wire:078]:   Found no devices!" this seemed odd as not seen this before.
g) I then decided to restore HA back to a backup prior to installing the upgrades to ESPHome and HA core.
h) Following the restore the sensors still not reporting and data even though ESPhome version 2024.9.0 and HA core version 2024.9.1 installed
I) I installed the YAML to the ESP32 device and hey presto the sensors started working and reporting data to HA.

So going back to pre-upgrade versions of ESPhome and HA core has rectified the issue. So it appears as far as I can see the issue is with the upgrades not with a power supply as no changes made to any wiring arrangements at all.

Also when loading the YAML to the ESPHome device I noticed the line in the logs that now says "[10:07:12][C][gpio.one_wire:080]:   Found devices:"

******* Question I have is is it the ESPhome upgrade thats causing the issue or the HA core upgrade???? **********

I have copied the full logs below:-
INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/esphome-web-51f44c.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-51f44c (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.4.0
INFO Installing platformio/espressif32 @ 5.4.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.4.0 has been installed!
INFO espressif32@5.4.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 has been installed!
INFO toolchain-xtensa-esp32@8.4.0+2021r2-patch5 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
INFO Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
INFO framework-arduinoespressif32@3.20005.220925 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40400.0
INFO Installing platformio/tool-esptoolpy @ ~1.40400.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Tool Manager: tool-esptoolpy@1.40400.0 has been installed!
INFO tool-esptoolpy@1.40400.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40700.0
INFO Installing platformio/tool-scons @ ~4.40700.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40700.0 has been installed!
INFO tool-scons@4.40700.0 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.1.3
INFO Installing esphome/AsyncTCP-esphome @ 2.1.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.3 has been installed!
INFO AsyncTCP-esphome@2.1.3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.2.2 has been installed!
INFO ESPAsyncWebServer-esphome@3.2.2 has been installed!
[... 277 lines cut ...]
[10:07:33][D][dallas.temp.sensor:054]: 'Flow': Got Temperature=30.3°C
[10:07:33][D][sensor:094]: 'Flow': Sending state 30.31250 °C with 1 decimals of accuracy
[10:07:43][D][dallas.temp.sensor:054]: 'Return': Got Temperature=31.5°C
[10:07:43][D][sensor:094]: 'Return': Sending state 31.50000 °C with 1 decimals of accuracy
[10:07:43][D][dallas.temp.sensor:054]: 'Flow': Got Temperature=30.4°C
[10:07:43][D][sensor:094]: 'Flow': Sending state 30.37500 °C with 1 decimals of accuracy
[10:07:53][D][dallas.temp.sensor:054]: 'Return': Got Temperature=31.6°C
[... cut ...]
[10:08:13][D][sensor:094]: 'Return': Sending state 31.50000 °C with 1 decimals of accuracy
[10:08:13][D][dallas.temp.sensor:054]: 'Flow': Got Temperature=30.3°C
[10:08:13][D][sensor:094]: 'Flow': Sending state 30.31250 °C with 1 decimals of accuracy
[10:08:23][D][dallas.temp.sensor:054]: 'Return': Got Temperature=31.5°C
[10:08:23][D][sensor:094]: 'Return': Sending state 31.50000 °C with 1 decimals of accuracy
[10:08:23][D][dallas.temp.sensor:054]: 'Flow': Got Temperature=30.3°C
[10:08:23][D][sensor:094]: 'Flow': Sending state 30.31250 °C with 1 decimals of accuracy

ssieb, 2024-09-20:
It can't be the HA upgrade, since that doesn't affect esphome on the device.  There were no direct changes to any of the 1-wire code, so I don't know what would have changed.  Is there any chance you have a logic analyzer or oscilloscope handy?
Btw, parasitic power is when the sensor doesn't have a VCC line connected and gets power from the data line.

deanhalllincoln, 2024-09-20:
Sorry since i retired I don't have access to logic analyser or oscilloscope. 
ill apply the HA core update just to make sure its not the issue. Ill hang fire updating ESPhome as that looks like the where the issue lies.
The sensor has VCC, ground and data input connected to the Esp32 so not parasitic.

deanhalllincoln, 2024-09-21:
Hi ssieb, just to feedback I have upgraded HA core and sensor are woing ok. So it appears it the ESPhome update that breaks the 1-wire sensors.

daernsinstantfortress, 2024-09-21:
I've just updated one of my own ESP8266 modules with two dallas probes to 2024.9.0 and everything is working ok (including temperature updates in HA 2024.9.2), so I suspect your issue must lie elsewhere.  I've had issues in the past when running multiple dallas sensors that you need to reduce the pull-up resistor on the data line so that there's enough current flowing to lift the data line up. Without this things get a bit marginal and start to go crazy, often working or not working based on things like ambient temperature. Do you know what value you are using?

```
INFO ESPHome 2024.9.0
INFO Reading configuration /config/esphome/garage-temperature.yaml...
INFO Starting log output from 192.168.0.34 using esphome API
INFO Successfully connected to garage-temperature @ 192.168.0.34 in 0.005s
INFO Successful handshake with garage-temperature @ 192.168.0.34 in 2.652s
[09:46:02][I][app:100]: ESPHome version 2024.9.0 compiled on Sep 21 2024, 09:43:28
[09:46:02][C][wifi:600]: WiFi:
[09:46:02][C][wifi:428]:   Local MAC: 50:02:91:E0:DF:16
[09:46:02][C][wifi:433]:   SSID: [redacted]
[09:46:02][C][wifi:436]:   IP Address: 192.168.0.34
[09:46:02][C][wifi:439]:   BSSID: [redacted]
[09:46:02][C][wifi:441]:   Hostname: 'garage-temperature'
[09:46:02][C][wifi:443]:   Signal strength: -60 dB ▂▄▆█
[09:46:02][C][wifi:447]:   Channel: 11
[09:46:02][C][wifi:448]:   Subnet: 255.255.254.0
[09:46:02][C][wifi:449]:   Gateway: 192.168.1.254
[09:46:02][C][wifi:450]:   DNS1: 192.168.1.20
[09:46:02][C][wifi:451]:   DNS2: 0.0.0.0
[09:46:02][C][logger:185]: Logger:
[09:46:02][C][logger:186]:   Level: DEBUG
[09:46:02][C][logger:188]:   Log Baud Rate: 115200
[09:46:02][C][logger:189]:   Hardware UART: UART0
[09:46:02][C][gpio.one_wire:020]: GPIO 1-wire bus:
[09:46:02][C][gpio.one_wire:021]:   Pin: GPIO4
[09:46:02][C][gpio.one_wire:080]:   Found devices:
[09:46:02][C][gpio.one_wire:082]:     0x18011441a8a0aa28 (DS18B20)
[09:46:02][C][gpio.one_wire:082]:     0x590114412941aa28 (DS18B20)
[09:46:02][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[09:46:02][C][dallas.temp.sensor:034]:   Address: 0x18011441a8a0aa28 (DS18B20)
[09:46:02][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[09:46:02][C][dallas.temp.sensor:036]:   Update Interval: 60.0s
[09:46:02][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[09:46:02][C][dallas.temp.sensor:034]:   Address: 0x590114412941aa28 (DS18B20)
[09:46:02][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[09:46:02][C][dallas.temp.sensor:036]:   Update Interval: 60.0s
[09:46:02][C][captive_portal:089]: Captive Portal:
[09:46:02][C][mdns:116]: mDNS:
[09:46:02][C][mdns:117]:   Hostname: garage-temperature
[09:46:02][C][esphome.ota:073]: Over-The-Air updates:
[09:46:02][C][esphome.ota:074]:   Address: garage-temperature.local:8266
[09:46:02][C][esphome.ota:075]:   Version: 2
[09:46:02][C][esphome.ota:078]:   Password configured
[09:46:02][C][safe_mode:018]: Safe Mode:
[09:46:02][C][safe_mode:019]:   Boot considered successful after 60 seconds
[09:46:02][C][safe_mode:021]:   Invoke after 10 boot attempts
[09:46:02][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[09:46:02][C][api:139]: API Server:
[09:46:02][C][api:140]:   Address: garage-temperature.local:6053
[09:46:02][C][api:142]:   Using noise encryption: YES
[09:46:04][D][dallas.temp.sensor:054]: 'Garage Temperature Rear': Got Temperature=14.6°C
[09:46:04][D][sensor:093]: 'Garage Temperature Rear': Sending state 14.62500 °C with 1 decimals of accuracy
[09:46:13][D][dallas.temp.sensor:054]: 'Garage Temperature Front': Got Temperature=14.2°C
[09:46:13][D][sensor:093]: 'Garage Temperature Front': Sending state 14.25000 °C with 1 decimals of accuracy
[09:46:14][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
```

deanhalllincoln, 2024-09-21:
Hi daernsinstantfortress, Thanks for the post. strnage yours works any mine does not. My resistor is 4.7k and was working prior to the update so dont see how the update could need a different resistor??
What YAML code are you using for the two sensors? would be interested to see if it differes from mine to see if there are any issues there.

daernsinstantfortress, 2024-09-21:
@deanhalllincoln Here's my config:
```
esphome:
  name: "garage-temperature"
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: <redacted>
  password: <redacted>

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Garage Temperature"
    password: <redacted>

captive_portal:

logger:
  level: DEBUG

api:
  encryption:
    key: <redacted>

ota:
  - platform: esphome
    password: <redacted>

one_wire:
  - platform: gpio
    pin: GPIO4
  
sensor:
  - platform: dallas_temp
    address: 0x18011441A8A0AA28
    name : "Garage Temperature Rear"
  - platform: dallas_temp
    address: 0x590114412941AA28
    name : "Garage Temperature Front"
```
From my own experience having built quite a few ESP8266 / ESP32 with temperature sensors, I've definitely had some with multiple sensors where I've needed to go lower than 4k7, so perhaps this is something worth a try. I'm yet to have a bad temperature sensor (quite surprising considering that they are bottom-drawer eBay specials!) but I've had more than one ESP8266 just randomly start to go weird on me and after messing about, just replaced it with a new one which solved the problem.  The other option is to swap IO ports on the module.

Personally, I would drop the resistor first, then swap IO lines and then just replace the ESP module and move on!

deanhalllincoln, 2024-09-21:
Thanks, I dont see any issues with my Yaml compatred to yours. Ill try a lower resistor see if that makes any differance.

By the way any ideas how many Dallas sensors you can have on on IO Pin as I would like to add more going forward? once I have this sorted.

daernsinstantfortress, 2024-09-21:
> By the way any ideas how many Dallas sensors you can have on on IO Pin as I would like to add more going forward? once I have this sorted.

No idea what the actual limit is, but I've had 5 on a single ESP8266 GPIO pin for years and it works well enough. That said, this is the one ESP that is a pig to update every time a new version comes out. I'd always thought it was a wifi connection issue (it's in the boiler room), but you've got me wondering now...!

emefff, 2024-09-22:
Yesterday I updated my 15 ESPs (6 different kinds) with 40+ DS18B20s to ESPHome 2024.9.0. Regardless of cable length I always use 4.7k pullup, longest cable is ~5m. I use microphone cables. I did not experience any problems after this update, although I've had plenty in the past with Dallas.

deanhalllincoln, 2024-09-22:
> Yesterday I updated my 15 ESPs (6 different kinds) with 40+ DS18B20s to ESPHome 2024.9.0. Regardless of cable length I always use 4.7k pullup, longest cable is ~5m. I use microphone cables (shielding is even floating). I did not experience any problems after this update, although I've had plenty in the past with Dallas.

Yep its weird, everything points to the update being the issue but if it was then loads of people would be posting about it as its a common usage of ESP devices. Ill have to try a few different things at my end (such as the resistor) till i find a solution.

1N4148, 2024-09-30:
Yeah, the Dallas code is funny. If I comment out the mqtt: client in a working multi sensor, multi 1-wire bus setup, some sensors fail. Enabling mqtt: again and all sensors are working. So probably some kind of timing issues in the code? Will try the experimental DS248x component now.

thimo-seitz, 2024-10-01:
I'm running ten DS18B20 on a ESP8266 for Years without a problem. After upgrade of the image on the ESP8266 there are plenty of checksum errors. After downgrading the the image on the ESP to 2024.8.3 everything works fine again. No checksum errors anymore.  On homeassistant itself there's esphome 2024.9.2 installed.

alistair23, 2024-10-08:
Same here. Before `2024.9` I had no issues, after updating to `2024.9.2` I get `scratch pad checksum invalid` errors

ssieb, 2024-10-08:
There have been no changes that directly affect the component.
Is there anyone having this issue that can get traces from a logic analyzer?

ssieb, 2024-10-08:
If you have a logic analyzer, then come find me on the esphome discord server.  Otherwise, you can't do it.

thimo-seitz, 2024-10-08:
I gues a "logic analyzer" is not a module that's simply compiled into a eshpome-image?

ssieb, 2024-10-08:
No, it's hardware.  Something like https://www.aliexpress.com/item/1005007076428972.html.

m1ch, 2024-10-16:
I am having similar issues. Yesterday I added three additional DS18B20 to one board and updated from 2023.x to 2024.9. 

This was the steps I performed:

- Activate DEBUG and install to board.
- Add each new DS18B20 individual, to get their addresses
- Add the new dallas_temp sensors to the yaml and install to board. After reboot, only four out of six one-wire sensors where detected.
- Remove all other peripherals for testing from the yaml. Now no DS18B20 was detected. 

My conclusion of this behavior is, that the one_wire component can't initialize the sensors during boot and thus they are not working correct at runtime. With lots of additional peripherals added, boot-up takes more time, so the the one_wire component detects more devices. I could not get boot-up to slow down enough to detect more than four sensors. 

I hope this can narrow down the issue, otherwise I can hook up a logic analyzer later.

ssieb, 2024-10-16:
@m1ch can you please find me on discord to discuss this?

fran6120, 2024-10-16:
I think I'm having problems since I moved from `dallas` component to `one_wire` component.

```
[00:27:17][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[00:27:17][W][dallas.temp.sensor:074]: 'temp_celula_interno' - reading scratch pad failed bus reset
[00:27:17][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
```

I tried changing the sensor and bus pull_up resistor to 3.3k, 4.7k, 5.6k, and 6.8k, but it didn't fix it.

Sometimes it randomly works for a while, but as soon as it works, it stops.

ssieb, 2024-10-16:
`bus reset failed`
Some sort of wiring problem or the sensor isn't responding.

1N4148, 2024-10-17:
It's not neccessary a wiring or sensor problem, see https://github.com/esphome/issues/issues/6260#issuecomment-2384137733

ssieb, 2024-10-17:
You didn't say how it was failing there.  Were you getting bus reset errors?

1N4148, 2024-10-17:
Sometimes bus reset errors, sometimes sensors not found, sometimes scratchpad invalid. 
Arduino + MQTT -> Scratchpad invalid (only sensors with a long wire)
Arduino - MQTT -> Sensors not found, bus reset error
ESP-IDF (regardless of MQTT or not) -> Everything fine

Cannot give you more detailed logs at the moment as it's my central heating controller and I've also switched over to DS2482-800 for stability reasons. 

Don't have a similar, complex environment at the moment to reproduce or connecting the logic analyzer.

For me this looks like a timing issue as it's independent of the pull up resistor value.

m1ch, 2024-10-18:
I did some further digging and fond some interesting things:

When I hooked up the logic analyzer I got the address 0xbe91a5f91d647f28, but in ESP-home, when I only had one sensor connected,  I got before the address 0xbe91a5f91d64ff28 (note the ff28 at the end instead of 7f28). 
From the LA readings and how one-wire works, it is clear, that also esp-home finds the device, when multiple sensors are connected, but ignores it. (I guess the CRC check is wrong for one of the IDs. Did not find a quick check to verify which one is correct)

When I change now the ID in my ESP-home config, I get correct readings. With the "wrong" ID I only get FFFFFFFFFF as a reply, which is also due to how one-wire works and, thus I get the CRC error message. 

I have following suggestions to improve the one-wire implementation:
* Add a CRC check in python, to check if an added sensor has a feasible ID. 
* Add a check if the reply of the sensor is only 1s (ffff), that there is no sensor with this ID connected, rather then there is a CRC error. 
* Add an output to the search algorithm of the one-wire initialization, if a device with a wrong CRC is detected. Also show the "wrong" ID, since it may work. 
* Add a longer pause between bits in the one-wire implementation. According to the specification, the maximum data rate of one-wire is 16.3 kbps which means 61.349us per bit. I saw in the output of the LA that ESP-home already starts after 60us with the following bit, if the bit transmitted was a 1. This seams to work, since the signal is high for sufficient time, but it is not in accordance to the spec. 

I hope this helps with a fix for this issues!

ssieb, 2024-10-18:
Checking the CRC in the config is a good idea.
If the CRC fails, then checking for all 0xff bytes would allow a more useful error message.
There is a CRC check in the scan, but you will only see that in serial logs.  The wrong id should never work.
The timing is very carefully according to spec, although there are issues with timing on an ESP that I'm hoping to workaround in the future.  I will check the search spec again, but it should be correct.

m1ch, 2024-10-18:
[image]

It's just some variations like this. I did not check the timings of the search algorithm in detail, if this is in spec.

ssieb, 2024-10-18:
It's only the bit internal timing that matters.  I don't think the inter-bit timing is even specified.

m1ch, 2024-10-18:
I also don't consider this point very relevant. The only thing that is mentioned in the spec is a max bitrate of 16.3 kbps, this implies a min interval between the bits, but it seems to work fine as is. 
This was just my thinking as a digital chip design engineer, who used to implement several RFID protocols in hardware. 
One of the issues there is to get enough power to the chip during transmission. The original intent from one-wire was to be used in parasitic mode and that you see in the spec. And so I saw the issue with getting sufficient power to the chip during transmission as well, specially since one-wire does not use an NRZ encoding. 
But for active mode it should be fine.

jurabln, 2024-11-04:
Any news on this topic? I can´t get it work neither with ESP32 nor with ESP8266.

daernsinstantfortress, 2024-11-04:
One of my ESP8266s with 5 x Dallas probes was dying - it was having some wifi connection issues, which I've seen before on other older devices. I did a straight like-for-like replacement with an ESP32 (nodemcu-32s) and have had zero issues shifting the config across. I have several other devices which have all been updated multiple times and I'm still not seeing any issues.

I wonder if there's a pattern in the GPIO port that is being used here to connect to the devices? In this particular case, I'm using GPIO21 which seems to work fine for my use case. Perhaps others could share their own config...

Heronimos, 2024-11-18:
Sadly I don't know the version before.  But was from before one_wire protocol but the other one (dallas)

jurabln, 2024-11-21:
Same with the latest release => device can´t be found

[21:32:04][I][app:100]: ESPHome version 2024.11.0 compiled on Nov 21 2024, 21:30:59
[21:32:04][C][wifi:600]: WiFi:
[21:32:04][C][wifi:428]:   Local MAC: [redacted]
[21:32:04][C][wifi:433]:   SSID: [redacted]
[21:32:04][C][wifi:436]:   IP Address: [redacted]
[21:32:04][C][wifi:439]:   BSSID: [redacted]
[21:32:04][C][wifi:441]:   Hostname: 'esp8266heatingflow'
[21:32:04][C][wifi:443]:   Signal strength: -57 dB ▂▄▆█
[21:32:04][C][wifi:447]:   Channel: 6
[21:32:04][C][wifi:448]:   Subnet: [redacted]
[21:32:04][C][wifi:449]:   Gateway: [redacted]
[21:32:04][C][wifi:450]:   DNS1: [redacted]
[21:32:04][C][wifi:451]:   DNS2: 0.0.0.0
[21:32:04][C][logger:185]: Logger:
[21:32:04][C][logger:186]:   Level: DEBUG
[21:32:04][C][logger:188]:   Log Baud Rate: 115200
[21:32:04][C][logger:189]:   Hardware UART: UART0
[21:32:04][C][gpio.one_wire:020]: GPIO 1-wire bus:
[21:32:04][C][gpio.one_wire:021]:   Pin: GPIO2
[21:32:04][W][gpio.one_wire:078]:   Found no devices!
[21:32:04][C][captive_portal:089]: Captive Portal:
[21:32:04][C][mdns:116]: mDNS:
[21:32:04][C][mdns:117]:   Hostname: esp8266heatingflow
[21:32:04][C][esphome.ota:073]: Over-The-Air updates:
[21:32:04][C][esphome.ota:074]:   Address: esp8266heatingflow.local:8266
[21:32:04][C][esphome.ota:075]:   Version: 2
[21:32:04][C][esphome.ota:078]:   Password configured
[21:32:04][C][safe_mode:018]: Safe Mode:
[21:32:04][C][safe_mode:019]:   Boot considered successful after 60 seconds
[21:32:04][C][safe_mode:021]:   Invoke after 10 boot attempts
[21:32:04][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[21:32:04][C][api:140]: API Server:
[21:32:04][C][api:141]:   Address: esp8266heatingflow.local:6053
[21:32:04][C][api:143]:   Using noise encryption: YES
[21:32:07][D][api:103]: Accepted 192.168.1.11
[21:32:07][W][component:237]: Component api took a long time for an operation (288 ms).
[21:32:07][W][component:238]: Components should block for at most 30 ms.
[21:32:08][D][api.connection:1446]: Home Assistant 2024.11.2 ([redacted]): Connected successfully
[21:32:50][I][safe_mode:041]: Boot seems successful; resetting boot loop counter

lanrat, 2024-11-22:
Have you verified you have a physical pull-up resistor on the 1-wire data line? I had issues until I added one.

jddonovan, 2025-01-27:
This is driving me crazy as well. After update, it no longer works for me either. Three sensors and bluetooth scanning using MQTT.

After some fiddling, I got it working, but promptly broke after adding an interval timer for bluetooth scanning. I'm also guessing some timing issue with initialising the sensors as it seems to be dependent on other code.

1N4148, 2025-01-27:
Did you try the 2024.12 release? There was a change in the OneWire functions which made the whole stuff imho quite stable. No more scratchpad errors in my environment (ESP32, 48x DS18B20 with long wires on 4x 1-Wire bus). 

Before that, it was quite sensible to the framework (IDF worked, Arduino not) and enables/disenabled components.

thimo-seitz, 2025-01-28:
Es sieht bei mir so aus, als wenn es wieder läuft mit der Version 2024.12.4.

1N4148, 2025-01-28:
Sure: [https://github.com/esphome/esphome/pull/7957](url)

vincent-tr, 2025-01-28:
> Sure: [https://github.com/esphome/esphome/pull/7957](url)

Ah nice! 😍 
Tricky change, thanks for pointing it out!

vincent-tr, 2025-01-28:
Seems OK after the update here too (I have 3 esp32 boards with a DS18b20 each, one was failing)

Thanks a lot for the fix!

jddonovan, 2025-01-29:
For me it still doesn't work. Using 2024.12.4.

Neither the esp-idf framework nor arduino.

I get values from the sensors sporadically by replugging the ESP. However a software restart doesn't seem to have any effect and all I get are checksum errors. The setup worked without any issue before the update. Mind you, I did re-solder from pin 3 to pin 1. I'll try resoldering it to pin 2 and another ESP board.

1N4148, 2025-01-29:
GPIO1/3 are also assigned to the UART so I switched a long time ago to GPIO15 and higher.

jddonovan, 2025-02-01:
> GPIO1/3 are also assigned to the UART so I switched a long time ago to GPIO15 and higher.

Not sure which ESP you have, but ESP32-C3 is pretty safe for GPIO ports 1 to 3. GPIO20 and GPIO21 are the UART0 ports.

jddonovan, 2025-02-04:
> For me it still doesn't work. Using 2024.12.4.
>
> Neither the esp-idf framework nor arduino.
>
> I get values from the sensors sporadically by replugging the ESP. However a software restart doesn't seem [...]

Well I resoldered it to pin 2, so the setup is as close to the working ESP unit as possible. Really no clue. Dallas sensors (or rather cheap clones) work for a few minutes after pulling the power and reconnecting. But not always.

Simple restart/reflash doesn't help at all.

ssieb, 2025-03-06:
Can some people test this with ESP32 (any variant)?
```yaml
external_components:
  - source: github://ssieb/esphome@onewire2
    components: [ gpio, esp32, one_wire, dallas_temp ]
    refresh: 1min
```

fran6120, 2025-03-06:
> Can some people test this with ESP32 (any variant)?
> 
> external_components:
>   - source: github://ssieb/esphome@onewire2
>     components: [ gpio, esp32, one_wire, dallas_temp ]
>     refresh: 1min

I have an esp32-c3 and when compiling I get the following error:

```
src/esphome/components/esp32/gpio.cpp: In member function 'virtual esphome::ISRInternalGPIOPin esphome::esp32::ESP32InternalGPIOPin::to_isr() const':
src/esphome/components/esp32/gpio.cpp:49:25: error: cannot convert 'volatile gpio_dev_s::<unnamed union>*' to 'volatile uint32_t*' {aka 'volatile long unsigned int*'} in assignment
     arg->in_reg = &GPIO.in;
                         ^~
src/esphome/components/esp32/gpio.cpp:50:30: error: cannot convert 'volatile gpio_dev_s::<unnamed union>*' to 'volatile uint32_t*' {aka 'volatile long unsigned int*'} in assignment
     arg->out_set_reg = &GPIO.out_w1ts;
                              ^~~~~~~~
src/esphome/components/esp32/gpio.cpp:51:30: error: cannot convert 'volatile gpio_dev_s::<unnamed union>*' to 'volatile uint32_t*' {aka 'volatile long unsigned int*'} in assignment
     arg->out_clr_reg = &GPIO.out_w1tc;
                              ^~~~~~~~
src/esphome/components/esp32/gpio.cpp:52:31: error: cannot convert 'volatile gpio_dev_s::<unnamed union>*' to 'volatile uint32_t*' {aka 'volatile long unsigned int*'} in assignment
     arg->mode_set_reg = &GPIO.enable_w1ts;
                               ^~~~~~~~~~~
src/esphome/components/esp32/gpio.cpp:53:31: error: cannot convert 'volatile gpio_dev_s::<unnamed union>*' to 'volatile uint32_t*' {aka 'volatile long unsigned int*'} in assignment
     arg->mode_clr_reg = &GPIO.enable_w1tc;
                               ^~~~~~~~~~~
src/esphome/components/esp32/gpio.cpp:56:25: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'in1'; did you mean 'in'?
     arg->in_reg = &GPIO.in1.val;
                         ^~~
                         in
src/esphome/components/esp32/gpio.cpp:57:30: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1_w1ts'; did you mean 'out_w1ts'?
     arg->out_set_reg = &GPIO.out1_w1ts.val;
                              ^~~~~~~~~
                              out_w1ts
src/esphome/components/esp32/gpio.cpp:58:30: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1_w1tc'; did you mean 'out_w1tc'?
     arg->out_clr_reg = &GPIO.out1_w1tc.val;
                              ^~~~~~~~~
                              out_w1tc
src/esphome/components/esp32/gpio.cpp:59:31: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1ts'; did you mean 'enable_w1ts'?
     arg->mode_set_reg = &GPIO.enable1_w1ts.val;
                               ^~~~~~~~~~~~
                               enable_w1ts
src/esphome/components/esp32/gpio.cpp:60:31: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1tc'; did you mean 'enable_w1tc'?
     arg->mode_clr_reg = &GPIO.enable1_w1tc.val;
                               ^~~~~~~~~~~~
                               enable_w1tc
*** [.pioenvs/meteo-c3/src/esphome/components/esp32/gpio.cpp.o] Error 1
========================== [FAILED] Took 7.50 seconds ==========================
```

As an additional note, there are more components in the configuration yaml in order to test the changes in real conditions and not just your component.

Could they be the cause?

ssieb, 2025-03-06:
Ah, I didn't test compiling with a C3.  That's the problem with doing low-level access like this.  Things can be slightly different between the variants.  I'll test compiles on all the variants and see if I can sort this out.

ssieb, 2025-03-07:
I think I've fixed the compile for all the variants that I know of.

XneddyX, 2025-03-16:
> Can some people test this with ESP32 (any variant)?
> 
> external_components:
>   - source: github://ssieb/esphome@onewire2
>     components: [ gpio, esp32, one_wire, dallas_temp ]
>     refresh: 1min

I have flash your fork an let it run over night. I will inform you.

XneddyX, 2025-03-20:
for me it is working fine

closed 2025-06-07 by ssieb