esphome/issues #6261: bme68x_bsec2_i2c platform on ESP8266 triggers a boot loop

ndandanov, 2024-09-19:
### The problem

Enabling [`bme68x_bsec2_i2c`](https://esphome.io/components/sensor/bme68x_bsec2.html) on ESP8266 (D1 mini) triggers a boot loop. The exact same configuration using [`bme680_bsec`](https://esphome.io/components/sensor/bme680_bsec.html) instead boots properly and seems to work well so far.

### Which version of ESPHome has the issue?

2024.9.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

bme68x_bsec2_i2c

### Example YAML snippet

```yaml
esphome:
  name: aqm
# Example configuration entry
esp8266:
  board: d1_mini
  framework:
    version: recommended

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  - platform: esphome
    password: !secret esphome_ota_password
    # Forward this TCP port to do OTA updates remotely
    port: 8266

wifi:
  ssid: !secret esphome_wifi_ssid
  password: !secret esphome_wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret esphome_ap_ssid
    password: !secret esphome_ap_password
captive_portal:
# Web server to provide web interface
web_server:
  port: 80
  local: true

# Minimal example configuration with common sensors
i2c:
  sda: D2
  scl: D1

bme68x_bsec2_i2c:
  address: 0x77
  model: bme680
  operating_age: 28d
  sample_rate: LP
[... 309 lines cut ...]
[00:34:09]3fffff50:  3fff0570 00000002 3fff2184 4020ef38  
WARNING Decoded 0x4020ef38: esphome::bme68x_bsec2::BME68xBSEC2Component::setup()
[00:34:09]3fffff60:  00000000 3fff32d8 00000000 3fff2130  
[00:34:09]3fffff70:  3fff2184 00000004 3fff056c 4021b4a5  
WARNING Decoded 0x4021b4a5: esphome::Application::setup()
[00:34:09]3fffff80:  0000000d 0000000d 3fff3764 feefeffe  
[00:34:09]3fffff90:  00000014 3fff05b8 3ffe9909 feefeffe  
[00:34:09]3fffffa0:  feefeffe feefeffe feefeffe 3fff0b08  
[00:34:09]3fffffb0:  3fffdad0 00000000 3fff0adc 4022d020  
WARNING Decoded 0x4022d020: loop_wrapper() at core_esp8266_main.cpp
[00:34:09]3fffffc0:  feefeffe feefeffe 3fffdab0 4010048d  
WARNING Decoded 0x4010048d: cont_wrapper
[00:34:09]<<<stack<<<
[00:34:09]
[00:34:09]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[00:34:09]
```

### Additional information

Board schematic:
https://oshwlab.com/funnybrum/IAQ_Board

Original firmware used with the board:
https://github.com/funnybrum/AQM

ndandanov, 2024-09-30:
Perhaps @kbx81 @neffs can help, according to [`CODEOWNERS`](https://github.com/esphome/esphome/blob/20cb2e147fd6bae669facaeeaf967c588f4b55d1/CODEOWNERS) :)

neffs, 2024-09-30:
Could you try this?
https://github.com/esphome/esphome/pull/4585#issuecomment-2282656538

We should probably document the limitations of using the library and component with ESP8266 better...

ireun, 2024-09-30:
@neffs i've tried to no luck, NodeMCU v2 here..

ssieb, 2024-09-30:
Looking at the crash log, I would agree that it's most likely running out of RAM.  Try removing the web server and captive portal.

rkeicher, 2024-09-30:
I had the same boot loop issue with my Wemos D1 and this additional parameters fix it.
Thanks for this solution.

Could you try this?
esphome/esphome#4585 (comment)<https://github.com/esphome/esphome/pull/4585#issuecomment-2282656538>

ireun, 2024-09-30:
> Try removing the web server and captive portal

Disabled both, still bootloop, sadly

ndandanov, 2024-10-01:
> Could you try this? [esphome/esphome#4585 (comment)](https://github.com/esphome/esphome/pull/4585#issuecomment-2282656538)
> 
> We should probably document the limitations of using the library and component with ESP8266 better...

Many thanks, @neffs! This change made the board boot successfully and it seems to work good so far.

It works even with the  web server and captive portal.

I had read that there might be issues with the low amount of RAM on the ESP8266 D1 mini, but had not seen the comment you linked. Perhaps it would be useful to make it more visible, e.g. by placing the contents on the [BSEC2 component page](https://esphome.io/components/sensor/bme68x_bsec2.html).

mquade, 2024-10-02:
Thanks @neffs
Adding the Flag solved my issue #6295

closed 2025-03-31 by ndandanov