esphome/issues #6263: Integral sensor: trapezoid method - weird summing?

baudneo, 2024-09-21:
### The problem

I have set up an esphome device with 2 CT clamps to measure my 240 V dryers current. I noticed after setting up HASS and esphome integral sensors that there were some discrepancies between HASS and esphome sensor summing.

I have 2 integral sensors created in esphome and 2 in HASS (trapezoid and left methods). After accumulating data for a while, the esphome left, trapezoid and HASS left sensors all report the same data. Only the HASS trapezoid method is different, as shown in this screenshot:
[image]

The power sensors are updated every 5 seconds using an `on_value` hook, so HASS and esphome are both getting the same data at approx the same interval (esphome isn't getting more data to work with than HASS, AFAIK).

> [!NOTE]
> [esphome docs](https://esphome.io/components/sensor/integration.html#integration-sensor) say that the default `integration_method` is `trapezoid`.

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9

### What platform are you using?

ESP32

### Board

yd-esp32-23 AKA esp32-s3-devkitc-1 (clone)

### Component causing the issue

integral sensor

### Example YAML snippet

```yaml
# Dryer with 2 30A CT Clamps on dual-pole 40A 240V circuit
esphome:
  name: esp32s3-test
  friendly_name: esp32s3 Dryer
  

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:
  level: INFO
# Enable Home Assistant API
api:
  encryption:
    key: "<redacted>"

ota:
  - platform: esphome
    password: "<redacted>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 10.0.2.31

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32S3-Test Fallback Hotspot"
    password: !secret hotspot_pw

captive_portal:
    
i2c:
  id: i2c0
  scl: 9
  sda: 8

ads1115:
  address: 0x48
  # Needed for CT clamp
  continuous_mode: on
  id: ads0

substitutions:
  friendly_name: esp32-dryer

sensor:
  - platform: ads1115
    multiplexer: 'A0_GND'
    gain: 6.144
    name: "${friendly_name} - A0 - Voltage"
    accuracy_decimals: 5
    internal: True
    id: ads_a0
    ads1115_id: ads0
    update_interval: 100ms

  - platform: ct_clamp
    # 1V/30A clamp
    sensor: ads_a0
    id: raw0
    name: "${friendly_name} - A0 - Raw"
    internal: True
    sample_duration: 300ms
    update_interval: 1000ms
    accuracy_decimals: 5

  - platform: ct_clamp
    # 1V/30A clamp
    # YHDC branded
    sensor: ads_a0
    id: current0
    name: "${friendly_name} - L2 Amps"
    sample_duration: 300ms
    update_interval: 1000ms
    accuracy_decimals: 5
    # calibrate with a multimeter
    filters: 
    #   # raw -> measured
      - calibrate_linear: 
        - 0 -> 0
        - 0.004 -> 0.13
        - 0.141 -> 4.394
        - 0.175 -> 5.36
        - 0.72875 -> 25.72
        - 0.833 -> 26.71
      - clamp:
          min_value: 0
          max_value: 30
          ignore_out_of_range: false

  - platform: ads1115
    multiplexer: 'A1_GND'
    gain: 6.144
    name: "${friendly_name} - A1 - Voltage"
    accuracy_decimals: 5
    internal: True
    id: ads_a1
    ads1115_id: ads0
    update_interval: 100ms

  - platform: ct_clamp
    # 1V/30A clamp
    # non YHDC
    sensor: ads_a1
    id: raw1
    name: "${friendly_name} - A1 - Raw"
    internal: True
    sample_duration: 300ms
    update_interval: 1000ms
    accuracy_decimals: 5

  - platform: ct_clamp
    # 1V/30A clamp
    # non YHDC
    sensor: ads_a1
    id: current1
    name: "${friendly_name} - L1 Amps"
    sample_duration: 300ms
    update_interval: 1000ms
    accuracy_decimals: 5
    # calibrate with a multimeter
    filters: 
    #   # V -> A
      - calibrate_linear: 
        - 0 -> 0
        - 0.15146 -> 4.00
        - 0.90015 -> 25.24
        - 0.91476 -> 26.14
      - clamp:
          min_value: 0
          max_value: 30
          ignore_out_of_range: false
  
  - platform: homeassistant
    entity_id: sensor.0x70b3d52b600114bb_voltage
    id: washer_plug_volts
    internal: True

  - platform: homeassistant
    entity_id: sensor.ac_frigidaire_plug_voltage
    id: ac_plug_volts
    internal: True

  - platform: template
    name: "${friendly_name} - HASS Voltage"
    id: voltage
    device_class: voltage
    state_class: measurement
    unit_of_measurement: V
    internal: True
    lambda: |-
      int volts = 120;
      if ( !(isnan( id(washer_plug_volts).state ) ) ) {
        volts = id(washer_plug_volts).state;
      } else if ( !isnan( id(ac_plug_volts).state ) ) {
        volts = id(ac_plug_volts).state;
      }
      return float(volts);
    update_interval: 5s
    on_value: 
      then:
        - lambda: |-
            float voltage = x;
            float power_L1 = voltage * id(current1).state;
            float power_L2 = voltage * id(current0).state;
            id(watts0).publish_state(power_L2);
            id(watts1).publish_state(power_L1);
            id(total_power).publish_state(power_L1 + power_L2);
            

  - platform: template
    name: "${friendly_name} - Total Power"
    id: total_power
    device_class: power
    state_class: measurement
    unit_of_measurement: W
    lambda: return 0;
    update_interval: never

  - platform: integration
    name: "${friendly_name} - Energy (trapezoid)"
    sensor: total_power
    id: total_energy
    time_unit: h
    #restore: true
    unit_of_measurement: kWh
    state_class: total
    device_class: energy
    # if in Wh turn to kWh
    filters: 
      - multiply: 0.001
  
  - platform: integration
    name: "${friendly_name} - Energy (left)"
    sensor: total_power
    id: total_energy_left
    time_unit: h
    integration_method: left
    #restore: true
    unit_of_measurement: kWh
    state_class: total
    device_class: energy
    # if in Wh turn to kWh
    filters: 
      - multiply: 0.001
  
  - platform: template
    name: "${friendly_name} - L2 Watts"
    id: watts0
    device_class: power
    state_class: measurement
    unit_of_measurement: W
    lambda: return 0;
    # let the voltage sensor set the state
    update_interval: never

  - platform: template
    name: "${friendly_name} - L1 Watts"
    id: watts1
    device_class: power
    state_class: measurement
    unit_of_measurement: W
    lambda: return 0;
    # let the voltage sensor set the state
    update_interval: never
  
  - platform: template
    name: "${friendly_name} - Total Amps"
    id: total_current
    device_class: current
    state_class: measurement
    unit_of_measurement: A
    lambda: return (id(current0).state + id(current1).state);
    update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
N/A
```

### Additional information

I opened [an issue in the HASS repo](https://github.com/home-assistant/core/issues/126356) and they pointed out that the Left and Trapezoid methods should not be returning the same value, so I thought I should open an issue here to see if there may be something going on.

ssieb, 2024-09-22:
I don't see how the values could be that different.  There must be a data issue somewhere.  Have you looked at the data?  The esphome code for trapezoid is correct.  You could try specifying that it's trapezoid to make sure it's using that.
You shouldn't have 2 ct_clamp sensors on the same input like that.  It's likely to cause you problems.  Use copy sensors for that.

baudneo, 2024-09-22:
Source data is consistent, I just recreated the trapezoid sensor with the method specified and will report back. Thanks for the tip about the copy sensor!
[image]

emefff, 2024-09-23:
I had a similar issue, I think (it has been a while). 
Do you have longer periods with a zero value? Then the first trapezoid can be very large, as it 'connects' to the last zero value, maybe from the day before. I think, that's what happened in my case, I don't have  a better explanation. 
What I did was to 'inject' an artificial power value into the data stream, at regular intervals. This way, my values got better. I called it 'tiny_power_value', with 1W it is tiny and does not contribute much to the energy value. What I actually do is, setting the numerical power value of this data stream equal to the 'tiny_power_value' with an automation, every hour or so. Maybe it helps.

baudneo, 2024-09-23:
After a day of ingesting days, there is a small difference. If you set precision to 3, they converge and then differ briefly.

[image]

closed 2024-09-23 by baudneo

ssieb, 2024-09-23:
I wonder if the reason the HA integration was so different was because of what @emefff mentioned.  I think HA drops duplicate values, so if it's 0 for a long time, when there's finally a higher value, the trapezoid has become really big.

baudneo, 2024-09-24:
[image]
[image]
[image]
[image]

ssieb, 2024-09-24:
What are you trying to show?  I'm guessing those are the esphome integration sensors and they're close as expected.  It was the HA trapezoid one that was way off.

baudneo, 2024-09-24:
Sorry, just more data for anyone in the future. The HASS one is the one I use for the energy dashboard and unrelated. I was more showing how they converge and differ slightly.