esphome/issues #6266: CONF_MODE instead of CONF_CHANNEL in i2s_audio/speaker/__init__.py

gbesacier, 2024-09-21:
### The problem

In this file https://github.com/esphome/esphome/blame/8e5d7337c8e9490c901cd17f59b72ed300f775b1/esphome/components/i2s_audio/speaker/__init__.py#L114 it should be
`cg.add(var.set_internal_dac_mode(config[CONF_MODE]))`
instead of
`cg.add(var.set_internal_dac_mode(config[CONF_CHANNEL]))`

The error was introduced two weeks ago. Committers were @nielsnl68 and @jesserockz

### Which version of ESPHome has the issue?

2024.9.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

zdenekchobot, 2024-10-28:
I confirm the error in esphome 2024.10.2, board esp32dev, framework esp-idf. Error during compilation:

/config/esphome/alarm-clock.yaml: In function 'void setup()':
/config/esphome/alarm-clock.yaml:384:63: error: no matching function for call to 'esphome::i2s_audio::I2SAudioSpeaker::set_internal_dac_mode(i2s_channel_fmt_t)'
                                                               ^
In file included from src/esphome.h:30,
                 from src/main.cpp:3:
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.h:50:8: note: candidate: 'void esphome::i2s_audio::I2SAudioSpeaker::set_internal_dac_mode(i2s_dac_mode_t)'
   void set_internal_dac_mode(i2s_dac_mode_t mode) { this->internal_dac_mode_ = mode; }
        ^~~~~~~~~~~~~~~~~~~~~
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.h:50:8: note:   no known conversion for argument 1 from 'i2s_channel_fmt_t' to 'i2s_dac_mode_t'

yggi, 2025-04-03:
I ran into a similar problem: 

Error during compilation:
```
src/main.cpp: In function 'void setup()':
src/main.cpp:209:40: error: cannot convert 'i2s_channel_fmt_t' to 'i2s_dac_mode_t'
  209 |   cyd_speaker->set_internal_dac_mode(::I2S_CHANNEL_FMT_ONLY_RIGHT);
      |                                      ~~^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                        |
      |                                        i2s_channel_fmt_t
In file included from src/esphome.h:26:
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.h:35:45: note:   initializing argument 1 of 'void esphome::i2s_audio::I2SAudioSpeaker::set_internal_dac_mode(i2s_dac_mode_t)'
   35 |   void set_internal_dac_mode(i2s_dac_mode_t mode) { this->internal_dac_mode_ = mode; }
      |                              ~~~~~~~~~~~~~~~^~~~
```

from this minimal config:

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf

i2s_audio:
  i2s_lrclk_pin: GPIO25
    
speaker:
  - platform: i2s_audio
    dac_type: internal
    mode: stereo
```

ElektronikCZ, 2025-04-15:
I have the same issue with an internal DAC. The internal DAC usage is quite poorly/confusingly documented, and seemingly neglected.