esphome/issues #6267: ESPTime::recalc_timestamp_local() is handling DST time wrongly

zvirja, 2024-09-22:
### The problem

Current implementation of ESPTime::recalc_timestamp_local() is removing an extra time for DST. It's not needed and as result it generates wrong output. Tested with `Europe/Copenhagen` timezone and time now.

### Which version of ESPHome has the issue?

 2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Additional information

Current implementation is

```c++
void ESPTime::recalc_timestamp_local(bool use_day_of_year) {
  this->recalc_timestamp_utc(use_day_of_year);
  this->timestamp -= ESPTime::timezone_offset();
  ESPTime temp = ESPTime::from_epoch_local(this->timestamp);
  if (temp.is_dst) {
    this->timestamp -= 3600;
  }
}
```

Just remove the following part:

```c++
  if (temp.is_dst) {
    this->timestamp -= 3600;
  }
```

as `ESPTime::timezone_offset()` is correctly calculates the offset and handles summer time inside.

zvirja, 2024-09-22:
Tagging @jesserockz as you added this in https://github.com/esphome/esphome/pull/6513

ssieb, 2024-09-22:
We've been looking at the time stuff and this is one of the issues.

blackjid, 2024-11-06:
I'm having this issue too.

eyal0, 2025-08-26:
`ESPTime::timezone_offset()` does not work correctly when the current time is near midnight at the user's timezone.  Even with removing the lines that you suggested, I think that you will find that the timezone offset is off by 24 hours sometimes.

There is a fix in #10426 that fixes ESPTime::timezone_offset() to correctly compute the timezone offset for the current moment, including dst.

closed 2025-09-02 by jesserockz

zvirja, 2025-09-04:
@eyal0 Amazing! I see that the method from the bug description was changed as well, so the issue must be fixed by now!

Thanks for getting back to me and for making it a bit better 😊